Marus/cortex-debug #288: Loading SVD from Device Support Pack

dwalkes, 2020-05-10:
Hi,

Thanks for sharing this project!

I'm attempting to use with [Renode](https://renode.readthedocs.io/en/latest/introduction/supported-boards.html) and this launch configuration on [this test project](https://github.com/Trellis-Logic/bluepill-starter):
```
        {
            "showDevDebugOutput": true,
            "cwd": "${workspaceRoot}",
            "executable": "./build/bluepill-starter.elf",
            "name": "Debug Renode",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "external",
            "preLaunchTask": "Build STM",
            "gdbTarget": "localhost:3333",
            "device": "STM32F103"
        }
```
Then manually starting Renode using the instructions [here](https://renode.readthedocs.io/en/latest/debugging/gdb.html#connecting-to-gdb).  
```
set bin @<path to working directory>/bluepill-starter/build/bluepill-starter.elf
s @scripts/single-node/stm32f103.resc
machine StartGdbServer 3333
```
I'm able to use interactive debug, however I haven't been able to come up with the correct settings for the `device` parameter to load the svd file, despite the fact that I"ve loaded the [STM32F1 device support pack](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug-dp-stm32f1&ssr=false#qna).  Each time I see `No SVD File Loaded: undefined".  If I replace the `device` attribute with:
```
            "svdFile": "./STM32F103.svd",
```
Where the .svd file is taken from [this link](https://raw.githubusercontent.com/fduignan/stm32f103c8t6/master/STM32F103.svd) the peripherals view works.

I'm wondering how I should specify the `device` argument to get it to find the svd file so I don't need to specify it myself as a workaround.

The README link to https://marcelball.ca/projects/cortex-debug/ is down for me today, perhaps it would be explained there.

Thanks for sharing and for any suggestions you have to resolve this.

haneefdm, 2020-05-11:
I am assuming you installed the support pack in vscode, which btw, I am not sure is being maintained. You can see the entire support pack here.

https://github.com/Marus/cortex-debug-dp-stm32f1

Btw, did you try `"device": "STM32F103xx"`. I believe it is expecting two alpha-numeric chars after the device's major type. The `device` setting is also used for other purposes and it needs to match to a real device

haneefdm, 2020-05-11:
Btw, it is safer to use the `svdFile` setting to specify. In your case, since you are using an `external` gdb server, the `device` setting is not really used for anything else. Getting the svd file direct from the MCU vendor is what I highly recommend.

https://community.st.com/s/question/0D50X00009XkWDkSAN/how-does-st-manage-svd-files

There can be bug fixes in SVD files that you may not be getting if you get files distribute by others.

dwalkes, 2020-05-11:
Thanks for the response!

> I am assuming you installed the support pack in vscode

I did.

> Btw, did you try "device": "STM32F103xx".

I did not, but I confirm this does work.

> Getting the svd file direct from the MCU vendor is what I highly recommend.

Thanks for the advice!

I've summarized in a new wiki page at https://github.com/Marus/cortex-debug/wiki/Specifying-SVD-Files.

closed 2020-05-11 by dwalkes