Marus/cortex-debug #290: Project website down?

itsmnjn, 2020-05-13:
I've tried to access the [project's website](https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/) but it seems like it's down. 

Is the owner working on updating it? I'm very much interested in using this plugin, but the documentation in the repo is pretty sparse, and it seems like there are a lot of configuration options.

akospasztor, 2020-05-14:
I was just wondering about the same, I have been unable to access the website for several days.
I remembered that in the past, there were two pages explaining the launch configuration and the swo configuration. It might be outdated, but here are some archive-snapshots of the webpage:
- https://web.archive.org/web/20180903235853/https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/
- https://web.archive.org/web/20180904022232/https://marcelball.ca/projects/cortex-debug/cortex-debug-swo-decoding-and-graphing/

itsmnjn, 2020-05-14:
Those links help very much, thanks!

closed 2020-05-14 by haneefdm

haneefdm, 2020-05-14:
We will be moving the documentation to the Wiki. There is already some content there

https://github.com/Marus/cortex-debug/wiki