Marus/cortex-debug #291: runToMain not working

victorcanoz, 2020-05-14:
In my debug configuration, `"runToMain": true` however the debugger does not stop at the start of the main function.
I'm able to set breakpoints elsewhere but not in the main function.

I am using FreeRTOS on a LPC55S69 cortex M33.
Below is the configuration:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "Debug/ferlim-detecteur-foulee.axf",
            "serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "device": "LPC55S69",
            "interface": "swd", // or "jtag" - but usually "swd"
            "serialNumber": "", //if you have more than one J-Link probe add the serial number here 
            "runToMain": true,
        }
    ]
  }
```
Am I missing something?

haneefdm, 2020-05-14:
Your launch.json looks good.

Does manually setting a breakpoint in main (with `"runToMain": false`) work?

When your program stops somewhere (or pause it), can you do an `info breakpoints` in Debug Console and see what gdb tells us.

victorcanoz, 2020-05-15:
I have no idea why whatsoever, but today when I tried again, the debugger successfully stopped in the main!
I just hope it will not happen again.
Thanks anyway

closed 2020-05-15 by victorcanoz