Marus/cortex-debug #292: SAMD21 ERROR: Failed to read current instruction. before Reset_Handler()

ASL07, 2020-05-14:
Hi,

Hope I can get some help with this issue.

Everything was working fine. But suddenly, debugging stopped working (I can't find the reason).
Flashing works fine but debugging crashes/hangs just before `void Reset_Handler(void)`

[image]

Here is my configuration:
```
"cwd": "${workspaceRoot}",
    "executable": "/Users/developer/Code/iplacecontrollerv2/.pio/build/prodino_gprs/firmware.elf",
    "name": "Cortex Debug",
    "svdFile": "/Users/developer/.platformio/platforms/atmelsam/misc/svd/ATSAMD21G18A.svd",
    "request": "launch",
    "interface": "swd",
    "type": "cortex-debug",
    "device": "ATSAMD21G18",
    "runToMain": false,
    "servertype": "jlink",
    "showDevDebugOutput": true
```

This is shown in OUTPUT window:

```
Downloading 16016 bytes @ address 0x00002000
Downloading 15984 bytes @ address 0x00005E90
Downloading 15984 bytes @ address 0x00009D00
Downloading 5800 bytes @ address 0x0000DB70
Downloading 92 bytes @ address 0x0000F218
Downloading 752 bytes @ address 0x0000F274
Writing register (PC = 0x    7038)
Received monitor command: reset
Resetting target
Read 4 bytes @ address 0x00007038 (Data = 0x4A12B510)
Reading register (MSP = 0xFFFFFFFC)
Reading register (PSP = 0xB4FFF7FC)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       0)
Read 1 bytes @ address 0x00007038 (Data = 0x10)
Performing single step...
ERROR: Failed to read current instruction.
...Target halted (Vector catch, PC = 0xFFFFFFFE)
Reading all registers
WARNING: Failed to read memory @ address 0xFFFFFFFE
Read 4 bytes @ address 0x2000032E (Data = 0xB570BD01)
Reading 64 bytes @ address 0x20000300
Reading register (MSP = 0xFFFFFFFC)
Reading register (PSP = 0xB4FFF7FC)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       0)
Read 4 bytes @ address 0x2000032E (Data = 0xB570BD01)

```

And the DEBUG CONSOLE shows:
```
0000004642: 24-var-create var_pSrc_256 * "pSrc"
0000004644: GDB -> App: {"token":24,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_pSrc_256"],["numchild","1"],["value","<optimized out>"],["type","uint32_t *"],["thread-id","1"],["has_more","0"]]}}
0000004646: 25-var-update --thread 1 --frame 0 --all-values var_pDest_256
0000004648: GDB -> App: {"token":25,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
0000004649: 26-var-create var_pDest_256 * "pDest"
0000004649: GDB -> App: {"token":26,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_pDest_256"],["numchild","1"],["value","<optimized out>"],["type","uint32_t *"],["thread-id","1"],["has_more","0"]]}}
0000010687: 27-data-evaluate-expression --thread 1 --frame 0 Reset_Handler
0000010687: GDB -> App: {"token":27,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["value","{void (void)} 0x7038 <Reset_Handler>"]]}}
0000066566: 28-stack-info-frame --thread 1 --frame 0
0000066566: GDB -> App: {"token":28,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["frame",[["level","0"],["addr","0x00007038"],["func","Reset_Handler"],["file","/Users/developer/.platformio/packages/framework-arduino-samd/cores/arduino/cortex_handlers.c"],["fullname","/Users/developer/.platformio/packages/framework-arduino-samd/cores/arduino/cortex_handlers.c"],["line","141"],["arch","armv6s-m"]]]]}}
0000066567: 29-exec-next --thread 1
0000066567: GDB -> App: {"token":29,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
0000066567: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
0000066582: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["reason","end-stepping-range"],["frame",[["addr","0xfffffffe"],["func","??"],["args",[]],["arch","armv6s-m"]]],["thread-id","1"],["stopped-threads","all"]]}]}
0000066604: 30-thread-list-ids
0000066604: GDB -> App: {"token":30,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
0000066605: 31-thread-info 1
0000066605: GDB -> App: {"token":31,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 57005"],["frame",[["level","0"],["addr","0xfffffffe"],["func","??"],["args",[]],["arch","armv6s-m"]]],["state","stopped"]]]]]}}
0000066610: 32-data-list-register-values N
0000066615: GDB -> App: {"token":32,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","0"],["value","0"]],[["number","1"],["value","0"]],[["number","2"],["value","0"]],[["number","3"],["value","1619391766"]],[["number","4"],["value","536872548"]],[["number","5"],["value","536872636"]],[["number","6"],["value","536872636"]],[["number","7"],["value","2875850736"]],[["number","8"],["value","4194287483"]],[["number","9"],["value","4226500351"]],[["number","10"],["value","3154105851"]],[["number","11"],["value","4227194345"]],[["number","12"],["value","536872408"]],[["number","13"],["value","0xfffffffc"]],[["number","14"],["value","536871727"]],[["number","15"],["value","0xfffffffe"]],[["number","25"],["value","1627389952"]],[["number","91"],["value","4294967292"]],[["number","92"],["value","3036674044"]],[["number","93"],["value","0"]],[["number","94"],["value","0"]],[["number","95"],["value","0"]],[["number","96"],["value","0"]]]]]}}
0000066636: 33-stack-list-frames --thread 1 0 20
0000066636: 34-stack-list-frames --thread 1 0 20
0000066637: GDB -> App: {"token":33,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0xfffffffe"],["func","??"],["arch","armv6s-m"]]],["frame",[["level","1"],["addr","0x2000032e"],["func","device"],["arch","armv6s-m"]]]]]]}}
0000066639: GDB -> App: {"token":34,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0xfffffffe"],["func","??"],["arch","armv6s-m"]]],["frame",[["level","1"],["addr","0x2000032e"],["func","device"],["arch","armv6s-m"]]]]]]}}
```

What I have tried to fix it:
Setting runToMain to true/false

Run info breakpoints after the error:
returns `No breakpoints or watchpoints`

Erasing flash and reflashing the bootloader

Tried debugging a different application

haneefdm, 2020-05-14:
If nothing changed with VSCode or this extension, your application didn't change either? If JLink cannot read the PC or any address, then gdb or Cortex-Debug can't work either. Does the PC look right to you?

My guess, you have hit a hard fault. I would backtrack your source code changes and figure what is causing it.

ASL07, 2020-05-16:
Hi,
I have found the problem. Somehow my bootloader or fuse configuration was corrupted.
In case someone else has the same problem, reflash the bootloader following https://learn.adafruit.com/how-to-program-samd-bootloaders/programming-the-bootloader-with-atmel-studio

closed 2020-05-16 by ASL07