Marus/cortex-debug #295: Variables window <unknown>

microwavesafe, 2020-05-21:
I am having a problem with the Variables window. Seemingly randomly the local variables will all be set to <unknown> on a break point. I have checked and the call stack looks OK and you can also look at the variables by setting a watch.

[image]

Once this has happened the local variables will not be shown again until I disconnect the debugger and reprogram / reconnect the target.

I am using version 0.3.7-pre2 of the extension and version 1.45.1 VSCode. I've attached the debug log output, hopefully it will help.

[variable_unknown_log.txt](https://github.com/Marus/cortex-debug/files/4661310/variable_unknown_log.txt)

microwavesafe, 2020-05-21:
A couple of updates. I have now seen the local variables come back to life after a few break point hits, but then disappear again. Also I have seen this error pop up.

```Could not get stack variables: Selected thread is running. (from stack-list-variables --thread 1 --frame 0 --simple-values)```

Which may have something to do with it?

RisinT96, 2020-06-06:
I've been having similar problems since a recent vscode update (can't say which one excatly), the variables in local/watch would stop loading, which usually requires re-starting the debug session, sometimes even that doesn't help, makes debugging a pain.

It's probably not a problem with the gdb interface since viewing variables through mouse-hover still works properly.

haneefdm, 2020-06-08:
I cannot duplicate this issue. I primarily use a Mac. Which platform is this happening on? The log file attached by @microwavesafe seems like Linux.  @RisinT96 Are you seeing this on Linux as well? Please respond. My current Linux setup is a bit botched.

There have been unexpected timing/ordering differences with VSCode and how it interacts with debuggers but something is still amiss.

Maybe I need more detailed steps to reproduce the problem. Any help is appreciated.

RisinT96, 2020-06-08:
Hello, I've been experiencing this on multiple Windows 10 setups, when using an RTOS plug-in (JLink GDB Server), the thread list is also affected sometimes.

microwavesafe, 2020-06-08:
I am on Linux. I think it may be JLink hardware related. Recently I've been doing some prototype code on an stm32f4-discovery board. I converted the onboard debugger to JLink compatible and under this arrangement it hasn't happened once. The onboard debugger is slower than the JLink plus, so I was wondering if this held any clues?

haneefdm, 2020-06-08:
Ohh, could this be RTOS related? One of the things that happen with RTOSes is that the thread information is totally bogus until after the C initialization runs and after the RTOS has actually been initialized. However, everything looks fine afterward. Generally, by the you get to main, it should stabilize (depends on the RTOS). There is not much this extension can do.

pyocd deals with it more elegantly.

If you have not enabled RTOS awareness with OpenOCD/JLink but are still seeing this issue, then we have a different problem.

microwavesafe, 2020-06-08:
> If you have not enabled RTOS awareness with OpenOCD/JLink but are still seeing this issue, then we have a different problem.

I only have this in my settings. I believe this means I haven't enabled RTOS awareness.

```            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "device": "STM32F070CB",
            "executable": "${workspaceRoot}/build/bin/output.elf",
            "request": "launch",
            "type": "cortex-debug",
            "showDevDebugOutput": true,
            "servertype": "jlink",
```

RisinT96, 2020-06-08:
Happens with and without RTOS support enabled.

I not mentioned that when RTOS support is enabled, it also sometimes breaks the stack view.

haneefdm, 2020-06-08:
Yes, JLink/OpenOCD provides junk stack/thread information (sometimes from the last run or whatever junk there is in SRAM). It only applies when threads are enabled and there is no fix. OpenOCD is especially bad with threads (too many bugs).

I will continue looking. Thanks. The following is a big clue to me...

https://github.com/Marus/cortex-debug/issues/295#issuecomment-631972127

RisinT96, 2020-06-08:
Hi again,
Playing around with a simple project, I managed to break the watch window, this is what it look like:
[image]
(the little wheel is spinining).
Unable to add or remove items in the watch window, nor interact with the expendable sections.

I had no error pop-up at all, don't see anything unusual in the debug console, nor the adapter output.

All it took was to set multiple breakpoints, and step into some functions, I'll try to pinpoint exactly what causes this.

JLink info:
```
SEGGER J-Link GDB Server V6.72d Command Line Version

JLinkARM.dll V6.72d (DLL compiled May 15 2020 16:47:06)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32L476VGTx
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32L476VGTx
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link STLink V21 compiled Aug 12 2019 10:29:20
Hardware: V1.00
S/N: 772535930
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 50000
```

haneefdm, 2020-06-08:
I have seen that spinning thing when the data structure is too large and/or user steps/continues too fast before the GUI has a chance to update after the previous pause. This is something new though.

Yes, the quick succession of commands is a problem. Requests from VSCode to the debugger  are getting queued up and they randomly fail because the context has totally changed for old requests. Again, this started with a recent update with VSCode. It doesn't mean we don't have a bug.

I am as frustrated as you all are and I can't even duplicate the problem. But I know it exists.

haneefdm, 2020-06-13:
Could you please try the following pre-release

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre3

Use the VSIX manual installation method to install the extension

RisinT96, 2020-06-26:
I tried using the new release, at first look it seems to work better Ethan the previous.
Haven't yet gotten the chance to simulate the behavior that breaks the watch/variables windows.
I'll update next week once I get a chance to test it.

haneefdm, 2020-06-26:
Note: V0.3.7 is released and now in the wild.

microwavesafe, 2020-06-29:
I have been using this for a few days and it seems better. Interestingly I still get the occasional 

``Could not get stack variables: Selected thread is running. (from stack-list-variables --thread 1 --frame 0 --simple-values)``

error, but this does not seem to affect the local variable window. It looks promising!

microwavesafe, 2020-07-01:
I am seeing <unknown> again today. It is the same problem as before, the variable in the Local window states <unknown>, but if I add the same variable to the watch window it display correctly.

sourcebox, 2020-08-16:
This issue also happens to me. What I noticed is that after initially starting the debug session, the variables are displayed correctly. Then, after doing some code modifications and pressing the restart button, the session will relaunch, but the variables all show "unknown". Hovering a variable in the source window or adding it to the watch list, will however show its value.

RisinT96, 2021-02-12:
Well, it seems like the issue still persists, sorry for the delay but the patches didn't fix it.
 Happens more rarely but still happens once in a while.

AndrewCapon, 2022-03-08:
Is there any fix for this, it happens all the time here. Basically the Variables window just doesn't work for me.

Hovering over a variable in the editor displays values perfectly, as well as the p command in the "Debug Console"

haneefdm, 2022-03-08:
@AndrewCapon  Not sure what release version you are using. I have been unable to truly reproduce this issue. It is unclear to me what steps to follow to reproduce this issue. I don't use JLink that frequently though. What is your setup? Did you customize your restart? Perhaps a copy of your launch.json?

But quite a bit has changed over the last few months. Do you mind trying the latest pre-release? You can enable the pre-release from within VSCode.

[image]

After you enable/install, it will look like this.

[image]

AndrewCapon, 2022-03-08:
Hi @haneefdm,

Thanks for the reply, unfortunately I am already using the Pre-Release version. So this is the version I am having the problems with.

haneefdm, 2022-03-08:
Could you enable debug `"showDevDebugOutput": "raw"` and copy all the text from the Debug Console and attach the file? It may be large. Maybe I can spot something.

I am also using Big Sur and st-link V3

AndrewCapon, 2022-03-08:
Thanks very much for your interest and help with this.

So this is after a start, hit breakpoint, restart (build), hit breakpoint and local not working:

```
29-interpreter-exec console "load"
-> ~"`/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf' has changed; re-reading symbols.\n"
`/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf' has changed; re-reading symbols.
-> ~"Loading section .text, size 0x28450 lma 0x24000000\n"
Loading section .text, size 0x28450 lma 0x24000000
-> 29+download,{section=".text",section-size="164944",total-size="2438981"}
-> 29+download,{section=".text",section-sent="16160",section-size="164944",total-sent="16160",total-size="2438981"}
-> 29+download,{section=".text",section-sent="164944",section-size="164944",total-sent="164944",total-size="2438981"}
-> ~"Loading section .exidx, size 0x8 lma 0x24028450\n"
Loading section .exidx, size 0x8 lma 0x24028450
-> 29+download,{section=".exidx",section-size="8",total-size="2438981"}
-> ~"Loading section .init_array, size 0x2c lma 0x24028458\n"
Loading section .init_array, size 0x2c lma 0x24028458
-> 29+download,{section=".init_array",section-size="44",total-size="2438981"}
-> ~"Loading section .fini_array, size 0x4 lma 0x24028484\n"
Loading section .fini_array, size 0x4 lma 0x24028484
-> 29+download,{section=".fini_array",section-size="4",total-size="2438981"}
-> ~"Loading section .data, size 0xdcc lma 0x24028488\n"
Loading section .data, size 0xdcc lma 0x24028488
-> 29+download,{section=".data",section-size="3532",total-size="2438981"}
-> ~"Loading section .isr_vector, size 0x298 lma 0x8000000\n"
Loading section .isr_vector, size 0x298 lma 0x8000000
-> 29+download,{section=".isr_vector",section-size="664",total-size="2438981"}
-> 29+download,{section=".isr_vector",section-sent="664",section-size="664",total-sent="169196",total-size="2438981"}
-> ~"Start address 0x24000894, load size 169196\n"
Start address 0x24000894, load size 169196
-> ~"Transfer rate: 110 KB/sec, 10574 bytes/write.\n"
Transfer rate: 110 KB/sec, 10574 bytes/write.
-> 29^done
30-interpreter-exec console "enable breakpoint"
-> =breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x24006d6e",func="KnobParameter::Process()",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",thread-groups=["i1"],times="1",original-location="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp:92"}
-> =breakpoint-modified,bkpt={number="2",type="breakpoint",disp="keep",enabled="y",addr="0x24006de8",func="KnobParameter::ResetActualValue(float)",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="105",thread-groups=["i1"],times="0",original-location="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp:105"}
-> 30^done
31-interpreter-exec console "monitor reset"
-> 31^done
32-interpreter-exec console "monitor reset halt"
-> @"target halted due to debug-request, current mode: Thread \n"
target halted due to debug-request, current mode: Thread
-> @"xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000, semihosting\n"
xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000, semihosting
-> 32^done
33-exec-continue
-> 33^running
-> *running,thread-id="all"
mi2.status = running
-> =breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x24006d6e",func="KnobParameter::Process()",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",thread-groups=["i1"],times="2",original-location="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp:92"}
-> ~"\n"

-> ~"Breakpoint 1, KnobParameter::Process (this=0x2000c7e4 <mainUI+32>) at KnobParameter.cpp:92\n"
Breakpoint 1, KnobParameter::Process (this=0x2000c7e4 <mainUI+32>) at KnobParameter.cpp:92
-> ~"92\t\treturn m_bChanged;\n"
92		return m_bChanged;
-> *stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c7e4 <mainUI+32>"}],file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
34-thread-list-ids
-> 34^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
35-thread-info 1
-> 35^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c7e4 <mainUI+32>"}],file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},state="stopped"}]
36-data-list-register-values N
37-stack-info-depth --thread 1 10000
-> 36^done,register-values=[{number="0",value="536922084"},{number="1",value="0"},{number="2",value="0"},{number="3",value="536922084"},{number="4",value="536922084"},{number="5",value="604148308"},{number="6",value="0"},{number="7",value="537001576"},{number="8",value="0"},{number="9",value="0"},{number="10",value="0"},{number="11",value="0"},{number="12",value="-1"},{number="13",value="0x2001fe68"},{number="14",value="604007625"},{number="15",value="0x24006d6e <KnobParameter::Process()+438>"},{number="25",value="1627389952"},{number="58",value="0"},{number="59",value="6755399441055744 [... cut ...]
-> 40^error,msg="-var-create: unable to create variable object"
watch: -var-create: unable to create variable object (from var-create watch_2527785aec91c116b56d99c12d975602dda5f351735d1cf7bf2b0323ae0f173b @ "m_modSlotsStorage[0]")41-stack-list-variables --thread 1 --frame 0 --simple-values
-> 41^done,variables=[{name="this",arg="1",type="KnobParameter * const",value="0x2000c7e4 <mainUI+32>"}]
42-var-update --thread 1 --frame 0 --all-values var_this_256
-> 42^done,changelist=[{name="var_this_256",in_scope="invalid",has_more="0"}]
```

AndrewCapon, 2022-03-08:
If I then try to expand `this` in the local window I get this:

```
43-var-list-children --all-values var_this_256
44-break-delete
Could not delete all breakpoints. TypeError: Cannot read property 'stdin' of null
45-target-detach
```

haneefdm, 2022-03-08:
From the output that started at `29-interpreter-exec console "load"` to command `41`, I don't see anything wrong except for bad watch variable evals. The request for local variables resulted in `this` with a value `0x2000c7e4` and no other locals. Does the value `0x2000c7e4` look right?

Then I see `42-var-update` on gdb variable  `var_this_256` that mirrors your variable `this`, but this variable has not been created/assigned. This is normal. It should have failed but it succeeded (`-> 42^done`) which is rather strange. We let it fail and then we create a gdb-variable. If it succeded then the variable was already created or exists somehow. But when we try to expand `43-var-list-children --all-values var_this_256`, it appears gdb has crashed

> Could not delete all breakpoints. TypeError: Cannot read property 'stdin' of null

That means gdb has crashed at this point. So the `stdin` to gdb has been (unexpectedly) closed. It appears that the `43-var-list-children --all-values var_this_256` caused gdb to crash. Notice no result from command `43`.

This is a different problem than the problem of having `'<unknown>'` values. Could you try a newer/different version of gdb? What version are you using?

AndrewCapon, 2022-03-08:
Gdb is: `GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git`

Here is the exact same breakpoint when it is working (first run):

```
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 -d "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf"
1-gdb-set target-async on
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" -f /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32h7x.cfg -c init -c "reset init" -c "gdb_breakpoint_override hard"
    Please check TERMINAL tab (gdb-server) for output from openocd
-> undefined=thread-group-added,id="i1"
-> ~"Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...\n"
Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...
-> 1^done
2-gdb-version
Reading symbols from objdump: Time: 27 ms
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Reading symbols from nm: Time: 52 ms
5-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/gdbsupport.init"
6-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/gdb-swo.init"
7-target-select extended-remote localhost:50000
8-interpreter-exec console "monitor reset halt"
9-target-download
10-interpreter-exec console "monitor reset halt"
11-enable-pretty-printing
12-interpreter-exec console "monitor arm semihosting enable"
-> 5^done
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1544\n"
Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1544
-> ~"1544\t\tfor (pSource = &_sidata, pDest = &_sdata; pDest != &_edata; pSource++, pDest++)\n"
1544		for (pSource = &_sidata, pDest = &_sdata; pDest != &_edata; pSource++, pDest++)
-> *stopped,frame={addr="0x24000894",func="Reset_Handler",args=[],file="../libdaisy//core/startup_stm32h750xx.c",fullname="/Volumes/Crucial X8 2TB/Development/daisy/libdaisy/core/startup_stm32h750xx.c",line="1544",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 7^connected
-> @"target halted due to debug-request, current mode: Thread \n"
target halted due to debug-request, current mode: Thread
-> @"xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000\n"
xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000
-> 8^done
-> 9+download,{section=".text",section-size="164944",total-size="2438981"}
-> 9+download,{section=".text",section-sent="16160",section-size="164944",total-sent="16160",total-size="2438981"}
-> 9+download,{section=".text",section-sent="164944",section-size="164944",total-sent="164944",total-size="2438981"}
-> 9+download,{section=".exidx",section-size="8",total-size="2438981"}
-> 9+download,{section=".init_array",section-size="44",total-size="2438981"}
-> 9+download,{section=".fini_array",section-size="4",total-size="2438981"}
-> 9+download,{section=".data",section-size="3532",total-size="2438981"}
[... 33 lines cut ...]
-> *stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c7e4 <mainUI+32>"}],file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
18-thread-list-ids
-> 18^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
19-thread-info 1
-> 19^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c7e4 <mainUI+32>"}],file="KnobParameter.cpp",fullname="/Volu [... cut ...]
watch: -var-create: unable to create variable object (from var-create watch_2527785aec91c116b56d99c12d975602dda5f351735d1cf7bf2b0323ae0f173b @ "m_modSlotsStorage[0]")26-stack-list-variables --thread 1 --frame 0 --simple-values
-> 26^done,variables=[{name="this",arg="1",type="KnobParameter * const",value="0x2000c7e4 <mainUI+32>"}]
27-var-update --thread 1 --frame 0 --all-values var_this_256
-> 27^error,msg="Variable object not found"
28-var-create var_this_256 * "this"
-> 28^done,name="var_this_256",numchild="1",value="0x2000c7e4 <mainUI+32>",type="KnobParameter * const",thread-id="1",has_more="0"
```

AndrewCapon, 2022-03-08:
And then a continue to hit the breakpoint again (local window now not working):

```
29-exec-continue --thread 1
-> 29^running
-> *running,thread-id="all"
mi2.status = running
-> @"halted: PC: 0x24006d70\n"
halted: PC: 0x24006d70
-> =breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x24006d6e",func="KnobParameter::Process()",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",thread-groups=["i1"],times="2",original-location="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp:92"}
-> ~"\n"

-> ~"Breakpoint 1, KnobParameter::Process (this=0x2000c810 <mainUI+76>) at KnobParameter.cpp:92\n"
Breakpoint 1, KnobParameter::Process (this=0x2000c810 <mainUI+76>) at KnobParameter.cpp:92
-> ~"92\t\treturn m_bChanged;\n"
92		return m_bChanged;
-> *stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c810 <mainUI+76>"}],file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
30-thread-list-ids
-> 30^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
31-thread-info 1
-> 31^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x24006d6e",func="KnobParameter::Process",args=[{name="this",value="0x2000c810 <mainUI+76>"}],file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},state="stopped"}]
32-data-list-register-values N
33-stack-info-depth --thread 1 10000
-> 32^done,register-values=[{number="0",value="536922128"},{number="1",value="0"},{number="2",value="0"},{number="3",value="536922128"},{number="4",value="536922128"},{number="5",value="604148308"},{number="6",value="0"},{number="7",value="537001544"},{number="8",value="0"},{number="9",value="0"},{number="10",value="0"},{number="11",value="0"},{number="12",value="-2139095041"},{number="13",value="0x2001fe48"},{number="14",value="604007625"},{number="15",value="0x24006d6e <KnobParameter::Process()+438>"},{number="25",value="1627389952"},{number="58",value="1000"},{number="59",value="0.0062893564615001119"},{number="60",value="211106232532992"},{number="61",value="-0.0029657153328876973"},{number="62",value="8.7954674357251852e-06"},{number="63",value="0.69314718069162029"},{number="64",value="1.0291255546633603e-314"},{number="65",value="0"},{number="66",value="0"},{number="67",value="0"},{number="68",value="0"},{number="69",value="0"},{number="70",value="0"},{number="71",value="0"},{number="72",value="0"},{number="73",value="-nan(0xfffff00000000)"},{number="90",value="1610612752"},{number="91",value="0x2001fe48"},{number="92",value="0x0"},{number="93",value="0"},{number="94",value="0"},{number="95",value="0"},{number="96",value="4"},{number="97",value="0"},{number="98",value="4.4765625"},{number="99",value="2213.75781"},{number="100",value="0.975629687"},{number="101",value="0"},{number="102",value="116"},{number="103",value="21741568"},{number="104",value="-0.907402873"},{number="105",value="-1.34229782e+14"},{number="106",value="0.44227621"},{number="107",value="-1.86522028e+38"},{number="108",value="1.79828668"},{number="109",value="3.48196239e+36"},{number="110",value="0"},{number="111",value="0"},{number="112",value="0"},{number="113",value="0"},{number="114",value="0"},{number="115",value="0"},{number="116",value="0"},{number="117",value="0"},{number="118",value="0"},{number="119",value="0"},{number="120",value="0"},{number="121",value="0"},{number="122",value="0"},{number="123",value="0"},{number="124",value="0"},{number="125",value="0"},{number="126",value="0"},{number="127",value="0"},{number="128",value="-nan(0x7fffff)"}]
-> 33^done,depth="7"
34-stack-list-frames --thread 1 0 6
-> 34^done,stack=[frame={level="0",addr="0x24006d6e",func="KnobParameter::Process",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="92",arch="armv7e-m"},frame={level="1",addr="0x24006bae",func="KnobParameter::Init",file="KnobParameter.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/KnobParameter.cpp",line="18",arch="armv7e-m"},frame={level="2",addr="0x2400692a",func="UIPage::InitKnob",file="UIPage.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/UIPage.cpp",line="16",arch="armv7e-m"},frame={level="3",addr="0x2400695e",func="UIPage::InitKnobDefaultRenderer",file="UIPage.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/UIPage.cpp",line="21",arch="armv7e-m"},frame={level="4",addr="0x24005b84",func="OscUI::Init",file="OscUI.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/OscUI.cpp",line="17",arch="armv7e-m"},frame={level="5",addr="0x240075f2",func="MainUI::Init",file="MainUI.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/MainUI.cpp",line="14",arch="armv7e-m"},frame={level="6",addr="0x240031f4",func="main",file="SimpleFMSynth.cpp",fullname="/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/SimpleFMSynth.cpp",line="86",arch="armv7e-m"}]
35-var-update --thread 1 --frame 0 --all-values watch_2527785aec91c116b56d99c12d975602dda5f351735d1cf7bf2b0323ae0f173b
-> 35^error,msg="Variable object not found"
[... cut ...]
-> 36^error,msg="-var-create: unable to create variable object"
watch: -var-create: unable to create variable object (from var-create watch_2527785aec91c116b56d99c12d975602dda5f351735d1cf7bf2b0323ae0f173b @ "m_modSlotsStorage[0]")37-stack-list-variables --thread 1 --frame 0 --simple-values
-> 37^done,variables=[{name="this",arg="1",type="KnobParameter * const",value="0x2000c810 <mainUI+76>"}]
38-var-update --thread 1 --frame 0 --all-values var_this_256
-> 38^done,changelist=[{name="var_this_256",in_scope="false",type_changed="false",has_more="0"}]
```

haneefdm, 2022-03-08:
Now that I see the entire session, I see that `28-var-create var_this_256 * "this"` was done already and rest of the session also makes sense. Now we are down to why gdb is crashing. Time to try a new/different version of gdb for you and me trying to use the same version as what you have -- 10.3-2021.10

From the last post of yours, I see commands `29-38`, and nothing seems amiss. We responded to VSCode with a new list of threads, frames, and variables just like the first time. What do you see in the Locals Window (you said "local window now not working" but I can't tell what that means. A screengrab of the area perhaps?

Btw, you did a "load" between the start and the restart? I don't think that came from the extension. May I ask why?

AndrewCapon, 2022-03-09:
Hi there,

For the start/restart load this is how the build system for this Daisy board is working, it does a build when you restart.

The problem is there without this though, basically for me the locals windows works only once, after hitting the first breakpoint. From that point on it doesn't work. `<unknown>'. This is without a restart, just using continue and hitting the breakpoint again.

I will try a different gdb today...

Here are two screen shots, this is just starting the debugger and hitting the breakpoint, then the second image is after continuing and hitting the breakpoint again:

1. Initial run debugger, hit breakpoint:

[image]

3. Continue and hit breakpoint again:

[image]

Now the complete log for the same thing but without the globals being open to make it easier to read. Also at the point that I am getting `<unknown>` I did a `p *this` to show gdb is still fine and working:

This is basically how I have been working with this, just displaying the variables using the `p` command, as long as I do this everything works and I can debug as normal. As long as I do not try to expand the `unknown` variables in the variables windows, if I do this is when it bombs out.

```
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 -d "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf"
1-gdb-set target-async on
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" -f /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32h7x.cfg -c init -c "reset init" -c "gdb_breakpoint_override hard"
    Please check TERMINAL tab (gdb-server) for output from openocd
-> undefined=thread-group-added,id="i1"
-> ~"Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...\n"
Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...
-> 1^done
2-gdb-version
Reading symbols from objdump: Time: 27 ms
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Reading symbols from nm: Time: 52 ms
5-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/gdbsupport.init"
6-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.4/support/gdb-swo.init"
7-target-select extended-remote localhost:50000
8-interpreter-exec console "monitor reset halt"
9-target-download
10-interpreter-exec console "monitor reset halt"
11-enable-pretty-printing
12-interpreter-exec console "monitor arm semihosting enable"
-> 5^done
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1544\n"
Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1544
[... 90 lines cut ...]
30-data-list-register-values N
31-stack-info-depth --thread 1 10000
-> 30^done,register-values=[{number="0",value="536922128"},{number="1",value="1132751509"},{number="2",value="8"},{number="3",value="0"},{number="4",value="536922128"},{number="5",value="604148308"},{number="6",value="0"},{number="7",value="537001544"},{number="8",value="0"},{number="9",value="0"},{number="10",value="0"},{number="11",value="0"},{number="12",value="-2139095041"},{number="13",value="0x2001fe48"},{number="14",value="604007625"},{number="15",value="0x24006d6e <KnobParameter::Process()+438>"},{number="25",value="-2130706432"},{number="58",value="1000.00011926651"},{number="59",value="0.0062893564615001119"},{number="60",value="211106232532992"},{number="61",value="-0.0029657153328876973"},{number="62",value="8.7954674357251852e-06"},{number="63",value="0.69314718069162029"},{number="64",value="1.7642974910131863e-07"},{number="65",value="5.1831404979823317e-315"},{number="66",value="0"},{number="67",value="0"},{number="68",value="0"},{number=" [... cut ...]
$1 = {m_pKnob = 0x20001734 <hw+1708>, m_fMin = 0, m_fMax = 1, m_fRange = 1, m_fValue = 0, m_fKnobValue = 0.264999986, m_uPresision = 3, m_bChanged = false, m_sName = 0x2401eeec " PW ", m_bEnabled = true, m_passthruDirection = KnobParameter::ptLower, m_fDistance = -0.264999986}
-> 37^done
{"token":37,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
38-stack-list-variables --thread 1 --frame 0 --simple-values
-> 38^done,variables=[{name="this",arg="1",type="KnobParameter * const",value="0x2000c810 <mainUI+76>"}]
39-var-update --thread 1 --frame 0 --all-values var_this_256
-> 39^done,changelist=[]
```

AndrewCapon, 2022-03-09:
I tried `GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git` exactly the same problem.

with `GNU gdb (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.90.20220202-git` the breakpoint is never hit!

```
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 -d "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf"
1-gdb-set target-async on
-> undefined=thread-group-added,id="i1"
-> ~"Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...\n"
Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...
-> ~"Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.\n"
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
-> ~"Use 'set mi-async'.\n\n"
Use 'set mi-async'.
-> 1^done
2-gdb-version
-> ~"GNU gdb (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.90.20220202-git\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin19.6.0 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://bugs.linaro.org/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Reading symbols from '/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf'
5-interpreter-exec console "info sources"
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" -f /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.2.2/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32h7x.cfg -c init -c "reset init" -c "gdb_breakpoint_override hard"
Please check TERMINAL tab (gdb-server) for output from openocd
-> ~"/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf:\n"
-> ~"(Full debug information has not yet been read for this file.)\n"
-> ~"\n"
-> ~"/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/_exit.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/write.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/sbrk.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/read.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/lseek.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/kill.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/isatty.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/getpid.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/fstat.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/libgloss/libnosys/close.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/libgcc/config/arm/lib1funcs.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/obj/gcc2/arm-none-eabi/thumb/v7e-m+dp/hard/libgcc/lib1funcs.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/libgcc/libgcc2.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/libgcc/config/arm/bpabi.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/obj/gcc2/arm-none-eabi/thumb/v7e-m+dp/hard/libgcc/bpabi.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/libgcc/config/arm/ieee754-df.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/obj/gcc2/arm-none-eabi/thumb/v7e-m+dp/hard/libgcc/ieee754-df.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/stdlib/nano-mallocr.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/isattyr.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/fstatr.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/readr.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/string/memmove.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/machine/arm/memchr.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/obj/newlib-nano/arm-none-eabi/thumb/v7e-m+dp/hard/newlib/libc/machine/arm/memchr.S, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/stdio/makebuf.c, /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-c [... cut ...]
28-thread-list-ids
-> 28^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
29-thread-info 1
-> 29^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x24000610",func="Reset_Handler",args=[],file="../libdaisy//core/startup_stm32h750xx.c",fullname="/Volumes/Crucial X8 2TB/Development/daisy/libdaisy/core/startup_stm32h750xx.c",line="1544",arch="armv7e-m"},state="stopped"}]
30-exec-continue
-> 30^running
-> *running,thread-id="all"
mi2.status = running
```

AndrewCapon, 2022-03-09:
I have just tested 11.2 from the console with my openocd and that doesn't work there either, so nothing to do with cortex-debug.

haneefdm, 2022-03-09:
Ummmm. I am confused as to what works and what doesn't. Note that we are stuck with `--interpreter=mi2`. Can't go backward or forwards (mi3) because gdb *guarantees* to break compatibility and many of our users are still stuck with older versions.

I do see a problem in the traces you supplied and it may not be cortex-debug but a serious issue regardless
```
34-var-update --thread 1 --frame 0 --all-values var_this_256
-> 34^done,changelist=[{name="var_this_256",in_scope="false",type_changed="false",has_more="0"}]
```
What the above says (due to `in_scope="false"`) that *The variable object does not currently hold a valid value but it may hold one in the future if its associated expression comes back into scope.*. In this case the associated expression is *`this`*. But it knows the value changed (because it is in the change list) but does not supply a new value (although I can see it when variables were listed). The PC is probably mid-scope change. This I think is a lie from gdb. If the variable `this` had a value at the same spot once before, it should have it again given the frame addresses are still same.

I have a couple of experiments queued up but since I cannot see the problem, I cannot test it either.

AndrewCapon, 2022-03-09:
Thanks.

Concerning what does work and what doesn't work here:

The first time you hit a breakpoint the locals window works.

When you continue and hit another breakpoint the locals window stops working displaying `<unknown>`

So basically it only works once.

I have tested with 10.2 and 10.3.

I cannot test with 11.2.

AndrewCapon, 2022-03-09:
I did think it may be something to do with hardware breakpoints in ram for a while but I think that is a red herring.

I have tested with 10.3, openocd and eclipse and everything is working fine there.

haneefdm, 2022-03-09:
Yes, it is a problem perhaps in the way cortex-debug is using gdb, but I can tell that gdb is definitely lying to us. Supposed to work and does work 99% of the time. Or else we would have never made a release. Not sure what is unique about your memory layout or elf file.

Since we cannot fix gdb, we have to find a workaround.  This won't be the first time. Again, since I can't duplicate it, I can take a shot in the dark and have you try out a fix...if you are up to it. Testing to make sure I did not break something else.

AndrewCapon, 2022-03-09:
Thanks very much, I would be happy to test.

Currently I have code executing from sram, I will do a little test to see if that may be the issue, will be different code as this stuff won't fit in the flash...

I can build cortex-debug here as well and maybe try to debug that, I would be stumbling around in the dark though :)

AndrewCapon, 2022-03-09:
Ok, getting the same issue with flash.

So I set it all up on a linux VM.

Using `GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git` and `xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:30)`

Everything works absolutely fine, no issues at all.

The OpenOCD I am using on the Mac is a different build `0.11.0+dev-00562-g5ab74bde0 (2022-02-13-15:49)`

I'll see if I can get that xPack one running on the Mac...

AndrewCapon, 2022-03-09:
Right, it seems it was the OpenOCD causing the problems. 

Using `xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-19:33)` on the Mac and everything works fine unless I do a Restart and then I still get the problem, I can live with that though.

So I'm guessing if you want to test the issue then use `0.11.0+dev-00562-g5ab74bde0 (2022-02-13-15:49)`, this was installed by brew.

haneefdm, 2022-03-09:
Ohhh. With OpenOCD, I don't trust anything that appears with homebrew or xPack. I use two different chip vendors and both ship their own versions of OpenOCD. The ST version is pretty close to what OpenOCD releases, but more recent device/board support, tested, and supported. These custom versions contain changes that are not available in any public version. For other chips, I build from the HEAD.

But other than getting the registers and reading memory, I am not sure how a version of OpenOCD affects this. OpenOCD knows nothing of your executable, how to interpret the stack, etc. It is all done by gdb using the PC/SP/LR registers. OpenOCD does handle breakpoints though.

AndrewCapon, 2022-03-09:
Hi,

I changed it to use the version from STM32CubeIde and that also works perfectly.

Concerning the issue still being there with a restart, I changed the Daisy supplied launch.json to just do a build rather than a build all and now everything works on a restart unless I change some code and then restart. Should this work:

```
      "preLaunchTask": "build",
      "preRestartCommands": [
        "load",
        "enable breakpoint",
        "monitor reset"
      ],
```

haneefdm, 2022-03-09:
Since we do a reset as well, did you try removing the `"monitor reset"`? Otherwise, there will be back-to-back resets. Also, your reset may not be a good one because that one will cause not just a reset but also a run which is super bad and it may hit un-intentional breakpoints, execute unwanted code, etc. Generally we don't use that at all in a debug session as we don't have any control over it and gdb is now confused because the program is running and it doesn't know about that.

Come to think of it, what you may need is the following
```JSON
      "preRestartCommands": [
        "monitor reset halt"   // May not need this but is safer. Remove if it is affecting the flow.
        "load"
      ],
```
the above will reset and halt at the reset handler and then perform a "load". Cortex-Debug will issue a "monitor reset halt" and from that point on, it is good to go.

AndrewCapon, 2022-03-10:
Hi,

Thanks for the info, I made these changes and still have `<unknown>` after a restart with code changes.

I'm not so bothered about this as everything is working correctly when starting from scratch now.

Thanks very much for all your help and time on this, it is really appreciated.

Cheers

Andy

haneefdm, 2022-03-10:
Could you please try out this release. It has to be installed manually using `Install from VSIX...` command

https://github.com/Marus/cortex-debug/releases/tag/v1.3.5-pre2

AndrewCapon, 2022-03-10:
Hi @haneefdm,

Tried it out, still the same problem with restart/build.

Also I have been having the same `<unknown>` problem in this and the pre-release version in code in IRQ handlers, or at least in the DMA one. Non IRQ code seems to work fine though, apart from the restart issue and some rare occurrences of <unknown>.

haneefdm, 2022-03-10:
I don't know about the IRQ thing because in ARM, IRQ's use a different stack pointer ut SP should still be the current SP (MSP vs PSP) based of SPSEL. Not up to me but up to OpenOCD and GDB on how they handle ISRs. Technically, ISRs should create a separate thread of execution. It does not surprise that is does not work (bugs, but most likely not ours).

Back to non-isr. Temporarily, if you totally remove the `preRestartCommands`, what happens?. Also what happens to if you single step after the breakpoint?

AndrewCapon, 2022-03-10:
Without the `preRestartCommands` everything is fine on restart, single stepping after breakpoint is always fine. Basically all the debugging has always worked perfectly apart from the Variables window.

haneefdm, 2022-03-10:
Ohhh... really. So, this is gdb doing things that I did not expect either. Makes no sense.

Can you send me the entire debug log -- from the start of the session to restart and then seeing <unknown>?

AndrewCapon, 2022-03-10:
so start debug, variables window working:

```
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 -d "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf"
1-gdb-set target-async on
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth" -s /Users/andrewcapon/Development/openocd/scripts -f /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.5-pre2/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32h7x.cfg -c init -c "reset init" -c "gdb_breakpoint_override hard"
    Please check TERMINAL tab (gdb-server) for output from openocd
-> =thread-group-added,id="i1"
-> ~"Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...\n"
Reading symbols from /Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf...
Finished reading symbols from objdump: Time: 31 ms
-> 1^done
Suppressing output for '2-gdb-version'
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Suppressing output for '5-symbol-info-variables'
Finished reading symbols from nm: Time: 57 ms
6-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.5-pre2/support/gdbsupport.init"
7-interpreter-exec console "source /Users/andrewcapon/.vscode/extensions/marus25.cortex-debug-1.3.5-pre2/support/gdb-swo.init"
8-target-select extended-remote localhost:50000
9-interpreter-exec console "monitor reset halt"
10-target-download
11-interpreter-exec console "monitor reset halt"
12-enable-pretty-printing
13-interpreter-exec console "monitor arm semihosting enable"
-> 6^done
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1542\n"
Reset_Handler () at ../libdaisy//core/startup_stm32h750xx.c:1542
-> ~"1542\t\tasm (\"ldr sp, =_estack\");\n"
1542		asm ("ldr sp, =_estack");
-> *stopped,frame={addr="0x24000894",func="Reset_Handler",args=[],file="../libdaisy//core/startup_stm32h750xx.c",fullname="/Volumes/Crucial X8 2TB/Development/daisy/libdaisy/core/startup_stm32h750xx.c",line="1542",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 8^connected
-> @"target halted due to debug-request, current mode: Thread \n"
target halted due to debug-request, current mode: Thread
-> @"xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000\n"
xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000
-> 9^done
-> 10+download,{section=".text",section-size="165360",total-size="2421022"}
-> 10+download,{section=".text",section-sent="16160",section-size="165360",total-sent="16160",total-size="2421022"}
-> 10+download,{section=".text",section-sent="165360",section-size="165360",total-sent="165360",total-size="2421022"}
-> 10+download,{section=".exidx",section-size="8",total-size="2421022"}
-> 10+download,{section=".init_array",section-size="40",total-size="2421022"}
-> 10+download,{section=".fini_array",section-size="4",total-size="2421022"}
-> 10+download,{section=".data",section-size="3532",total-size="2421022"}
-> 10+download,{section=".isr_vector",section-size="664",total-size="2421022"}
-> 10+download,{section=".isr_vector",section-sent="664",section-size="664",total-sent="169608",total-size="2421022"}
-> 10^done,address="0x24000894",load-size="169608",transfer-rate="906384",write-rate="10600"
-> @"target halted due to debug-request, current mode: Thread \n"
target halted due to debug-request, current mode: Thread
-> @"xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000\n"
xPSR: 0x01000000 pc: 0x24000894 msp: 0x20020000
-> 11^done
-> 12^done
-> @"semihosting is enabled\n"
semihosting is enabled
-> @"\n"

-> 13^done
[... 231 lines cut ...]

At this point getting `<unknown>` in variables window.

`p this` works fine though:

```
p this
55-stack-select-frame --thread 1 0
56-interpreter-exec console "p this"
-> 55^done
-> ~"$2 = (FMVoices * const) 0x2000a98c <fmVoices>\n"
$2 = (FMVoices * const) 0x2000a98c <fmVoices>
-> 56^done
{"output":"","token":56,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
57-stack-select-frame --thread 1 0
58-stack-list-variables --thread 1 --frame 0 --simple-values
-> 57^done
-> 58^done,variables=[{name="uVoice",type="uint_fast8_t",value="0"},{name="this",arg="1",type="FMVoices * const",value="0x2000a98c <fmVoices>"}]
59-var-update --thread 1 --frame 0 --all-values var_uVoice_256
-> 59^done,changelist=[{name="var_uVoice_256",in_scope="invalid",has_more="0"}]
60-var-update --thread 1 --frame 0 --all-values var_this_256
-> 60^done,changelist=[{name="var_this_256",in_scope="invalid",has_more="0"}]
```

Can continue debugging, hitting breakpoints and using `p` to display things and everything is fine.

haneefdm, 2022-03-10:
Head scratcher !?!?!? Can you try the following your launch.json? I know it is out of the left-field but I can explain if it works (seriously doubt it but worth a shot).

```JSON
           "postRestartCommands": [
                "monitor gdb_sync",
                "stepi",
            ],
```

AndrewCapon, 2022-03-11:
Still get the `<unknown>` with that in, only difference seems to be the debugger starts in the rest handler.

haneefdm, 2022-03-11:
All out of ideas. What if the breakpoint is slightly before or after?

I have a hack in mind but have to sleep on it.

AndrewCapon, 2022-03-11:
It doesn't seem to make any difference where the breakpoint is as far as I can see.

I have some more info though.

If I run the debugger and don't break at all. Then do a restart/build and add a breakpoint then the variable window is working.

So this problem only occurs if a breakpoint has been hit previously.

AndrewCapon, 2022-03-11:
And some more info:

If I run the debugger and pause, Then do a restart/build and add a breakpoint then the variable window is working.

If I have hit a breakpoint, then continue, then do a restart/build it is broken.

So it seems to be breakpoint related

haneefdm, 2022-03-11:
Having gotten some sleep, I am getting a better picture. Back to the first trace
```
29-interpreter-exec console "load"
-> ~"`/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf' has changed; re-reading symbols.\n"
`/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.elf' has changed; re-reading symbols.
```
First of all, this will NOT work in Cortex-Debug and can cause silent failures. If the executable changes, we don't re-read the symbols and re-create all the globals/statics. Chances are they are still the same but not guarnteed. GDB has re-read the symbols and our world view vs GDB can now be different.

When gdb reloads the symbols, we have no idea what it did with all the temporary gdb-variables we created. Sems like they are still there but may not be reliable. Use at your own risk.

This does not explain why it works after the first breakpoint but results cannot be trusted.

Bottom line, we do not expect your executable to have changed once the session starts. If we did, our Restart handling would have been different -- as in you would not have to add a `"load"` command yourself.

I know MS has changed the definition of a 'Restart' a couple of times and they didn't used to trigger a re-build on a 'Restart'. Now they do. They went back and forth on this. This is why we introduced the 'Reset' button and most people like this button more as we know exactly what it does and does not rely on MS changing their policy. As it stands, it is much cleaner to stop the session and start over. There are many things that are done at startup that aren't done on a Restart.

Btw, with openocd, you can use it to load a new image instead of using gdb. To do this, you need to convert your elf file to a hex file and use the command
```JSON
    "monitor program {/Volumes/Crucial X8 2TB/Development/daisy/SimpleFMSynth/build/SimpleFMSynth.hex}"
```
I don't recommend the above because if you rebuild. That reminds me. If you use and RTOS OpenOCD/JLink/etc will only detect an RTOS at the start. Never again. Another reason why Restart is bad.

I would like to close this issue and start new ones if needed. We kinda hijacked this issue which at this point is unrelated.

closed 2022-03-11 by haneefdm

AndrewCapon, 2022-03-11:
Thanks for all the help as well, very good of you.