Marus/cortex-debug #296: 'cortex peripherals' window doesn't show up for me

nihalpasham, 2020-05-23:
I've placed the official `svd file` in the root folder of my project.  My `launch.json` is as follows

``` json
"version": "0.2.0",
     "configurations": [
           {
             "type": "cortex-debug",
             "request": "attach",
             "servertype": "openocd",
             "cwd": "${workspaceRoot}",
             "executable": "./target/thumbv7em-none-eabihf/debug/test-boot",
             "preLaunchTask": "boot build",
             "name": "Debug attach (OpenOCD)",
             "device": "nRF52840_xxAA",
             "configFiles": [
                 "interface/cmsis-dap.cfg", 
                 "target/nrf52.cfg"
             ],
             "svdFile": "${workspaceRoot}/nrf52840.svd",
         },
```
However, the cortex-peripheral window does not show up. Am I missing something?

**Details**
MCU - nrf52840-mdk (From makerdiary)
Screenshot -
[image]
SVD file location screenshot -
[image]

haneefdm, 2020-05-23:
Hmmm, you should see the `PERIPHERALS` section even if there is no svd file. I notice that this is an 'attach` session type. Does this also happen for a `launch` session type?

Could you also give us the version info for VSCode using 'About Visual Studio Code'

nihalpasham, 2020-05-23:
Yes, its the same for a `launch` type session as well. 

[image]

**VsCode version info:**
```
Version: 1.45.1 (system setup)
Commit: 5763d909d5f12fe19f215cbfdd29a91c0fa9208a
Date: 2020-05-14T08:27:35.169Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18363
```
In addition to the above issue, I noticed the `variables` window tends to be a bit glitchy i.e. it sometimes populates `locals` with variable-information of a previous function call and in some cases, it doesn't load anything (gives me a constant `loading-type circular animation` ).

nihalpasham, 2020-05-28:
Hi @haneefdm - just checking if there's an update. Also, was wondering - could I be missing some setting in my installation?

haneefdm, 2020-05-28:
I will look at it over this weekend. In the meantime, can you try the following?

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre2

You have to download the vsix file and use VSCode to Install the extension. Do NOT double click on it.

nihalpasham, 2020-06-08:
I think I stumbled upon the answer (and its deceptively hilarious). 
1. Right click the 'cortex-registers' tab  and select the option to show cortex-peripherals. 
2. Screenshot attached.

[image]

haneefdm, 2020-06-08:
I wonder how the Peripherals Window even got turned off. I didn't even down about the list of windows you can show hide.

About the variables window, we are tracking this in #295 or it could be a separate isssue

closed 2020-06-08 by haneefdm