Marus/cortex-debug #297: Starting debugging fails with special characters in path

vmichal, 2020-05-23:
Using version 0.3.6 VS code is unable to initialize debugging session, most likely because of non ASCII char 'á' in username.
Output from the "Debug Console":

`Please check OUTPUT tab (Adapter Output) for output from openocd.exe
Launching server: "openocd.exe" "-c" "gdb_port 50000" "-s" "C:\Users\Já\source\repos\ams-sw\ams/build" "-f" "C:\Users\Já\source\repos\ams-sw\ams/fse07-ecua Debug.cfg" "-c" "tpiu config internal C:/Users/JA30B~1/AppData/Local/Temp/tmp-11856mz3ZL4O5oICZ uart off 72000000 500000"
Launching GDB: "arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
1-gdb-set target-async on
2-interpreter-exec console "source c:/Users/Já/.vscode/extensions/marus25.cortex-debug-0.3.6/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor reset halt"
5-target-download
6-interpreter-exec console "monitor reset halt"
7-enable-pretty-printing
8-interpreter-exec console "EnableITMAccess"
9-interpreter-exec console "BaseSWOSetup 143"
10-interpreter-exec console "SetITMId 1"
11-interpreter-exec console "ITMDWTTransferEnable"
12-interpreter-exec console "DisableITMPorts 0xFFFFFFFF"
13-interpreter-exec console "EnableITMPorts 0x1"
14-interpreter-exec console "EnableDWTSync"
15-interpreter-exec console "ITMSyncEnable"
16-interpreter-exec console "ITMGlobalEnable"
17-interpreter-exec console "DisablePCSample"
undefinedC:\Program Files (x86)\GNU Tools Arm Embedded\9 2019-q4-major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from C:\\Users\\J�\\source\\repos\\ams-sw\\ams/build/AMS.elf...\n"}]}
Reading symbols from C:\Users\J�\source\repos\ams-sw\ams/build/AMS.elf...
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"c:/Users/Já/.vscode/extensions/marus25.cortex-debug-0.3.4/support/gdbsupport.init: No such file or directory.\n"}]}
c:/Users/Já/.vscode/extensions/marus25.cortex-debug-0.3.6/support/gdbsupport.init: No such file or directory.
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","c:/Users/Já/.vscode/extensions/marus25.cortex-debug-0.3.6/support/gdbsupport.init: No such file or directory."]]}}`

haneefdm, 2020-05-23:
Unfortunately, I don't think there is a solution. gdb itself has a problem with such path names

You may have to specify DOS style 8.3 path-names. There are many GNU tools that fail and we can't fix those issues. Not even Microsoft has a solution for this.

https://github.com/Microsoft/vscode-cpptools/issues/1998

closed 2020-05-23 by haneefdm

vmichal, 2020-05-24:
Indeed you are right. I have just made several tests and - apparently - whole gdb does not cope well with non-ASCII characters. Thanks for clarification, @haneefdm!