Marus/cortex-debug #299: watch window symbols for address with multiple meanings

lidavid, 2020-06-03:
Hello,

I have a multistage application that starts with a bootloader and then jumps into an application.
To get all the symbols of both of the elf's i load the application symbols into GDB with a postLaunchCommand : `add-symbol-file application.elf 0x10000000` - this works great. My issue is that because this is a multi staged application the symbols in RAM have two meanings. for example  when i execute `info symbol 0x20000554` i get :
> symbol_A in section .bss of bootloader.elf
   symbol_b in section .bss of application.elf

My question is how is it decided what symbol to show in the WATCH window (when there is an address and the symbol is displayed in "<>") ? and if there is any way to select appropriate symbol ?

haneefdm, 2020-06-03:
Cortex-Debug can only display the symbols from one elf file and it is the one you specify in the `launch.json`. This applies to globals/statics.

For stack, though we use whatever gdb tells us at the time. Elf file is not used.

What goes into the Watch window (expressions) are evaluated in the context of the current stack by gdb and it may be local, static, or global. gdb decides. We simply display the result from gdb and sometimes, the result could be an error or nothing.

If this answers your question, please close the issue.

lidavid, 2020-06-03:
Yes, this answers my question. Thanks. 

Any idea how can i tell GDB to prefer one symbol over the other ?

haneefdm, 2020-06-03:
Nope, I don't know. You can always use a physical address with a typecast and you get exactly what you want, of course, the symbol may move when you recompile.

closed 2020-06-03 by lidavid