Marus/cortex-debug #300: Can't launch - works fine using arm-none-eabi-gdb and JLinkGDBServer manually

BarrettLowe, 2020-06-03:
When try to launch my app, nothing really happens. I get the little pop-up debugger window with buttons to pause and step etc but there is no jlink connection, no run to main, no anything else. I looked for logs but I'm not finding anything. Debug console is empty, the output windows look uninteresting...I have no clue where to look to diagnose this. 

I'm on windows and doing all of this manually using gdb and the gdb server works exactly as expected. My launch is below for reference but it doesn't feel like that's my problem.  I'm happy to get more information, just don't know where to look!
```
...
        {
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/cmakebuild/debug/P2/application/P2.elf",
            "name": "Debug P2 App",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "MIMXRT1064DVL6A",
            "interface": "swd",
            "runToMain": true,
            "showDevDebugOutput": false,
            "svdFile": "${workspaceRoot}/NXP_SDK/i.MXRT1064/MIMXRT1064.xml"
        },
...
```

haneefdm, 2020-06-03:
Enabling `showDevDebugOutput` may help. So, you see the "Debug Console" totally empty? There should at least be a couple of lines in there.

In the "Output" window, look for "Adapter Output" section where raw output from JLink server goes. Btw, did you specify the JLInk Gdb sever path? You can do this in user settings.

[image]

BarrettLowe, 2020-06-03:
When I enabled `showDevDebugOutput` I got this in the debug console: `0000000010: Reading symbols from 'C:\p4\blowe_UTB/cmakebuild/debug/P2/application/P2.elf'`

Then nothing else. There is no Adapter Output section. This is with all extensions disabled except for the cortex-debug extension.

BarrettLowe, 2020-06-03:
Tried to fix the slash direction because that looked suspect to me but no change.

haneefdm, 2020-06-03:
Ah, you are running into a bug that was recently addressed. Is this an IAR generated executable?

Can you try this beta release?

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre2

Please download the VSIX file. Install instructions:

https://stackoverflow.com/questions/42017617/how-to-install-vs-code-extension-manually

BarrettLowe, 2020-06-03:
Wha-bam! It works. Thanks! 

Do you have a link for the bug? I apparently didn't have the correct search terms.

closed 2020-06-03 by BarrettLowe

haneefdm, 2020-06-03:
I believe it is Issue #270