Marus/cortex-debug #305: Can't read SWO data with OpenOCD on a STM32F765NG

GrandChris, 2020-06-17:
On debugging on a STM32F765NG on Windows, OpenOCD throws a "Unable to open path: C:/Users/---UserName---/AppData/Local/Temp/tmp-147161rWrPfzywL5x - Unable to read SWO data." error message. 

### Cortex Debug configuration
```json
{
            "type": "cortex-debug",
            "request": "attach",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}/build",
            "executable": "ZEUS_STM32F765NG.elf",
            "name": "Debug (OpenOCD)",
            "device": "STM32F765NG",
            "svdFile": "../scripts/openOCD/STM32F7x7.svd",
            "showDevDebugOutput": true,
            "configFiles": [
                "../scripts/openOCD/stm32f7generic.cfg"
            ],
            "preLaunchTask": "Upload STM32F765NG if not already uploaded",
            "swoConfig": 
            {
                "enabled": true,
                "decoders": [
                    {
                        "type": "graph",
                        "port": 0,
                        "encoding": "unsigned",
                        "graphId": "1",
                        "scale": 1
                    },
                    {
                        "type": "graph",
                        "port": 1,
                        "encoding": "unsigned",
                        "graphId": "2",
                        "scale": 1
                    },
                    {
                        "type": "graph",
                        "port": 2,
                        "encoding": "unsigned",
                        "graphId": "3",
                        "scale": 1
                    },
                ],
                "cpuFrequency": 200000000,
                "swoFrequency":   2000000,
                "source": "probe"
            },
            "graphConfig": [
                {
                    "label": "CPU time",
                    "timespan": 30,
                    "type": "realtime",
                    "annotate": false,
                    "maximum": 105,
                    "minimum": 0,
                    "plots": [
                        {
                            "graphId": "1",
                            "label": "data 1",
                            "color": "#53753c"
                        },
                        {
                            "graphId": "2",
                            "label": "data 2",
                            "color": "#955f20"
                        },
                        {
                            "graphId": "3",
                            "label": "data 3",
[... 139 lines cut ...]
```
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.262304
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : Listening on port 50000 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000298 msp: 0x2002a098
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10016451
Info : flash size = 1024 kbytes
Info : Single Bank 1024 kiB STM32F76x/77x found
Info : flash size = 1024 bytes
```

haneefdm, 2020-06-17:
@GrandChris Could you do me a favor? Please remove the lines
```
init
reset init
```
at the end of the config file. I think this is too early because we have not yet finished configuring the `tpiu` if you notice the OpenOCD command-line

An `init` is done automatically by OpenOCD at the end of all configuration anyways. We may still do a smarter timeout but I want to see what happens.

haneefdm, 2020-06-18:
@GrandChris I also implemented the timeout. It is on my branch in my fork. Please give me feedback

https://github.com/haneefdm/cortex-debug/tree/external-swo

This change actually has more impact that might affect other gdb-servers like "external". See PR #276

And, I do want to know the results of the experiment I mentioned earlier https://github.com/Marus/cortex-debug/issues/305#issuecomment-645409363 to help others out who are using things a bit differently. It will be a big help for me.

GrandChris, 2020-06-18:
If I run it without 
"init"
"reset init" 
it seems to work. But if I look at the Adapter output, the requst 8000 kHz speed message is missing,
which means that the "reset-init" function (declared in "target/stm32f7x.cfg") is no longer be called.
Also my upload script, which also depends on this configure script now throws 50% of the time
"Error: corrupted fifo read pointer 0xffffffff".

I could probably work around the new issues but I think It should not be required because I think that creating the configure script would require a too deep understanding of how Cortex Debug works.

I may try your branch in the evening.

haneefdm, 2020-06-18:
Okay, thanks. Please let me know if the new timeout code works.

In case you have issues, I edited the same file you did, just different code and async, non-blocking timeout. I was unable to test it myself. I tested the code in a standalone app, outside VSCode.

Deltawolf, 2021-07-08:
Hi all,

I'm fairly new to all of this. I'm having a hard time deciphering if a fix was ever merged to master. I'm experiencing the same problem, and I'm not sure if this is the same root cause or something I've configured incorrectly (always a high probability).  Any update?

haneefdm, 2021-07-08:
Sorry, this code was never merged. I had no way of testing it and never got a response on if my changes worked or not. I needed to know if it works on Windows and one of Linux/MacOS

I think it is a good enhancement.

Deltawolf, 2021-07-08:
I think I was able to look at your branch and copy the file in. Before it was creating the temp file, but it was not able to open it. Now it is not creating the temp file. I'm double checking that I copied it correctly. I'm on Windows 10 x64.

Deltawolf, 2021-07-08:
We are free to roam the country.

Ok, I'm able to see output in this format in the text file. Nothing to output console. Not sure what to do from here. In VS Code this can be rendered as a "SOH" control character, and they are surrounding every character.

haneefdm, 2021-07-08:
Could you try writing something much longer (longer than 80 chars). VSCode will add a new-line every 80 chars.

I am assuming you are writing 8-bit characters in your FW. 16 or 32 bit writes will not be compatible for a console-decoder. The text file will also have decoding information btw, since SWO can have multiple ports and formats and they all come into the same file

haneefdm, 2021-07-08:
Also add some new-lines into your FW writes. File based stuff has all kinds of buffering going on on both reads and writes and I am wondering if that could be a problem.

Deltawolf, 2021-07-08:
All of my printf lines are ended with \n. Here are the functions in use. I have it encoding as ASCII now instead of leaving the encoding blank. The functions I'm using to write are listed below this paste dump

We are free to roam the country.
00:00:00:00:00:01
Board rev: 0
mqtt wait
StartNetworkTask
MAC Address Initialization: 00:00:00:00:00:01
     €     €     €     €     €     €     €     €     €     €MQTT task notified. Setting up initial connection and connecting...
Set to subscribe to topic: Configuration/Assignment/Set/00:00:00:00:00:01
Set to subscribe to topic: Configuration/Time/Timezone
Set to subscribe to topic: Device/Firmware/
Set to subscribe to topic: Device/Firmware/v1/#
MQTT DNS Request started....
MQTT DNS Request returned!
MQTT DNS had no answer for 'broker.nc.X.net'. Setting default broker IP of XX.XXX.X.XX
Generated Client ID: 00:80:E1:00:00:01_<00000-00000>
Connecting to MQTT broker at 'XX.XXX.X.XX'...
mqtt_connect return -4
mqtt_connect return -4
MQTT Connection Status: 0

'int _write(int32_t file, uint8_t *ptr, int32_t len)
{
    
    for(int i = 0; i<len; i++)
        ITM_SendChar((*ptr++)); 
    return len;
}

__STATIC_INLINE uint32_t ITM_SendChar (uint32_t ch)
{
  if (((ITM->TCR & ITM_TCR_ITMENA_Msk) != 0UL) &&      /* ITM enabled */
      ((ITM->TER & 1UL               ) != 0UL)   )     /* ITM Port #0 enabled */
  {
    while (ITM->PORT[0U].u32 == 0UL)
    {
      __NOP();
    }
    ITM->PORT[0U].u8 = (uint8_t)ch;
  }
  return (ch);
}

'

haneefdm, 2021-07-08:
Okay, this may be a silly question. Where are you looking for the output from SWO. It should be in the `Output` Tab and the drop down should have something like `SWO: ITM [port: 0, type: console]`

If this is working, then lets roll back and print small strings. We are polling the SWO file aggressively (perhaps too aggressively) every 2ms.

haneefdm, 2021-07-08:
Even if that `Output` tab is empty, it should be in the dropdown list along with `Adapter Output` and other types of output

Deltawolf, 2021-07-08:
It's not silly, it's easy to miss, but I am looking at that output tab and
not seeing anything. I'm only able to open the temp file after ending the
debug.

Deltawolf, 2021-07-08:
It is being added to the list correctly. No information is being piped in
though (or however it is being handled on the back).

haneefdm, 2021-07-08:
>> I'm only able to open the temp file after ending the debug.

If you can't see it, neither can we :-) Do you see/list at least the file is created before quitting the debug session? Looks like openocd is not flushing the output buffers after writes and they are force flushed by the C-runtime on exit. The default buffer size is typically large (0.5-4KB)

Deltawolf, 2021-07-08:
Ok, good to know. Now we are getting somewhere. Yes, the fix you posted
before allowed me to create and find the file at launch. It's just listed
as 0kb until stop

haneefdm, 2021-07-08:
if you do a `dir` or `ls -l` on that file, do you see a non-zero size before you quit the session?

Deltawolf, 2021-07-08:
Still 0 size using dir.

Deltawolf, 2021-07-08:
It looks like opening the file will force it to update if I open it in Windows, but not before then. It still won't show in the output tab if I open and close it.

haneefdm, 2021-07-08:
We already have the file open and constantly reading it. All we get is zero bytes. You reading the file should also do the same thing. The output will actually be buffered inside (user space) in openocd and it has no idea someone is trying to read it.

How did you/we configure the TPIU. If we did it you should see it in the Debug Console as args to openocd

Deltawolf, 2021-07-08:
Launching server: "openocd.exe" "-c" "gdb_port 50000" "-s" "C:\Repos\XX\Devices\" "-f" "C:\Repos\XX\Devices\ocd_debug.cfg" "-c" "tpiu config internal C:/Users/XXXXXX~1/AppData/Local/Temp/tmp-93325jzdDyNNdDn8 uart off 64000000 2000000"
Launching GDB: "C:\Repos\arm_gcc\2020-q4-major\bin\arm-none-eabi-gdb-py.exe" "-q" "--interpreter=mi2"
.
.
.
17-interpreter-exec console "EnableITMAccess"
18-interpreter-exec console "BaseSWOSetup 31"
19-interpreter-exec console "SetITMId 1"
20-interpreter-exec console "ITMDWTTransferEnable"
21-interpreter-exec console "DisableITMPorts 0xFFFFFFFF"
22-interpreter-exec console "EnableITMPorts 0x1"
23-interpreter-exec console "EnableDWTSync"
24-interpreter-exec console "ITMSyncEnable"
25-interpreter-exec console "ITMGlobalEnable"
26-interpreter-exec console "DisablePCSample"

haneefdm, 2021-07-08:
I don't think file-based IO works well on Windows at all because of all of these buffering issues. Other platforms may also have an issue. Originally, this change was for using openocd as an "external" server as in you start the server yourself and not by this extension.

I think we can do a tcp based IO... I am currently stopped because I cannot build the extension. MS is updating vscode and they are half done. Can't compile anything.

Deltawolf, 2021-07-08:
I'm pretty new, but is there any keywords I can look up or things I can do in the interim? I was using semi-hosting for debug prior, but once I started messing with interrupts in FreeRTOS I got blocked and had to switch to ITM. I'm using st-linkv2 and trying to find a way to pipe ITM into any of the consoles.

Deltawolf, 2021-07-08:
When I was manually setting this up earlier, I can open a port for telnet (not sure what other connections are possible), but I get the control characters and I have to have an external powershell open.

haneefdm, 2021-07-08:
It took me 3+ hours and now I can compile the extension again.

Would you be willing to test my changes in Windows? I can do the best I can for MacOS. This change will use tcp based IO for SWO rather than file(s).

It may take a couple of hours

Deltawolf, 2021-07-08:
Yeah, I'm down to check it out. I have to go home in about an hour or so, but I can check it first thing in the morning if you are still working on it.

Deltawolf, 2021-07-08:
Would failing this test impact the issue we are working on given it is still somewhat functioning?

.vscode\extensions\cortex-debug-master\binary_modules\v14.16.0\win32\x64\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Deltawolf, 2021-07-08:
Ok, I'm heading home. I got the serial port updated and it is passing the tests now. I was stepping through the file.ts, and the program control flow seemed odd, I never fell into the assignment block shown below. I'm not sure if its how the fs.open is setup that the debugger won't allow you to break-point in there or what was happening. Instead I was only able to step from fs.open() to  setTimeout(). Just wanted to confirm that this piece was working as intended since it was something we changed. Good luck tonight and I'll chat in the morning
'else {
                this.fd = fd;
                this.interval = setInterval(this.read.bind(this), 2);
                this.connected = true;
                this.emit('connected');
            }'

haneefdm, 2021-07-09:
I pushed the changes to my branch. You can try it out. No more file-based IO for OpenOCD. Switched to tcp in OpenOCD since it supports that now.

Even this has to change. Cortex-Debug is using a deprecated tcl command to set the TPIU. The TPIU setting is now per processor/DAP based which matters when you have more than one processor.

I tested my changes on a Mac. There is now less dependency on the OS now that the file/fifo usage is gone.

PS: The reason you did not see the single-step work as you expected is that a LOT of Javascript is asynchronous If you put a breakpoint in the else-part and let it continue, it would have stopped where you expected eventually. So, ya debugging is hard until you get used to it.

Deltawolf, 2021-07-09:
Do I need to change my debug configuration at all?  Running on probe is giving me the same results as yesterday so far.

haneefdm, 2021-07-09:
Nope. you shouldn't have to. Currently the `"source"` can be `"probe"` or `"socket"` or `"file"`, but not `"serial"` because that invokes a serial port communication channel.
```
        {
            "cwd": "${workspaceRoot}",
            "executable": "./Debug/firststm.elf",
            "name": "Debug OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32F429ZI",
            "configFiles": [
                "firststm Debug OpenOCD.cfg"
            ],
            "swoConfig": {
                "cpuFrequency": 168000000,
                "swoFrequency":   1500000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "port": 0,
                        "label": "hdm",
                        "showOnStartup": true
                    }                    
                ],
                // "profile": true,
                "enabled": true
            },
            "runToMain": true
        },
```

Deltawolf, 2021-07-09:
I copied your swoConfig section to be extra sure and just changed by frequencies. Same results as yesterday. Temp file at 0kb, doesn't write until opened or debugger stopped. Nothing in the console, but it showOnStartup is pulling it up automatically now which is nice.

haneefdm, 2021-07-09:
Oops, I gave you the production `launch.json`. Here is the debug one which will use the debug version of the debug adapter. Notice the `"debugServer": 4711` line. Note that tmp files are no longer used for OpenOCD. Check the Debug Console you will see a very different command-line to OpenOCD from before
```JSON
        {
            "cwd": "${workspaceRoot}",
            "executable": "./Debug/firststm.elf",
            "name": "Debug OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32F429ZI",
            "debugServer": 4711,
            "configFiles": [
                "firststm Debug OpenOCD.cfg"
            ],
            "swoConfig": {
                "cpuFrequency": 168000000,
                "swoFrequency":   1500000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "port": 0,
                        "label": "hdm",
                        "showOnStartup": true
                    }                    
                ],
                // "profile": true,
                "enabled": true
            },
            "runToMain": true
        },
```
```

Btw, if you want to post multi-line code- then you can do a

```
```JSON
      any text including
           new lines 
'``
```

I 3 backticks to start and end a block of code. First 3 first 3 backticks can have a language

Deltawolf, 2021-07-09:
I'm not getting the tpiu config commands on launch right now.

Launching server: "openocd.exe" "-c" "gdb_port 50000" "-s" "C:\Repos\xxx\Devices\" "-f" "C:\Repos\xxx\Devices\ocd_debug.cfg"

Deltawolf, 2021-07-09:
Ok, I tried it and got 

"connect ECONNREFUSED 127.0.0.1:4711"

haneefdm, 2021-07-09:
I am seeing `"-c" "gdb_port 50000"`. You should see something like. You will see a lot more stuff

```
Launching server: "openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "/Users/hdm/STM32CubeIDE/workspace_1.1.0/firststm" "-f" "firststm Debug OpenOCD.cfg" "-c" "tpiu config internal :50003 uart off 168000000 1500000"
```

haneefdm, 2021-07-09:
That means that the debug version of the adapter cannot start because there is another version running on port 4777. Quit all instances of VSCode. Use the task manager or process viewer and see if there is a process `node` running and if so, kill it.

Deltawolf, 2021-07-09:
Let me make sure I set it up correctly since I'm a noob. I have deleted the original plugin and added a new folder in .vscode/extensions containing the github source files. binary_modules is updated with the serial-port.sh through bash. Launch vs code and hit debug right?

I'm going to restart my computer because I do not see the process open and vs code has been restarted.

Deltawolf, 2021-07-09:
Same results after restart. Still getting the refused connection

haneefdm, 2021-07-09:
To debug without modifying the VSCode extensions, there is a better way (for me)

* git clone the coretex-debug repo and set to the right branch
* cd cortex-debug
* npm install
* Start VSCode and open this folder (since VSCode is in my path, I do "code ." on the command line)
* Using Command Pallette "Run Task...-> npm: watch". This will start a background compile process
* From the debugger, I use the "Extension + Debug Server" which will debug both the front-end and back-end at the same time. You can make changes to sources and they get compiled when you save.
* Above will start a new VSCode window where you load your FW project.
* If you make any changes to Cortex-Debug, you have to quit the FW VSCode window, terminate the debug adapter in the Cortex-Debug VSCode and restart debugging.

Yes, it seems like a LOT but could have been worse. To debug a Native C++ project I had to use two computers and the list of things to do (can go wrong) was two pages long.

Deltawolf, 2021-07-09:
I'll try your workflow after this. Thanks for letting me know and being so patient with me. Hopefully we'll get this up and running in a moment.

haneefdm, 2021-07-09:
Those steps are good to know, especially to debug (or help me out). Here is a VSIX file that you can download and install. It will replace your current extension and do not use the 4711 port in your launch.json because it will be like you are really using a released version of the extension

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.12-pre2

This is a painful way for me to release it because no good way to update/track with changes and you can't debug the extension with it.

Deltawolf, 2021-07-09:
Ok, I'll swap over. I'm getting this error on npm install
Error installing vscode.d.ts: Error: Request returned status code: 404

I was googling trying to figure out how to get around this

haneefdm, 2021-07-09:
This is what I spend more than 6 hours yesterday until I finally got it working. Even the example extension from Microsoft failed. I tried so many things and in the end, not sure what worked. But, if you already had node_modules dir under the cortex-debug, remove it and then re-run `npm install`.

Make sure your node version is up to date

To install from the VSIX file I posted above, see
https://stackoverflow.com/questions/42017617/how-to-install-vs-code-extension-manually

Deltawolf, 2021-07-09:
Launching server: "openocd.exe" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "C:\Repos\xxxx\Devices\" "-f" "C:\Repos\xxxx\Devices\ocd_debug.cfg" "-c" "tpiu config internal :50003 uart off 64000000 2000000"

console opens, but no text still.

Deltawolf, 2021-07-09:
No problem. I'll look through your changes and see if I can step through and find what is going on. Thanks for your help again.

Deltawolf, 2021-07-09:
Ok, getting closer. I figured out how to get the npm install to work. I saw a lot of problems around credentialing so I just forked your branch and it let me start working. I was missing the 'dist' folder to launch the debug server though so I had to copy that over manually, but now we are jamming.

With uncaught exceptions on, I was able to see this when my board starts into main.

Exception has occurred: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 4. Received 5
t boundsError (internal/buffer.js:83:9)
	at writeU_Int8 (internal/buffer.js:738:5)
	at Buffer.writeUInt8 (internal/buffer.js:745:10)
	at ITMDecoder.rxWriteByte (c:\Repos\cortex-debug\dist\extension.js:45985:23)
	at ITMDecoder.processByte (c:\Repos\cortex-debug\dist\extension.js:46049:26)
	at SWOCore.handleData (c:\Repos\cortex-debug\dist\extension.js:46199:29)
	at SocketSWOSource.emit (events.js:315:20)
	at Socket.<anonymous> (c:\Repos\cortex-debug\dist\extension.js:46807:51)
	at Socket.emit (events.js:315:20)
	at addChunk (internal/streams/readable.js:309:12)
	at readableAddChunk (internal/streams/readable.js:284:9)
	at Socket.push (internal/streams/readable.js:223:10)
	at TCP.onStreamRead (internal/stream_base_commons.js:188:23)
	at TCP.callbackTrampoline (internal/async_hooks.js:131:14)

This is coming from writeU_Int8 in internal/buffer.js
This is the buffer being passed in
buf = {
  "0": 208,
  "1": 128,
  "2": 208,
  "3": 192,
  "4": 128,
}

This is ITMDecoder in full
ITMDecoder ={
  _events: {
    "software-event": function () { [native code] },
    "hardware-event": function () { [native code] },
    synchronized: function () { [native code] },
    "lost-synchronization": function () { [native code] },
    timestamp: function () { [native code] },
    overflow: function () { [native code] },
  },
  _eventsCount: 6,
  _maxListeners: undefined,
  syncBuffer: {
    _elements: [
      208,
      192,
      128,
      208,
      32,
      128,
    ],
    _first: 5,
    _last: 0,
    _size: 6,
    _evictedCb: undefined,
    _end: 4,
  },
  status: 3,
  rxCount: 5,
  timestamp: 0,
  rxBuffer: {
    "0": 208,
    "1": 128,
    "2": 208,
    "3": 192,
    "4": 128,
  },
  rxTargetLength: 5,
  rxPacketType: 3,
  rxPort: -1,
}

haneefdm, 2021-07-09:
The `dist` folder contains the result of the TypeScript compilation. If it did not exist, then a compile did not happen. the task 'npm: watch' should have created this folder.

If you copied it from somewhere else, then it might not be right. You can manually compile the extension by opening a Terminal (I do it within VSCode) and doing `npm run-script compile`

Deltawolf, 2021-07-09:
Everything is working now. I lowered the swofrequency now that everything is connecting and it is no longer throwing the exception. It is weird because these settings worked in the SWV viewer through stm32cubeprogrammer, but adjusting them here was the final touch.

Deltawolf, 2021-07-09:
Ok that is good to know. I may not have ran it again when I had to restart earlier. I made the crucial mistake of making the cortex-debug repo in the extension folder, and vs code kept converting it. I got everything figured out when I actually put it in my repo folder.

haneefdm, 2021-07-09:
Something is bothering me. That exception is an SWO packet overlow and we are getting a 6th one. Of course, SWO is like a TX-UART so things can go wrong with frequencies.

That code has not changed in 3+ years.

Deltawolf, 2021-07-09:
I'd assume it is me making a dumb mistake before you fault the code for it lol. If you want, we can leave this open and I can make a calendar reminder to check back in on Wednesday to see if I've had any weirdness between now and then if that would be helpful.

haneefdm, 2021-07-09:
I will merge to the Master on @Marus repo. and call it done. I can close this issue as well once I do a PR the normal way.

I will chalk it up to a UART anamoly but it is good to know that it can happen.

Thanks for all your help. I think it made it better overall. I need more volunteers :-(

Deltawolf, 2021-07-09:
If I have the availability, I don't mind helping. I just never actually sign into Github. Is there a way you can ping me through email or like a Discord or something? I am have until the end of the summer to have access this setup and board, but after that, I won't be interning here anymore.

haneefdm, 2021-07-09:
@Deltawolf You can `Watch` this repo and you get email notifications. See the top right of this page.

I am currently working on supporting [SEGGER RTT](https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/) with OpenOCD and a new Memory Window. The new Memory Window is based on [MS Hex Editor](https://marketplace.visualstudio.com/items?itemName=ms-vscode.hexeditor) which I thought was pretty cool. I contributed a few changes there and I need to do one more PR for a bug fix.

I found (and fixed) a bug in OpenOCD-RTT as well so that is stalled until I do a PR which they don't accept in general. The way it is done in OpenOCD, it can work with any probe -- no need for JLink. I have better luck with Microsoft.

RTT is cool btw, it is super fast, no need to mess with frequencies Caveat, only works while you are debugging. RTT can be hooked up to printf

Another thread I got going is WSL2 support.

Deltawolf, 2021-07-09:
Ok cool. I will go look at the watch feature. I wish we had a Segger here, they look like that would make life easier than this st-link. A memory window would be a nice to have as well. I haven't checked out WSL2 yet. I set it up on my home pc, but I'm rarely there anymore so I haven't fiddled with it.

I'll try to dig around see how I can contribute here and there and hopefully I'll learn more while I'm at it. Thanks for your time yesterday and today! You've been awesome.

closed 2021-10-11 by haneefdm