Marus/cortex-debug #306: Truncated call stack

bugadani, 2020-06-19:
Hi!

I'm debugging a Rust application running on an nRF52 (cortex-m4-r0p1) using a j-link probe and j-link server.

Most of the time the debugging experience is fine. However, my callstack seems to be truncated to 20 or so elements, even when `backtrace` in debug console lists 51.

haneefdm, 2020-06-19:
Getting 20 at a time is the default in VSCode. There should be a UI element to get more.

Can you attach a screenshot if the call-stack window at the bottom?

bugadani, 2020-06-19:
[image]
[image]

Like this? Definitely nothing to get more

haneefdm, 2020-06-19:
VSCode initially had a hardcoded limit of 20 frames. After many complaints, looks like they fixed it. However, we were not aware they fixed it and we have to tell VSCode that "we can get more frames if it wants". You will now see the following in the Call Stack Window if there are more frames available

[image]

The way VSCode UI was designed, it will only load 20 additional frames. Not in our control.

I prepared you a preview release so you can verify with Rust. Especially for accuracy.

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre4
Download the VSIX file and Install the extension manually. Instructions below
https://stackoverflow.com/questions/42017617/how-to-install-vs-code-extension-manually

bugadani, 2020-06-19:
Wow that was fast, thank you! The preview works perfectly.

haneefdm, 2020-06-19:
I am hoping to make a release this weekend. Enjoy

closed 2020-06-19 by haneefdm