Marus/cortex-debug #308: Connect gdb to ip instead of localhost to debug from WSL2

carlolars, 2020-06-23:
Hi

I'm trying to debug from VSCode running remotely in WSL2 using OpenOCD, which needs to run on the Windows side.

The plugin successfully starts openocd on the windows side, I just added `"openOCDPreConfigLaunchCommands": [ "bindto 0.0.0.0" ]` so that it also listens to the WSL network interface and not only 127.0.0.1, and I have verified that it works by manually starting gdb in WSL2 and connecting to openocd started by the plugin using `<hostip>:50000`.

But how do I configure the plugin so that gdb connects to `<hostip>:50000` instead of `localhost:50000`?

haneefdm, 2020-06-23:
You can use whatever server you want with an external server.
```JSON
            "servertype": "external",
            "gdbTarget": "host:NNNN",
```
Use your own host (any hostname or ip-addr accessible from your device) and port$. Be sure to start the server first and then launch the debugger.

closed 2020-06-23 by haneefdm

carlolars, 2020-06-26:
Alright that works! Not what I'd hoped for since it introdues an additional step but I can live with it.

Here is how I made it work with VSCode running remotely in WSL2:

## Cortex-Debug in WSL2
1. Make sure that `openocd.exe` is found in WSL2 (`.../OpenOCD/0.10.0-13/bin` added to PATH) and that it is executable. 
2. Create a VSCode task that starts OpenOCD and configures it to listen to all interfaces (`"bindto 0.0.0.0"`), see _tasks.json_ below.
3. Create a VSCode launch configuration that connects to the OpenOCD gdb-server running in Windows-land, see _launch.json_ below.
    * The IP address in the `gdbTarget` (172.31.240.1) is the IP address of the Windows host. It can easily be found by in WSL2 do `cat /etc/reslov.conf` where the _nameserver_ IP is the IP of the Windows host. The IP will change when rebooting Windows!
4. Before starting a debug session make sure that the OpenOCD task is running. The task will stay until stopped or VSCode is closed so it usually only has to be started once.

### tasks.json
```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Launch OpenOCD",
            "command": "openocd.exe",
            "args": [
                "-f",
                "${input:openocd.config}",
                "-c",
                "\"bindto 0.0.0.0\""
            ],
            "type": "shell",
            "isBackground": true,
            "problemMatcher": []
        },
    ],
    "inputs": [
        {
            "id": "openocd.config",
            "type": "pickString",
            "description": "Select the configuration file for OpenOCD",
            "options": [
                "board/stm32f0discovery.cfg"
            ],
            "default": "board/stm32f0discovery.cfg"
        }
    ]
}
```

### launch.json
```
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "name": "(gdb-remote) Debug",
      "type": "cortex-debug",
      "request": "launch",
      "servertype": "external",
      "cwd": "${workspaceRoot}",
      "runToMain": true,
      "executable": "path/to/executable",
      "svdFile": "path/to/.svd",
      "gdbTarget": "172.31.240.1:3333",
    },
  ],
}
```

haneefdm, 2020-06-26:
Nice

RisinT96, 2020-06-26:
Maybe add this to the wiki?

haneefdm, 2020-06-26:
Yes, take a stab at it. I can review it later.

As part of OpenOCD startup, perhaps you also edit launch.json from a script?

carlolars, 2020-06-26:
> Also note that the windows host address changes on each **reboot/reset of the wsl2 instance**, it can be found at `/etc/resolv.conf`

I think that the Windows host keeps the same IP, since the Ethernet adapter "vEthernet (WSL)" on the Windows host is not changing. But the guest (WSL2) IP is changing every time it restarted. I tested with `wsl --shutdown` a few times just now.

RisinT96, 2020-06-26:
> I think that the Windows host keeps the same IP, since the Ethernet adapter "vEthernet (WSL)" on the Windows host is not changing. But the guest (WSL2) IP is changing every time it restarted. I [...]

My mistake then, I remember troubleshooting host access from my wsl2, and reading on one of the forums that the IP changes, maybe it happens on machine reboots.
Don't have access to that machine right now so I can't confirm.

carlolars, 2020-06-26:
> As part of OpenOCD startup, perhaps you also edit launch.json from a script?

Actually I have  
`"gdbTarget": "${env:WSL_HOST_IP}:3333"`  
in my _launch.json_, and in my _.bashrc_ I've added  
`export WSL_HOST_IP=$(cat /etc/resolv.conf | sed -rn 's|nameserver (.*)|\1|p')`

haneefdm, 2020-06-26:
There may be other solutions as well. Huge discussion here...

https://github.com/microsoft/WSL/issues/4210

carlolars, 2020-06-26:
> 
> 
> > As part of OpenOCD startup, perhaps you also edit launch.json from a script?
> 
> Actually I have
> `"gdbTarget": "${env:WSL_HOST_IP}:3333"`
> in my _launch.json_, and in my _.bashrc_ I've added
> `export WSL_HOST_IP=$(cat /etc/resolv.conf | sed -rn 's|nameserver (.*)|\1|p')`

I lied. I actually have `"debug.gdbTarget": "${env:WSL_HOST_IP}:3333"` in my _.vscode/settings.json_, and in the _launch.json_ I have `"gdbTarget": "${config:debug.gdbTarget}"`, because it seems like `${env:...}` doesn't work in my _launch.json_. Figured it had to be a reaons why I had it setup like that...

haneefdm, 2020-06-26:
Yes, the `env:` is looking for an environment var. in the Windows env.  So unless you start VSCode from the same bash that contains that variable, VSCode will not see it.

You can create a shortcut that  does the same, but you already have a good workaround

carlolars, 2020-06-26:
> Yes, the `env:` is looking for an environment var. in the Windows env. So unless you start VSCode from the same bash that contains that variable, VSCode will not see it.

Funny thing, it seems like I've managed to always start VSCode from within WSL2, where that env.var is set, when I've debugged during the last couple of days since it clearly doesn't and shouldn't work when starting VSCode from the shortcut 😅. So my "fix" to automate it is not a good one.

rwalton-256, 2021-09-20:
Not sure what has changed, when I tried this it looks like openocd server is actually running on WSL2. So I changed gdbTarget to :3333 and it works.

haneefdm, 2021-09-20:
@rwalton-256  I have no idea as I don't know your entire environment. But from our side, nothing changed in the last few weeks in this aspect.

Note that the gdbTarget is given to gdb -- we don't mess with it. ANd gdb simply connects to what you give -- at this point, it is up to your network and operating systems.