Marus/cortex-debug #309: Option to show variables in a different base

kisielk, 2020-06-23:
Often (actually in my case, usually) it's preferable to see variables in hex or binary. Currently it seems it's only possible to see them in decimal. Would be great if the popup menu provided an option to see them in another base.

haneefdm, 2020-06-23:
The UI for Variables Window belongs to VSCode. Not sure there is a way to customize it. Yeah, we can make a global setting for the format but not something we want to spend cycles on.

However, in the watch window, you can enter `expr,format` where `expr` can be any C expression and format can be one of `x b o, etc`, stuff supported by gdb

closed 2020-06-23 by haneefdm

chetrf, 2020-07-02:
If you use the memory viewer you can view hex bytes.  Not sure if this will help with your particular situation, but it was perfect form mine.  I am debugging a serial protocol and needed to see all the values in a buffer in hex.  You just use the "Cortex-Debug: View Memory" command from the command pallet.  You can then enter your variable name instead of an address, then enter how many bytes you want to see.
[image]

haneefdm, 2020-07-02:
Haha, I forgot about that. I added that feature to the memory window so you can use any C expression for an address. Works best with global variables though.

@Marus, creator of this extension did a great job with the presentation though. That was not me.

chetrf, 2020-07-02:
Any thoughts on the right way to type in an expression in the watch window to get hex for arrays?  The comma x syntax is working for individual variables but, I have not figured it out yet for arrays.
[image]

chetrf, 2020-07-03:
BTW, that view memory feature is pretty sweet.  I was able to open several windows and watch my data get passed around and drill in on some off by one errors pretty quickly.  I am very grateful for everyones work on the Cortex-Debug extension.  It has made it possible to spend most of my day in vscode instead of Eclipse.

Evaneee, 2020-11-14:
gdb command "set output-radix 16" can help easily see the hex format