Marus/cortex-debug #310: Deleted and disabled breakpoints still breaks execution

peterfing, 2020-07-07:
Had an installation working, but after upgrading Visual Studio Code from 1.46.0 to 1.46.1, deleted and disabled breakpoints still make execution break, restarting the debugging seems to clear them, I have tried to rollback to 1.46.0 but the problem still exist. My target device is LPC54114J256_M4 and using SEGGERs J-Link GDB V6.80a to connect to target.

diggit, 2020-07-10:
I've observed same behavior with STM32 + ST-Link (up to date) + openocd (upstream). Not sure about cortex-debug and vscode versions. I guess, that debug output could tell more. At least if breakpoint removal request arrives to gdb and openocd.

haneefdm, 2020-07-10:
In the Debug Console, you can do "info breakpoints" to see what breakpoints
are actually being set.

It is possible, the device/openocd is not clearing breakpoints properly.
You can read the debug registers to see as well. You can look up the
addresses in the TRM

peterfing, 2020-07-13:
Thank you for your reply! The "info breakpoints" command helps a lot to understand what is going on. I also discovered that I am able to delete my "ghost breakpoints" by using "clear" in the debug console when it has been hit without the need to restart the debugger. So far I have not seen a pattern yet in when my "ghost breakpoints" appears.

haneefdm, 2020-07-13:
You can set `"showDevDebugOutput": true` in launch.json to see what breakpoints are being set by this extension. You will see all the chatter between this extension and gdb. It is kinda hard to understand but ...

zePanzerfaust, 2020-07-25:
I got the same behaviour.
Cortex-Debug 0.3.7

The board used is a FRDM-K64F. Happens with older and the latest SEGGER J-Link V680e.

`Debug Console`
```
Please check OUTPUT tab (Adapter Output) for output from /Applications/SEGGER/JLink_V680e/JLinkGDBServer
Launching server: "/Applications/SEGGER/JLink_V680e/JLinkGDBServer" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MK64FN1M0xxx12"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /project/dir/build/Debug/project.elf...
done.
ethernetif_input (netif=0x2002ffa0) at /project/dir/lwip/port/enet_ethernetif.c:143
143	    while ((p = ethernetif_linkinput(netif)) != NULL)
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
```

Here the code wants to break in some ethernet code, but there is no breakpoint set.

The problem goes away if I remove `"runToMain": true`. Then it initially breaks in the ResetHandler and from then on, I can debug as usual.

This was also happening in CLion. Even though I copied the applicable settings over. Now either some parameters cause it or it's not necessarily a plugin problem. Because I don't know if CLion has something similar to break at main. In CLion the behaviour was a bit different. It did not reset the target. Whenever I set or removed a breakpoint it randomly stopped, wherever the code was at that moment.

`Adapter Output`
```
SEGGER J-Link GDB Server V6.80e Command Line Version
JLinkARM.dll V6.80e (DLL compiled Jul 13 2020 19:01:31)
Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MK64FN1M0xxx12
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 MK64FN1M0xxx12
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:00:28
Hardware: V1.00
S/N: 621000000
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00003C8C (Data = 0xF0006878)
Read 4 bytes @ address 0x2002FE5C (Data = 0x00010D6B)
Read 4 bytes @ address 0x2002FE5C (Data = 0x00010D6B)
Read 4 bytes @ address 0x00010D6A (Data = 0xFE75F7FE)
Read 4 bytes @ address 0x2002FE58 (Data = 0x2002FE80)
Read 4 bytes @ address 0x2002FE4C (Data = 0x2002FFA0)
Read 4 bytes @ address 0x2002FE4C (Data = 0x2002FFA0)
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x00003C8C)
Received monitor command: reset
Resetting target
Downloading 1024 bytes @ address 0x00000000
Downloading 16 bytes @ address 0x00000400
Downloading 4096 bytes @ address 0x00000410
Downloading 4096 bytes @ address 0x00001410
Downloading 4096 bytes @ address 0x00002410
Downloading 4096 bytes @ address 0x00003410
Downloading 4096 bytes @ address 0x00004410
Downloading 4096 bytes @ address 0x00005410
Downloading 4096 bytes @ address 0x00006410
Downloading 4096 bytes @ address 0x00007410
Downloading 4096 bytes @ address 0x00008410
Downloading 4096 bytes @ address 0x00009410
Downloading 4096 bytes @ address 0x0000A410
Downloading 4096 bytes @ address 0x0000B410
Downloading 4096 bytes @ address 0x0000C410
Downloading 4096 bytes @ address 0x0000D410
Downloading 4096 bytes @ address 0x0000E410
Downloading 4096 bytes @ address 0x0000F410
Downloading 4096 bytes @ address 0x00010410
Downloading 4096 bytes @ address 0x00011410
Downloading 4096 bytes @ address 0x00012410
Downloading 4096 bytes @ address 0x00013410
Downloading 4096 bytes @ address 0x00014410
Downloading 4096 bytes @ address 0x00015410
Downloading 104 bytes @ address 0x00016410
Downloading 8 bytes @ address 0x00016478
Downloading 4 bytes @ address 0x00016480
Downloading 4 bytes @ address 0x00016484
Downloading 156 bytes @ address 0x00016488
Writing register (PC = 0x     4c4)
Read 4 bytes @ address 0x000004C4 (Data = 0x480EB672)
Received monitor command: reset
Resetting target
```

`launch.json`
```
{
    "type": "cortex-debug",
    "request": "launch",
    "name": "<project_name> J-Link",
    "cwd": "${workspaceRoot}",
    "executable": "${command:cmake.launchTargetPath}",
    "serverpath": "/Applications/SEGGER/JLink_V680e/JLinkGDBServer",
    "servertype": "jlink",
    "device": "MK64FN1M0xxx12",
    "interface": "swd",
    "jlinkscript": "",
    //"runToMain": true,
    "armToolchainPath": "${env:GNU_ARM_GCC_TOOLCHAIN_DIR}",
    "svdFile": "${workspaceRoot}/MK64F12.xml"
}
```

haneefdm, 2020-07-25:
@zePanzerfaust I cannot reproduce this. I don't know where the ghost breakpoints are coming from. Can you do the following as well? I mean we can't set breakpoints you haven't asked for, so I am puzzled. Could it be you had an exception or a fault? That will look like a breakpoint to gdb/us

https://github.com/Marus/cortex-debug/issues/310#issuecomment-656658821

If you enable `"showDevDebugOutput": true` then you can see (with a bit of an effort) what breakpoints are actually being set.

My vscode version info below
```
Version: 1.47.2
Commit: 17299e413d5590b14ab0340ea477cdd86ff13daf
Date: 2020-07-15T18:18:50.054Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.5.0
```

zePanzerfaust, 2020-07-25:
The binary should work. If no debugger is attached I can connect to the device and it sends me data as it should. I also changed the code, reflashed and checked if it works.
```
Reading symbols from '/project/dir/build/Debug/project.elf'
Finished reading symbols
Please check OUTPUT tab (Adapter Output) for output from /Applications/SEGGER/JLink_V680e/JLinkGDBServer
Launching server: "/Applications/SEGGER/JLink_V680e/JLinkGDBServer" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MK64FN1M0xxx12"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
1-gdb-set target-async on
2-interpreter-exec console "source /Users/<user>/.vscode/extensions/marus25.cortex-debug-0.3.7/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
8-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from /project/dir/build/Debug/project.elf..."}]}
Reading symbols from /project/dir/build/Debug/project.elf...
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"done.\n"}]}
done.
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"0x0000444c in sys_now () at /project/dir/lwip/port/sys_arch.c:737\n"}]}
0x0000444c in sys_now () at /project/dir/lwip/port/sys_arch.c:737
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"737\t}\n"}]}
737	}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["frame",[["addr","0x0000444c"],["func","sys_now"],["args",[]],["file","/project/dir/lwip/port/sys_arch.c"],["fullname","/project/dir/lwip/port/sys_arch.c"],["line","737"]]],["thread-id","1"],["stopped-threads","all"]]}]}
Not implemented stop reason (assuming exception): undefined
GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"connected","results":[]}}
GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
Resetting target
GDB -> App: {"token":5,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["address","0x4c4"],["load-size","91652"],["transfer-rate","366608000"],["write-rate","3160"]]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
Resetting target
GDB -> App: {"token":7,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":8,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
9-break-insert -t --function main
GDB -> App: {"token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","-break-insert: Unknown option ``-function''"]]}}
10-thread-list-ids
GDB -> App: {"token":10,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
11-thread-info 1
GDB -> App: {"token":11,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 57005"],["frame",[["level","0"],["addr","0x000004c4"],["func","Reset_Handler"],["args",[]],["file","/project/dir/startup/startup_MK64F12.S"],["fullname","/project/dir/startup/startup_MK64F12.S"],["line","304"]]],["state","stopped"]]]]]}}
```

haneefdm, 2020-07-25:
Okay, this might help. As you can see, we are not setting any user breakpoints from our extension. But somehow, old breakpoints are not being erased.

Does this happen let's say that you have two breakpoints but just one has been disabled?

I see another problem. We try to set a breakpoint at main and it has always worked with gdb. Looks like now, it is failing. What version of gdb?
```
9-break-insert -t --function main
GDB -> App: {"token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","-break-insert: Unknown option ``-function''"]]}}
10-thread-list-ids
```

haneefdm, 2020-07-25:
@zePanzerfaust Please give us the gdb version #. Looks like they broke the protocol that has been there for ages

zePanzerfaust, 2020-07-25:
The previous recording should have used the following gdb.
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

zePanzerfaust, 2020-07-25:
I never had breakpoints set. Nor did I disable one.

zePanzerfaust, 2020-07-25:
Sorry gdb version is:
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git

zePanzerfaust, 2020-07-25:
In case VS Code took the arm-none-eabi-gdb in the path, it is "GNU gdb (GDB) 7.9".
GNU gdb (GDB) 7.9
This GDB was configured as "--host=x86_64-apple-darwin19.0.0 --target=arm-none-eabi".

haneefdm, 2020-07-25:
OMG, they broke compatibility. --function has been removed. I have no idea which documentation to follow anymore or why they gdb is changing protocols without updating MI version number

One version says...
```
 -break-insert [ -t ] [ -h ] [ -f ] [ -d ] [ -a ]
    [ -c condition ] [ -i ignore-count ]
    [ -p thread-id ] [ location ]
If specified, location, can be one of:

linespec location
A linespec location. See Linespec Locations.

explicit location
An explicit location. GDB/MI explicit locations are analogous to the CLI’s explicit locations using the option names listed below. See Explicit Locations.

‘--source filename’
The source file name of the location. This option requires the use of either ‘--function’ or ‘--line’.

‘--function function’
The name of a function or method.

‘--label label’
The name of a label.

‘--line lineoffset’
An absolute or relative line offset from the start of the location.
```
And another says
```
 -break-insert [ -t ] [ -h ] [ -r ]
    [ -c condition ] [ -i ignore-count ]
    [ -p thread ] [ line | addr ]
If specified, line, can be one of:

function
filename:linenum
filename:function
*address
The possible optional parameters of this command are:

`-t'
Insert a tempoary breakpoint.
`-h'
Insert a hardware breakpoint.
`-c condition'
Make the breakpoint conditional on condition.
`-i ignore-count'
Initialize the ignore-count.
`-r'
Insert a regular breakpoint in all the functions whose names match the given regular expression. Other flags are not applicable to regular expresson.
```
I found a couple of other instances as well where they broke it. Just doing a face palm. They are not supposed to break compatibility like that!?!?!

zePanzerfaust, 2020-07-25:
Definitely doesn't help in developing third-party apps :).
I saw slight differences in command line options between Linux and macOS versions of the same application, but I assume you used the same host?

haneefdm, 2020-07-26:
I think I now know why some breakpoints persist. These are function breakpoints and some logic is missing to remove breakpoints once set. Working on a fix

peterfing, 2020-08-13:
Great to hear that you have found a reason on why it is not working and are working on a fix.
Do you know from what version (of the SEGGERs J-Link GDB?) the compatibility was broken as a temporary solution could be to rollback to an older version?

haneefdm, 2020-08-14:
@peterfing I have no idea when it broke. Perhaps it was always broken. Let us know if the rollback worked.

fordp2002, 2020-09-23:
I am pretty sure I am seeing this issue too.

PGIII, 2020-12-01:
I also seem to be experiencing this with a JLink, seems to only happen if I remove breakpoints while the device is running, if I pause the device and then remove the breakpoints they clear no problem

dzarda, 2021-05-17:
Confirmed with Black Magic Probe. I use `clear` to delete all or `delete` when on the line in question.

haneefdm, 2021-09-09:
I cannot duplicate this issue. I have tried it many times it seems to work properly now. I don't own a BMP so this may be specific to BMP. Please reopen if issues persist.

closed 2021-09-09 by haneefdm