Marus/cortex-debug #312: Disconnecting when attached pauses target

diggit, 2020-07-10:
In debug session configured as `attach`, target gets paused when we get attached. (same behavior in `launch` type session). This is not critical for me, as I usually attach when something bad happens and pausing at this moment does not cause problems. Even though, it would be nice to leave target running unaffected.

**Main issue** is, when you disconnect from `attach` session, target pauses. Can't think of scenario, where such behavior would be wanted.
There is workaround, you can use `launch` session and just clear  launch commands `"overrideLaunchCommands": [ "" ],`.

haneefdm, 2020-07-10:
You can issue a continue right after an attach as part of post startup sequence. It does stop momenterarily but then continues execution. This is the normal trick.

During detach, this an openocd bug. It is supposed to clear all breakpoints, disable debug, and then continue execution. Not what happens. Their whole detach sequence is wrong. I fixed it for Cypress version of OpenOCD but OpenOCD is not receptive to bug fixes. They have a ton of bugs wrt. RTOS threads as well that I fixed. These bugs are in the common code.

closed 2020-07-10 by haneefdm

diggit, 2020-07-11:
Thanks for clarification of situation. I am surprised that OpenOCD did not accept bugfix patches. Do you have link to your patchset in OpenOCD [zylin system](http://openocd.zylin.com)? I'd like to have unbiased opinion :)