Marus/cortex-debug #314: Debug not working FreeRTOS

Krasnov-Max, 2020-07-11:
[marus25.cortex-debug]: Proposed API is only available when running out of dev or with the following command line switch: --enable-proposed-api marus25.cortex-debug

Версия: 1.47.0
Фиксация: d5e9aa0227e057a60c82568bf31c04730dc15dcd
Дата: 2020-07-09T08:30:34.302Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
ОС: Linux x64 5.4.0-40-generic

haneefdm, 2020-07-11:
Could you please explain a bit more. Are you saying when vscode with a proposed API, this extension does not work for FreeRTOS? But, what is not working. There is not enough to figure out what is wrong.

We usually do not test with proposed APIs

Krasnov-Max, 2020-07-11:
OK. 
Until FreeRTOS has started, debugging and step-by-step tracing works, after starting the OS, debugging hangs, and step-by-step tracing does not work. No reaction to breakpoints.
 From messages only that sent earlier.

Marus, 2020-07-24:
Hum - the proposed API thing is a bit strange - shouldn't to my knowledge be anything in there that should be using proposed APIs - can you think of any changes you've done recently @haneefdm that might have inadvertently used some proposed APIs?

Also, could you provide your launch.json file @Krasnov-Max - your behaviour is a little bit surprising - in the past more people have had issues with trying to hit breakpoints and stuff when using FreeRTOS before it has finished launching (as some of the memory structures GDB expects are not initialized at that point - so if it triggers a breakpoint before FreeRTOS has finished launching it can cause issues with the GDB backend that don't recover)

haneefdm, 2020-07-24:
No, I don't touch the proposed APIs. Don't even look at them until published :-) Unless I am the one who proposed the API change which I didn't

Marus, 2020-07-25:
@haneefdm Yeah - I didn't expect that you had - just can't figure out why it would be giving that error message - looks to be an up-to-date VSCode - so shouldn't be anything we are touching that is using any of the proposed APIs (I mean if you go back to 1.42 or something we probably do use things that were proposed at that point) - so that one is baffling.

haneefdm, 2020-07-25:
@Marus I will look into it over the weekend. If it is a real problem, I can report back to MS and nip it in the bud

Krasnov-Max, 2020-07-25:
{
     "configurations": [
    {
        "name": "openocd",
        "request": "launch",
        "type": "cortex-debug",
        "cwd": "${workspaceRoot}",
        "servertype": "openocd",
    //    "serverArgs": ["--serial", "333f69063447313236531057"],
        "interface": "swd",
        "v1": false,
        "runToMain": true,
        "executable": "build/Thermo_Solder.elf",
        "svdFile": "/home/maxx/svd/STM32F3/STM32F302.svd",
        
        "configFiles": [
            "/usr/local/share/openocd/scripts/interface/stlink.cfg",
            "/usr/local/share/openocd/scripts/target/stm32f3x.cfg" 
        ]
        
    }
]
}

haneefdm, 2020-07-25:
It works fine for me with the insiders build or using the command-line option. For Cortex-Debug FreeRTOS is nothing special unless you enabled rtos (which you have not done in launch.json) and even then it is nothing special.

In your openocd configuration files, are you enabling rtos, which is how I enable it? Please check the "Adapter Output" in the "OUTPUT" tab. You will see all the output from OpenOCD. It would be nice for us to see that as well. OpenOCD has bugs when rtos detection is enabled and it does lock up when you do a step in some cases. You will see that happen in the "Adapter Output"

[image]

danielschenk, 2021-03-25:
I'm not able to debug either, when enabling FreeRTOS integration. Breakpoints are not being hit, and the pause button is not responsive. This is my configuration:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "pqs-firmware",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "armToolchainPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin",
            "rtos": "FreeRTOS",
            "serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
            "interface": "swd",
            "device": "stm32f777zi",
            "svdFile": "${workspaceRoot}/project/source/predm_main_cubemx/STM32F7x7.svd",
            "runToEntryPoint": "main"
        }
    ]
}
```

haneefdm, 2021-03-25:
If you hover over `rtos` it will give you help. For JLink, it needs to be a pathname (full or relative to JLink install-dir) to the plugin dll/dylib including the extension.

danielschenk, 2021-03-25:
> If you hover over `rtos` it will give you help. For JLink, it needs to be a pathname (full or relative to JLink install-dir) to the plugin dll/dylib including the extension.

Incorrect. It tells me this:
[image]

If I try to launch with an invalid value like "bla", it even gives me a nice error about it (which doesn't happen with "FreeRTOS"):
[image]

The question now is: is there a problem in the implementation, or are the documentation and parameter checking logic wrong?

danielschenk, 2021-03-25:
One more sign that it doesn't necessarily have to be a path to a custom plugin: I have been using the setting "embOS" with J-Link fine all the time until trying FreeRTOS.

haneefdm, 2021-03-25:
I believe the doc. is in error.

Bullet proof way is to specify full path name for the extension. It is untrue that JLink only supports those two RTOSes. People can create their own, not every hard. I think there may be an issue in the code as well. For example, JLink ships with support for ChbiOS that wasn't there before (screenshot of the JLink install-dir below)

[image]

danielschenk, 2021-03-26:
If I specify the full path to the plugin (dll in my case, as I'm on Windows) I experience the same issues. So I think the issue is not in the way how this is specified in the launch file.

I found that I get this error in my Adapter Output:
```Setting breakpoint @ address 0x0800D7EE, Size = 2, BPHandle = 0x0004
Setting breakpoint @ address 0x0800FC38, Size = 2, BPHandle = 0x0005
Starting target CPU...
...Breakpoint reached @ address 0x0800FC38
ERROR: FreeRTOS maximum used priority is unreasonably big, not proceeding:     8002.
```

However, our `configMAX_PRIORITIES` is set to 20.

In STM32CubeIDE I'm able to debug the same application with FreeRTOS support just fine. However, that IDE has its own older version of the J-Link software, which is 6.80b, while VSCode uses my system-wide one, which is 6.88c. To rule out the version as a possible cause, I changed my launch file to use the J-Link files of STM32CubeIDE, and this makes no difference.

haneefdm, 2021-03-26:
Everyone has a problem with this. GDB servers require a variable called `uxTopUsedPriority` to exist and properly initialized. At some point FreeRTOS removed that variable but gdb servers still support old and new variants of FreeRTOS and hence that variable to exist even though it does not add anything to their logic.

Isn't `configMAX_PRIORITIES` a `#define`? Whag JLink (and OpenOCD and others) are looking for is for a global variable called `uxTopUsedPriority`

Just google it ad see how many people run into this. You will also find code to work around it. Note that the `uxTopUsedPriority` has to survive compiler and linker optimizations

https://forum.segger.com/index.php/Thread/3536-SOLVED-Thread-Aware-FreeRTOS-debugging-seems-to-ignore-pxOverflowDelayedTaskList/

danielschenk, 2021-03-26:
Here you can see the variables used by the debugger when using STM32CubeIDE:
```
Loading RTOS plugin: C:\ST\STM32CubeIDE_1.4.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.jlink.win32_1.4.0.202007081208\tools\bin\GDBServer\RTOSPlugin_FreeRTOS.dll...
RTOS plugin (API v1.0) loaded successfully
RTOS plugin: Loaded
Received symbol: pxCurrentTCB (0x20020D24)
Received symbol: pxReadyTasksLists (0x20020D30)
Received symbol: xDelayedTaskList1 (0x20020ED8)
Received symbol: xDelayedTaskList2 (0x20020EEC)
Received symbol: pxDelayedTaskList (0x20020D28)
Received symbol: pxOverflowDelayedTaskList (0x20020D2C)
Received symbol: xPendingReadyList (0x20020F08)
Received symbol: xTasksWaitingTermination (0x20020F34)
Received symbol: xSuspendedTaskList (0x20020F20)
Received symbol: uxCurrentNumberOfTasks (0x20020EC0)
Received symbol: uxTopUsedPriority (0x00000000)
Received symbol: uxTopReadyPriority (0x20020ED4)
Received symbol: vPortEnableVFP (0x0800910C)
Received symbol: FreeRTOSDebugConfig (0x00000000)
All mandatory symbols successfully loaded.
```

And here when using vscode:
```
Loading RTOS plugin: C:/ST/STM32CubeIDE_1.4.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.jlink.win32_1.4.0.202007081208/tools/bin/GDBServer/RTOSPlugin_FreeRTOS.dll...
RTOS plugin (API v1.0) loaded successfully
RTOS plugin: Loaded
Received symbol: pxCurrentTCB (0x20020D24)
Received symbol: pxReadyTasksLists (0x20020D30)
Received symbol: xDelayedTaskList1 (0x20020ED8)
Received symbol: xDelayedTaskList2 (0x20020EEC)
Received symbol: pxDelayedTaskList (0x20020D28)
Received symbol: pxOverflowDelayedTaskList (0x20020D2C)
Received symbol: xPendingReadyList (0x20020F08)
Received symbol: xTasksWaitingTermination (0x20020F34)
Received symbol: xSuspendedTaskList (0x20020F20)
Received symbol: uxCurrentNumberOfTasks (0x20020EC0)
Received symbol: uxTopUsedPriority (0x00000000)
Received symbol: uxTopReadyPriority (0x20020ED4)
Received symbol: vPortEnableVFP (0x0800910C)
Received symbol: FreeRTOSDebugConfig (0x00000000)
All mandatory symbols successfully loaded.
```
As you can see, in both cases `uxTopUsedPriority` is indeed at a null address. In the thread you mentioned, Segger says that they added support for using `uxTopReadyPriority` instead. I suppose the plugin recognizes that the address of `uxTopUsedPriority` isn't valid and switches to the other variable.

Still I wonder, why does it work in STM32CubeIDE but not in vscode, with the exact same firmware in my uC and the exact same RTOS plugin? I even tried using the same `arm-none-eabi-gdb` version now, which also doesn't help.

haneefdm, 2021-03-26:
We have nothing to do with JLink or FreeRTOS or their plugin. Just trying to help you out. It is really between your elf file and JLink's gdbserver. We launch the gdbserver (exact command-line shown in Debug Console) and then never talk to it again. That conversation happens between gdb and the gdbserver.

There are hundreds of links on the internet about this and I just pasted one.

haneefdm, 2022-01-28:
Closing this because it really has nothing to do with Cortex-Debug. Not that I can tell

closed 2022-01-28 by haneefdm