Marus/cortex-debug #315: "No source file named..." error for breakpoint in .c file

explorer098, 2020-07-16:
Hey,

I am new to Visual Studio Code and use this great extension to debug a project for an Adafruit Feather platform (SAMD21) . In my project I also use the Arduino extension. When I insert breakpoints in my .ino file (Arduino "main" file), debugging works as expected. However, I make use of a seperate .c file located in my workspace forlder. There, inserting breakpoints throws the following error (I replaced the actual path name components with letters a-e for clarity):

"No source file named d:\a\b\c\d\e\CFILE.c. (from break-insert "d:\\a\\b\\c\\d\\e\\CFILE.c:318")"

I am sure, that this file exists in the workspace!
Obviously there is a difference in the number of "\", might this cause the problem?

Here is my launch.son:
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": ".//.build/${fileBasename}.elf",
            "request": "launch",
            "serverpath": "D:/Programs/JLink_V614b/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "device": "ATSAMD21G18",
            "runToMain": true,
            "armToolchainPath": "C:/Users/E/AppData/Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin"
        }
    ]
}

Thanks for your support!

haneefdm, 2020-07-16:
Do you have any special/Unicode characters in the file-name or spaces? GDB only works with ASCII path names. There is no workaround for that.

explorer098, 2020-07-16:
Thanks for your comment @haneefdm. I read about this problem in other issues. My path does only have ASCII characters.
I noticed, that in the build folder, which is generated from the underlying Arduino (gcc) build-process, there is also a .c file generated with the exact same name as my original source file directly in the workspace folder. If I open it and set breakpoints, it works! 
This is the (simplified) structure of my folder:
Workspace Folder
--> .ino file (breakpoints work)
--> CFILE.c  (breakpoints don't work)
--> CFILE.h
-->.build folder
-----> .elf file (the on to wich I am pointing the executable path in launch.json)
-----> sketch folder
-------------> CFILE.c (breakpoints work)

I do not understand why the breakpoints work in my .ino file, but not in that .c file in the same path but probably it's rather my lack of knowledge about how the debug process works.

haneefdm, 2020-07-16:
I am glad it works. It really depends on your build process and which .c file **actually** got compiled. Maybe there is a post-processing step before compiling. I am not familiar with Arduino development env.

I am going to close this issue if that is okay.

closed 2020-07-16 by haneefdm

rei-vilo, 2021-02-12:
I had the same error message but, in my case, I've just forgotten to specify `-g3` as build option.