Marus/cortex-debug #316: After starting debug session, vscode defaults to "TERMINAL" tab instead of "DEBUG CONSOLE"

Teslafly, 2020-07-22:
After pressing f5 to compile and start debugging, the active tab in the "problems/output/terminal/debug console" area is "terminal" with the output of gcc. If the program compiles successfully I would like it to automatically switch over to the "debug console" tab. When you are in a quick "change something, f5, debug" loop, it becomes a pain to click over to the debug console every time when it should recognize that's where I want to be since the code is uploaded and I am halted at main.

Even an option to do so in launch.json such as:
`"console": "internalConsole"`
would be nice if  there is a reason to not have this behavior default.
https://stackoverflow.com/questions/49572658/vs-code-starts-debugging-in-integrated-terminal-instead-of-debug-console

haneefdm, 2020-07-22:
This is completely out of our control. We don't do a build or know its status. It is done by VSCode and finally, we only respond to a request to start debugging and even then we have a tiny bit of control. It is all up to VSCode

For me, it always switches to Debug Console already. I wonder if something changed in core VSCode

haneefdm, 2020-07-22:
I will look into VSCode API to see if there is a way to force a switch to "Debug Console".

I believe this area is going through some overhaul. Now you can drag other stuff (left side panel items) into this panel. Waiting for things to settle down.
[image]

closed 2021-09-11 by haneefdm