Marus/cortex-debug #317: SWO Graphing

Gasman2014, 2020-07-24:
Thanks for this extension.

I am trying to get graphing working but am hitting a problem.  With this launch.json

```
          {
            "type": "console",
            "label": "SWO [1]",
            "port": 1,
            "encoding": "utf8"
          },
```

I get this console output;

```
[2020-07-24T07:34:21.686Z]   0
[2020-07-24T07:34:21.882Z]   1
[2020-07-24T07:34:22.086Z]   2
[2020-07-24T07:34:22.283Z]   3
[2020-07-24T07:34:22.495Z]   4
[2020-07-24T07:34:22.694Z]   5
[2020-07-24T07:34:22.885Z]   6
[2020-07-24T07:34:23.079Z]   7
[2020-07-24T07:34:23.291Z]   8
[2020-07-24T07:34:23.486Z]   9
[2020-07-24T07:34:23.670Z]   10
[2020-07-24T07:34:23.874Z]   11
...
```

I also get the same output if the encoding is set to ascii.

However, when I try to graph it, I get this:

[image]

I was expecting a monotonically increasing stair step plot.

This is the relevant bit of my launch.json. I have tried most, if not all, combinations of encodings. I have also tried with and without the first ascii decoder. It appears that the data is not recognised as an integer stream but is being decomposed into ascii with a framing error. Any advice?

```
        "decoders": [
          {
            "type": "graph",
            "port": 1,
            "encoding": "unsigned",
            "graphId": "1",
            "scale": 1
          },
        ]
      },
      "graphConfig": [
        {
          "label": "Graph 1",
          "timespan": 5,
          "type": "realtime",
          "annotate": false,
          "maximum": 128,
          "minimum": 0,
          "plots": [
            {
              "graphId": "1",
              "label": "Increment",
              "color": "#53753c" 
            }
          ]
        }
      ]
```

Marus, 2020-07-24:
Hi,

By the look of your output what you are currently sending over the SWO is string representations of the data, separated by new lines (hence why the print out nicely with the console). What the graphing system expects is actual binary data to be sent over - so say if you were using 8 bit values then each value would be one single byte sent over with the actual value (so 0x00 0x01 0x02 ...). There should be configuration settings to use most common (although I won’t guarantee all) binary formats for the data. 

Currently there is nothing in the grapher to parse data from ASCII/UTF8 string output. Generally I’d recommend putting each different bit of data on a separate ITM port.

Marus, 2020-07-24:
To clarify a bit further - something like the following could be used to send the value over an ITM port (this is using the the STM32F10x CMSIS header (`stm32f10x.h`) and `core_cm3.h` - may require some variation depending on your controller):

```
void ITM_SendValue (int port, uint32_t value)
{
    if ((ITM->TCR & ITM_TCR_ITMENA_Msk) && (ITM->TER & (1UL << port)))
    {
        while (ITM->PORT[port].u32 == 0);
        ITM->PORT[port].u32 = value;
    }
}
```

You should not be packing multiple values into a single ITM frame either.

Gasman2014, 2020-07-24:
Thanks - that's good advice about the ITM format and generally very helpful. I will have another look at the library I have been using and see if I can get that to write a byte stream.

Marus, 2020-07-24:
I expect most libraries will have something similar - if not I expect that snippet above will work with minimal or no modifications as long as you are using CMSIS headers - as ITM/SWO is part of the main core and not vendor specific.

I'll close this for now - but if you can't get it to work with this information please re-open and we'll see if we can find other problems.

closed 2020-07-24 by Marus

Gasman2014, 2020-07-28:
Thanks. I have got this working now however there is still one issue - the default graphing mode interpolates between the last and current reading. This may be reasonable and appropriate for some scalar quantities e.g. temperature but a non interpolated graph option should be available.  

Would you like me to open this as a new issue?

Example;
```
void loop()
{
  for (n = 1; n < 10; ++n)
  {
    ITM_SendValue(1, n);
    delay(100);
  }
  delay(100);
}
```
Expected result - a staircase plot like this.
[image]

Actual result - each value is plotted and the trace continues horizontally until the next value arrives - at which point the two values are interpolated;

[image]

Marus, 2020-07-28:
Yes - you could create a new issue there to enhance that with a non-interpolated mode. Have to look at the graphing library I'm using to see how hard that would be to implement (point form would be easy for sure - but a stepped line like you were expecting might take a bit more work)