Marus/cortex-debug #318: Incomprehensible error

Sanya-zik, 2020-07-24:
Hi guys!

I've been trying to run the debugger for two days, but still got no result.

I'm using st-link + stm32f407vet6.

Running OpenOCD and GDB from terminals everything works. I can do "next", "step", etc.

But an error is happening in vscode. The openocd server starts up normally, but when gdb is connected it says:

```
Reading symbols from .....
0x000032f6 in <stm32f4xx_hal::delay::Delay as embedded_hal::blocking::delay::DelayUs<u32>>::delay_us (self = 0x1000ff9c, us = 200000) at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f4xx-hal-0.8.3\src/delay.rs: 69
69 while! Self.syst.has_wrapped () {}
Not implemented stop reason (assuming exception): undefined
```

The line that begins with the address `0x000032f6 in ...` changes with every start.

In this case, breakpoints are ignored wherever I put them.

Sometimes after starting the debugger, execution stops for some reason at line 489 (pub unsafe ...) in the `cortex-m-rt-0.6.12/src/lib.rs` file:
```
// ...
# [doc (hidden)]
# [no_mangle]
pub unsafe extern "C" fn Reset () ->! {
    extern "C" {

        // These symbols come from `link.x`
        static mut __sbss: u32;
        static mut __ebss: u32;

        static mut __sdata: u32;
        static mut __edata: u32;
        static __sidata: u32;

    }
    // ...
```
There is a stop in other files, but very rarely.

Maybe I'm doing something wrong, but I've tried everything and it still doesn't work.

Debug console from start to stop on the line `pub unsafe extern "C"`
```
Reading symbols from 'D:\stm32\projects\p-blink-01\target\thumbv7em-none-eabihf\debug\p-blink-01'
Finished reading symbols
Please check OUTPUT tab (Adapter Output) for output from openocd.exe
Launching server: "openocd.exe" "-c" "gdb_port 50000" "-s" "D:\stm32\projects\p-blink-01" "-f" "interface/stlink.cfg" "-f" "target/stm32f4x.cfg" "-c" "tpiu config internal C:/Users/User/AppData/Local/Temp/tmp-9020Q7ai8AYVF28L uart off 48000000 2000000"
Launching GDB: "arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
1-gdb-set target-async on
2-interpreter-exec console "source c:/Users/User/.vscode/extensions/marus25.cortex-debug-0.3.7/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor reset halt"
5-target-download
6-interpreter-exec console "monitor reset halt"
7-enable-pretty-printing
8-interpreter-exec console "EnableITMAccess"
9-interpreter-exec console "BaseSWOSetup 23"
10-interpreter-exec console "SetITMId 1"
11-interpreter-exec console "ITMDWTTransferEnable"
12-interpreter-exec console "DisableITMPorts 0xFFFFFFFF"
13-interpreter-exec console "EnableITMPorts 0x1"
14-interpreter-exec console "EnableDWTSync"
15-interpreter-exec console "ITMSyncEnable"
16-interpreter-exec console "ITMGlobalEnable"
17-interpreter-exec console "DisablePCSample"
undefinedC:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
[... 184 lines cut ...]
  "executable": "./target/thumbv7em-none-eabihf/debug/p-blink-01",
  "name": "Debug (OpenOCD)",
  "device": "STM32F407VET6",
  "configFiles": [
    "interface/stlink.cfg",
    "target/stm32f4x.cfg"
  ],
  "svdFile": "${workspaceRoot}/.vscode/STM32F407.svd",
  "swoConfig": {
    "enabled": true,
    "cpuFrequency": 48000000,
    "swoFrequency": 2000000,
    "source": "probe",
    "decoders": [
      {
        "type": "console",
        "label": "ITM",
        "port": 0
      }
    ]
  },
}
```

Let me know if you need any more information on this error, I'll try to get it.

Marus, 2020-07-24:
Unfortunately, I don't know to much about rust (I want to learn more - but there is never enough time) - and why it might not be able to interpret things properly when run through the extension (we do some things were we parse things from objdump which may break if something ends up in a format we didn't expect - but I wouldn't have expected this to effect simply setting/hitting breakpoints). 

I know some people (@Samonitari and @perlindgren have I believe) had used it in the past with rust - although maybe something has changed that has caused it to stop working. Don't know if they might have any feedback about what they may have been doing differently.

haneefdm, 2020-07-24:
I did modify the parser for the objdump output to handle de-mangled names for C++ and Rust and to handle IAR generated excutables.. @muttering-oldman Can you provide just the executable and take a look. And, what do the list of breakpoints look like.
```
Not implemented stop reason (assuming exception): undefined
```
I have seen this all the time because when a reset happens, it appears like a breakpoint be hit or a hard fault but the reason gdb receives is unknown reason. It is not fatal.

But from this issue, I don't understand what is actually not working. gdb may not be liking the elf file produced to enable breakpoints?!?!

muttering-oldman, 2020-07-24:
@haneefdm I'm afraid I didn't quite get it about the breakpoint list. Where do I need to look? Is this an extension output?
Could you please tell me in more details what exactly should I do?

haneefdm, 2020-07-24:
Hmmm, unless it comes from disassembly, we don't mess with path names or line numbers. They go directly to gdb interprets them.

The double back-slashes and forward slash are generally okay. Even multiple of them. They are generally collapsed into one

From your last attachment, everything seems to have worked. gdb did not fail, did it? So, what is not working?

muttering-oldman, 2020-07-24:
Breakpoints do not interrupt program execution. I put them everywhere in the code, but none of them worked. Perhaps I am doing something wrong.

haneefdm, 2020-07-24:
In the debug console, can you do a `info breakpoints` and then continue. It might show you the addresses where the breakpoint is actually set. We need to make sure those addresses are valid. This can happen if source does not match the executable being debugged which may not match what is actually programmed in the device. Everything has to be in sync

muttering-oldman, 2020-07-25:
After connecting, I executed the command "load". 
Then I executed "bt" to view the stack. 
I proceeded with "continue" and then my console was locked. I pressed ctrl+c to stop and ran "bt" again to view the stack.
Then I set two breakpoints to "b main.rs:34" and "b main.rs:37", and then I executed "info b" to view all breakpoints.

But unfortunately the breakpoints are ignored and the execution is not interrupted. I'll continue researching and supplement the information.

```
$ arm-none-eabi-gdb -q target\thumbv7em-none-eabihf\debug\p-blink-01
Reading symbols from target\thumbv7em-none-eabihf\debug\p-blink-01...
(gdb) target remote :3333
Remote debugging using :3333
0x000013ee in cortex_m::peripheral::syst::<impl cortex_m::peripheral::SYST>::has_wrapped (
    self=0x16e36000)
    at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-0.6.3\src\peripheral/syst.rs:136
136             self.csr.read() & SYST_CSR_COUNTFLAG != 0
(gdb) load
Loading section .vector_table, size 0x1a8 lma 0x0
Loading section .text, size 0x475c lma 0x1a8
Loading section .rodata, size 0xbf0 lma 0x4910
Start address 0x475a, load size 21748
Transfer rate: 98 KB/sec, 5437 bytes/write.
(gdb) bt
#0  Reset ()
    at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.6.12\src\lib.rs:489
(gdb) continue
Continuing.

Program received signal SIGINT, Interrupt.
0x000032fa in <stm32f4xx_hal::delay::Delay as embedded_hal::blocking::delay::DelayUs<u32>>::delay_us (
    self=0x1000fe14, us=500000)
    at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f4xx-hal-0.8.3\src/delay.rs:69 69                  while !self.syst.has_wrapped() {}
(gdb) bt
#0  0x000032fa in <stm32f4xx_hal::delay::Delay as embedded_hal::blocking::delay::DelayUs<u32>>::delay_us (self=0x1000fe14, us=500000)
    at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f4xx-hal-0.8.3\src/delay.rs:69 #1  0x00003240 in <stm32f4xx_hal::delay::Delay as embedded_hal::blocking::delay::DelayMs<u32>>::delay_ms (self=0x1000fe14, ms=500)
    at C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f4xx-hal-0.8.3\src/delay.rs:32 #2  0x000002f2 in p_blink_01::__cortex_m_rt_main () at src\main.rs:36
#3  0x00000208 in main () at src\main.rs:15
(gdb) b main.rs:34
Breakpoint 1 at 0x2be: file src\main.rs, line 34.
(gdb) b main.rs:37
Breakpoint 2 at 0x2f6: file src\main.rs, line 37.
(gdb) info b
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x000002be in p_blink_01::__cortex_m_rt_main at src\main.rs:34
2       breakpoint     keep y   0x000002f6 in p_blink_01::__cortex_m_rt_main at src\main.rs:37
```

haneefdm, 2020-07-25:
Looks like the program is in some sort of a blocking delay loop when you interrupted the program

Is it possible you have an infinite loop or the delay is too large? If gdb cannot stop at a breakpoint, then there is not much Cortex-Debug can do.

haneefdm, 2020-07-25:
I should close this issue because the message is coming from OpenOCD and it does not impede anything and it has always been there on a reset. We know that gdb stopped but it doesn't know what the reason is and we are reporting back to you.

I think we are mixing up multiple issues and the message may be incomprehensible but what else should it say? The bug is really with OpenOCD where if the program stops, it should always give a reason.

closed 2020-07-25 by haneefdm