Marus/cortex-debug #319: Cannot start debug, when flash address is not set to default value in linker script

maschgit, 2020-07-28:
Hello,
my application is splitted in several memory areas via linker script file.
The main application is starting not at address 0x08000000, but at 0x08010200.
The Program counter address is: 0x08010200.
The stack pointer address is:  0x0C010200.
When I try to start to debug I got the following failure:
`Please check OUTPUT tab (Adapter Output) for output from C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe
Launching server: "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "XMC4500-F144x1024"
Launching GDB: "C:\GnuToolChain\9_2019_q4_major\bin\arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
undefinedC:\GnuToolChain\9_2019_q4_major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from <path>/application.elf...
0x000000e2 in ?? ()
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
SWO enabled successfully.
Warning: the current language does not match this frame.
Exception condition detected on fd 552`

When I try to debug the same *.elf file in Segger Ozone, debugging works without a problem.
My expectation is, that the debugger automatically knows where to flash and where to start the application, like Ozone is doing this.
I can reproduce this behavior, by just changing the location.
Do I need to set a parameter to be able to debug in VSC or ist this a failure?

Marus, 2020-07-28:
That's an odd one - I can't think of anything on the extension side that should effect this - as we simply provide the file+line locations to GDB. Have you tried to see if that same executable can be properly debugged by `arm-none-eabi-gdb` directly in the command line - as if GDB cannot interpret the executable properly the extension will not be able to.

haneefdm, 2020-07-28:
Could it be that in Ozone, you are using a combined bin file to program the flash, whereas in VSCode you are just using the elf file? It still doesn't explain what you are seeing.

As an experiment,  can you try programming via Ozone and debugging via VSCode? To do this you may have to use `overrideLaunchCommands` and not do a target-download. Just do a reset

maschgit, 2020-07-29:
Thank you for your answers.
@Marus I tried your suggestion. The result ist, that debugging works fine. I tried it several times and it works always.
The console output is:
[image]

@haneefdm In ozone I am using the exactly same elf file.
->To do this you may have to use 'overrideLaunchCommands' and not do a target-download. Just do a reset
-->> How to configure 'overrideLaunchCommands'?

I observed two or three times, that after ending debugging with ozone or command line and directly after this starting debugging with vsc, vsc is doing everything fine. I was able to debug the elf file.
But I couldn't surly reproduce this behavior. So I can't tell you how to get this. :-(
After erasing the memory in the µController and restarting the debug process in vsc, I get again the same "faulty" behavior.

haneefdm, 2020-07-29:
What I don't see in the gdb output is loading the target. I see the following sequence
```
target remote localhost:50000
monitor reset
```
And, then, it looks like it is attaching to a running program and gdb is asking you for a file to load for symbols as, without it, it won't work. In JLink output, do you see the programming happen?

I will give you two options below. The first one bypasses programming

```
    "overrideLaunchCommands" : [
            "monitor halt",
            "monitor reset",
            "-target-download",
            "monitor reset",
            "enable-pretty-printing"
    ]
```
This is what Cortex-Debug does to start the debugging session. You can change this sequence or comment out the download and the following reset

I believe. in Ozone, you can use an elf file to debug but a bin/hex file to program. If so, you can do the same. target-download makes gdb send the programming regions to link, but if you have a bin/hex file, you can replace target-download with

```
    "preLaunchCommands": [
        "exec-file path-name-to-hex-or-elf-file"
     ]
```
Oh, you should enable debug output to see exactly what is going on
```
    "showDevDebugOutput": true
```

maschgit, 2020-07-31:
@haneefdm 
>What I don't see in the gdb output is loading the target. I see the following sequence
--> you will find this at both line, where the first grey line is. Beginning with: "<ld/Debug/...."

>And, then, it looks like it is attaching to a running program and gdb is asking you for a file to load for symbols as, without it, it won't work. In JLink output, do you see the programming happen?
-->>No. With the command "load file.elf" segger will be started to erase the µController and flash with new data.
After this I wrote the command "file projectfile.elf". What the user sees is the second grey marked line "<ld/..".
Then the third grey marked line loads the symbols. This is where the text is written: "Reading symbols from...."

The described procedure I took from [here](https://mcuoneclipse.com/2015/03/25/command-line-programming-and-debugging-with-gdb/). There you will finde the section "**Segger GDB Server**"

>I believe. in Ozone, you can use an elf file to debug but a bin/hex file to program.
-->That is not correct. Because I only create an elf file, there is no bin or hex file.
Ozone loads, flashes and debugs the elf file.

@Marus I tried to debug from command line. 
Flashing and debugging works perfectly. 
I created some text files with the output of the command line and from VSC.
The first three files are output files where the start address in the linker script file is 0x08000000.
In this case, as I wrote, debugging is also possible in VSC.
[JLinkServerRunning.txt](https://github.com/Marus/cortex-debug/files/5006293/JLinkServerRunning.txt)
[arm-none-eabi-gdb-running.txt](https://github.com/Marus/cortex-debug/files/5006294/arm-none-eabi-gdb-running.txt)
[VSCDebugLogWorking.txt](https://github.com/Marus/cortex-debug/files/5006295/VSCDebugLogWorking.txt)
Note: the VSCDebugLogWorking.txt only contains the first path of output. After the last line there are functions from the code written. I removed them. I think the flash and boot process is enough.

In the second case the start address is 0x0800C000, because at 0x08000000 is the bootloader located.
In this case command line is still working, but VSC is not.
I get the following output:
[JLinkServerModifiedLinkerScript.txt](https://github.com/Marus/cortex-debug/files/5006306/JLinkServerModifiedLinkerScript.txt)
[arm-none-eabi-gdb-ModifiedLinkerScript.txt](https://github.com/Marus/cortex-debug/files/5006307/arm-none-eabi-gdb-ModifiedLinkerScript.txt)
[VSCHangsAfterFlashing.txt](https://github.com/Marus/cortex-debug/files/5006309/VSCHangsAfterFlashing.txt)

The linker script file is the infineon default linker script:
[XMC4500x1024.txt](https://github.com/Marus/cortex-debug/files/5006325/XMC4500x1024.txt)
In this file I changed the start address from 0x08000000 to 0x0800C000.

I talked to a colleague of mine. He said, maybe in the extension somewhere is set, that an application can only start at default address 0x08000000. Values which are unequal ending in this failure.
It could be, that I do something wrong. But I have no idea what. To my opinion such a use case, where different memory sections are defined (using bootloader and application) is nothing special or unusual.

haneefdm, 2020-07-31:
I am out of my depth here wrt. to JLink, scratching my head. Hopefully, you always erased before you run your experiments

I have never seen the `<;d/<Path>`. I don't understand

```
(gdb) monitor device = XMC4500-F144x1024
Selecting device: XMC4500-F144x1024
<ld/<Path>/ProjetFile.elf
```
Is that a prompt from gdb where you typed the file name? Thing is who is interpreting the elf file

1. Is JLink server programming directly from the elf file?
2. Or gdb sending programming instructions to program-specific memory areas when `-target-download` is executed

Cortex-Debug uses the latter method which does not seem to be working. If that is failing, then gdb is not interpreting the elf file properly. It feels like both Cortex-Debug and command-line are doing the same thing but using a machine interface (MI) rather than the human interface. Yes, there are differences. To emulate it exactly, you can use the MI as well because all the commands are printed in the Debug Console, then we can see where the problem is. You have to start gdb exactly like the way Cortex-Debug does and use the same sequence, alter it and figure out what is going wrong. Following were the commands send to gdb. You can skip the 2nd command which has to do with SWO

```
1-gdb-set target-async on
2-interpreter-exec console "source c:/Users/MyName/.vscode/extensions/marus25.cortex-debug-0.3.7/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
```
Something is wrong with command #6 above. FYI: We don't do anything to where the program starts or what is in the reset vector.

Since, I don't have your board or the elf file ....

If it is the first method, JLink may be interpreting the elf file more accurately than gdb. Perhaps there is a workaround we can try using a `monitor` command to do the programming

haneefdm, 2020-07-31:
To launch gdb in MI mode, use the following
```bash
C:\GnuToolChain\9_2019_q4_major\bin\arm-none-eabi-gdb.exe -q --interpreter=mi2 <path-to-elf-file>
```

maschgit, 2020-08-03:
@haneefdm I did what you said.
Here is the copied log: 
[arm-none-eabi-gdb-ModifiedLinkerScript-WithCMDParameter.txt](https://github.com/Marus/cortex-debug/files/5015360/arm-none-eabi-gdb-ModifiedLinkerScript-WithCMDParameter.txt)

The J-Link Server output is:
[JLinkServerModifiedLinkerScriptOutput_WithModifiedCMDParamterIn-arm-none-eabi-gdb.txt](https://github.com/Marus/cortex-debug/files/5015388/JLinkServerModifiedLinkerScriptOutput_WithModifiedCMDParamterIn-arm-none-eabi-gdb.txt)

maschgit, 2020-10-07:
Hello.
I recreated my linker script file.
Since then, I can debug from out of VSC.
Thank you for your help.

closed 2020-10-07 by maschgit