Marus/cortex-debug #322: pyOCD support probably not up do date

tdjastrzebski, 2020-08-03:
I suggest improving pyOCD support. Since November 2018 and [this commit](https://github.com/mbedmicro/pyOCD/commit/bed8ca061c1c00f57d79550ee292e24cf5617f77#diff-2eeaed663bd0d25b7e608891384b7298
) **pyocd-gdbserver** as a startup command is deprecated.
`0000782:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.`

In the log attached below I see another problem `Target type 'stm32f746zgtx' not recognized.` My Nucleo-746ZG board is auto-detected as target **stm32f746zgtx**, which is unsupported. **stm32f746zg** target is supported. I do not find a way to override target type.

```
0000782:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
0000974:ERROR:gdb_server:uncaught exception: Target type 'stm32f746zgtx' not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md> for how to install additional target support.
Traceback (most recent call last):
  File "c:\program files\python36\lib\site-packages\pyocd\board\board.py", line 61, in __init__
    self.target = TARGET[self._target_type](session)
KeyError: 'stm32f746zgtx'
```

``` yaml
{
    "name": "Debug (pyOCD)",
    "type": "cortex-debug",
    "request": "launch",
    "servertype":"pyocd",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceRoot}/BUILD/NUCLEO_F746ZG/GCC_ARM-DEBUG/${workspaceRootFolderName}.elf",
    "device": "STM32F746ZG",
    "v1": false,
    "svdFile": "${workspaceRoot}/STM32F7x6.svd",
    "runToMain": true
}
```

haneefdm, 2020-08-03:
I know. I already made the changes but did not push them out. Reason: it will be a breaking change for people who set a pathname for the server executable, and I don't have the bandwidth to deal with that.

So, I told myself, let's just stay in the compatibility mode until pyocd truly stops supporting it. Maybe I am wrong. When I look at pyocd source code, yes there are a lot of differences between the two different launch types

For now, you can create a shell script for `pyocd-gdbserver` to launch `pyocd` in the new way

Maybe the time has come to switch. I don't use pyocd on a regular basis but any advice is appreciated on what the usage model is. Do people specify a full pathname for us to find or just use the system path? I am talking Win/Linux/Mac

I like pyocd BTW, but they should not have deprecated that way.

tdjastrzebski, 2020-08-03:
@haneefdm I think the question is how many people still use almost two-years-old pyOCD release 0.13 and earlier and how much new pyOCD functionality, fixes and improvements is not exposed by cortex-debug due to self-imposed limitations.  
Frankly, I do not know the answer but I think it is at least time to consider moving forward.  
I am relatively new to embedded programming but my experience so far is rather painful, probably due to lack of years of experience. Whatever it is worth, cortex-debug, despite its limited documentation and examples is the tool which allowed me to move forward.

haneefdm, 2020-08-03:
I will wait for @Marus to comment for a few days. Give him a chance to reflect

patrislav1, 2020-11-03:
> If compatibility with older pyOCD versions is a real concern, I suggest popping up VS Code warning window for some time if detected pyOCD version is too old, before the change is made.

Another way would be to provide `pyocd-gdbserver` or `pyocd-legacy` as yet another server type, for the legacy pyOCD, if there are users still depending on it. Then the `pyocd` server type could be allowed to move forward - people who update Cortex-Debug but don't update pyocd then just have to change the server type in their `launch.json`.

patrislav1, 2020-11-07:
> I already made the changes but did not push them out. Reason: it will be a breaking change for people who set a pathname for the server executable, and I don't have the bandwidth to deal with that.

By the way, @haneefdm, would you be willing to share that code? I need the current version of `pyocd` for my project, but I can't get the current cortex-debug to work with it.

I would also be willing to contribute, if that helps getting the necessary changes upstream.

stephendpmurphy, 2021-01-14:
I second the updating of support for pyocd to use the latest and recommended gdb server commands. Just recently created my own DAP-Link board  and running into the same problems as @tdjastrzebski.

I am managing to make it work though by using the ```serverArgs``` config option. The target is an STM32F413RG which is not natively supported in pyOCD (I had to add support using the ```pyocd packs``` command).

My limited amount of testing appears to show it working and allows me to set breakpoint and step through my source. Take note that the ```--target``` and ```stm32f413rg``` are their own params in the serverArgs array.
```json
{
    "name": "Debug (pyOCD)",
    "type": "cortex-debug",
    "request": "launch",
    "servertype":"pyocd",
    "cwd": "${workspaceRoot}",
    "executable": "./build/my-firmware.elf",
    "device": "stm32f413rg",
    "runToMain": true,
    "v1": false,
    "serverArgs": [
        "--target",
        "stm32f413rg"
    ]
}
```

patrislav1, 2021-02-07:
I would like to point out that while pyocd users are supposed to use a outdated version, openocd users are now [supposed to use a bleeding edge version that's not even in the OS package managers yet](https://github.com/Marus/cortex-debug/wiki/Home/_compare/ea6b22363c28bb44a7030c36f4dc19ef9cc139cb...cefc809ccddf3f36ca0e080f0b5adad4cb31ebb5).

Maybe it's time to update the pyocd interface in Cortex-Debug eventually, I would do it myself if I had any typescript skills.

newbrain, 2021-02-21:
I'll add my two cents:
The only reason `pyocd-gdbserver` is still around might be this very extension (BTW: thanks @Marus, it's great!).
The other deprecated tools will probably be removed from next version.
See the last point in this [pyOCD PR](https://github.com/pyocd/pyOCD/pull/1080).

I second @tdjastrzebski  idea of a warning - if one is using such an old pyOCD, it's really time to upgrade.

Marus, 2021-02-22:
I do think we probably should head towards the new interface for pyocd - yeah - while it may cause a few people to have to update configurations can't hold on to the past forever.

Like @haneefdm thought I don't typically use pyocd - although it looks like it now works with STLink (and possibly J-Link) hardware - so hopefully I should be able to get this running to try out things and hopefully update

tdjastrzebski, 2021-02-24:
I know this may be a bit provocative but several months have passed and, while I am still new to embedded programming, my perspective has changed a bit.  
Why not to drop pyOCD support and continue with OpenOCD only? pyOCD seems to be good only for nucleo (but not all) and similar boards but not for custom hardware while OpenOCD is good for both.
Drop [stutil](https://github.com/stlink-org/stlink) support as well. It's last release was almost a year ago, not much new development, no support for new MCUs like STM32H7A3.
The only thing I would add to cortex-debug is flash support for OpenOCD.
If license permits, I would consider redistributing cortex-debug with OpenOCD - that means fast start for new enthusiasts and reliable debugging = good experience. Is not it all what counts?

RisinT96, 2021-02-24:
What do you mean pyOCD support is only good for nucleo? It supported any Cortex M mcu, while they might not have out-of-the boa support for all MCUs, you can relatively easily add support for new ones with industry standard CMSIS packs.

newbrain, 2021-02-24:
> pyOCD seems to be good only for nucleo 

[citation needed]

Really, I use cortex-debug + pyOCD with , let me check:
* NXP: LPCXPRESSO43S67, MIMXRT1021EVK, OKDO E1 (LPC55S9), MKE04Z8 & LPC1769 (self built stuff), 
* Many self built boards with STM32F0xx and F4xx
* Now, the Raspberry Pi foundation RP2040.
* and yes, Nucleos (F411, L476, F401) and Discos (F746, F429, L476)

newbrain, 2021-02-24:
> OpenOCD is always such a pain to set up and launch.
Yes, so true. I only use it as it comes precanned in VisualGDB.
Every time I tried to set it up for myself I went mad.

tdjastrzebski, 2021-02-24:
I meant: nucleo and similar. If I count it correctly, pyOCD supports only 14 ST processors, mostly those found in Nucleo boards. I think it is a bit surprising since, best to my knowledge, pyOCD development is lead by ST folks.
I agree, OpenOCD is a bit pain, recent builds are difficult to find but, personally, I found it very stable and versatile.

patrislav1, 2021-02-24:
Here's a non-exhaustive list of devices covered by pyocd through cmsis packs. Good luck covering all those with openocd

https://www.keil.com/dd2/

haneefdm, 2021-02-24:
My $0.02. pyOCD is actually pretty good and responsive to bug fixes. OpenOCD bugs leaks like crazy, not good at threaded apps., too many forks. A bug fix can take years to get resolved with OpenOCD

newbrain, 2021-02-24:
>  If I count it correctly

Here is the rub, see @patrislav1 comment.
I've yet to find something it would not handle. I never got OpenOCD working with the 43S67 for example (you are in a maze of twisty little ~~passages~~ configuration files, all alike)

> pyOCD development is lead by ST folks.

As far as I know, it's tied to Arm, not ST.
Its most active (currently) maintainer is, in fact, from Arm.

> recent builds are difficult to find 

And this (along with the atrocious complexity*) is really a problem.
From their page (updated in 2018!):
> OpenOCD is available in many different forms from numerous places, this page should serve as a guide directing you to the option most appropriate for the usecase.

(*) I don't deny it has (probably?) more features. I maintain they are difficult to use. pyOCD is extremely accessible.

tdjastrzebski, 2021-02-24:
@newbrain Exactly, this is my point, OpenOCD has more features, supports more processors, seems to be more stable and better supported - at least this is my experience. But the same time it is more complex and less accessible.  
And here is where cortex-debug comes into play!

RisinT96, 2021-02-24:
How is it more stable and better supported if you have to read an entire wiki just to find which version you need to download?
Are you just trolling us?

newbrain, 2021-02-24:
I think you should reevaluate your statement about "supports more processors".
Did you check the link in @patrislav1  comment?
Is there any mainstream or even obscure **Arm Cortex-M** missing?
Did you count them?
The total is **7173** different devices.

Yes, pyOC is for Arm Cortex-m.
That, and the fact it does not really support JTAG (yet) are the only thing missing, IME.

Note that we are discussing this on an extensions called "cortex-debug" so I would expect the lack of support for other architectures the epitome of moot point!

tdjastrzebski, 2021-02-24:
All right, that was just an idea, probably not the best one.
What I am missing from pyOCD is support for STM32H7A3 and OpenOCD supports this chip without problems.

newbrain, 2021-02-24:
> support for STM32H7A3 

Ok, at this point I totally agree with @RisinT96:

[image]

tdjastrzebski, 2021-02-24:
Well, `pyocd list --targets` does not mention STM32H7A3. Where does this list come from?

newbrain, 2021-02-24:
Now we are talking, and possibly, listening...
I ask you (again) to check the link: https://www.keil.com/dd2/
All the MCUs in that list are supported by pyocd, you just need to issue:
`> pyocd  pack -i <<desired model>>`
(see below for more complete help)

This is the result:
```
(venv) ~\Git\pyOCD [LinuxLibUSB ≡]> pyocd pack -i stm32h7a3
Downloading packs (press Control-C to cancel):
    Keil::STM32H7xx_DFP::2.7.0
(venv) ~\Git\pyOCD [LinuxLibUSB ≡]> pyocd list --target | Select-String 'stm32h7a'    
  stm32h7a3agixq            STMicroelectronics      STM32H7A3AGIxQ               STM32H7 Series, STM32H7A3   pack     
  stm32h7a3aiixq            STMicroelectronics      STM32H7A3AIIxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3igkx             STMicroelectronics      STM32H7A3IGKx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3igkxq            STMicroelectronics      STM32H7A3IGKxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3igtx             STMicroelectronics      STM32H7A3IGTx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3igtxq            STMicroelectronics      STM32H7A3IGTxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3iikx             STMicroelectronics      STM32H7A3IIKx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3iikxq            STMicroelectronics      STM32H7A3IIKxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3iitx             STMicroelectronics      STM32H7A3IITx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3iitxq            STMicroelectronics      STM32H7A3IITxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3lghxq            STMicroelectronics      STM32H7A3LGHxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3lihxq            STMicroelectronics      STM32H7A3LIHxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3nghx             STMicroelectronics      STM32H7A3NGHx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3nihx             STMicroelectronics      STM32H7A3NIHx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3qiyxq            STMicroelectronics      STM32H7A3QIYxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3rgtx             STMicroelectronics      STM32H7A3RGTx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3ritx             STMicroelectronics      STM32H7A3RITx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3vghx             STMicroelectronics      STM32H7A3VGHx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3vghxq            STMicroelectronics      STM32H7A3VGHxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3vgtx             STMicroelectronics      STM32H7A3VGTx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3vgtxq            STMicroelectronics      STM32H7A3VGTxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3vihx             STMicroelectronics      STM32H7A3VIHx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3vihxq            STMicroelectronics      STM32H7A3VIHxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3vitx             STMicroelectronics      STM32H7A3VITx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3vitxq            STMicroelectronics      STM32H7A3VITxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3zgtx             STMicroelectronics      STM32H7A3ZGTx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3zgtxq            STMicroelectronics      STM32H7A3ZGTxQ               STM32H7 Series, STM32H7A3   pack
  stm32h7a3zitx             STMicroelectronics      STM32H7A3ZITx                STM32H7 Series, STM32H7A3   pack
  stm32h7a3zitxq            STMicroelectronics      STM32H7A3ZITxQ               STM32H7 Series, STM32H7A3   pack

(venv) ~\Git\pyOCD [LinuxLibUSB ≡]>
```

This is the internal help for the `pack` command:
```
(venv) ~\Git\pyOCD [LinuxLibUSB ≡]> pyocd pack -h  
usage: pyocd pack [-h] [-v] [-q] [-c] [-u] [-s] [-f GLOB] [-i GLOB] [-n] [-H]

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         More logging. Can be specified multiple times.
  -q, --quiet           Less logging. Can be specified multiple times.

pack operations:
  -c, --clean           Erase all stored pack information.
  -u, --update          Update the pack index.
  -s, --show            Show the list of installed packs.
  -f GLOB, --find GLOB  Report pack(s) in the index containing matching device part numbers.
  -i GLOB, --install GLOB
                        Download and install pack(s) containing matching device part numbers.

pack options:
  -n, --no-download     Just list the pack(s) that would be downloaded, don't actually download anything.
  -H, --no-header       Don't print a table header.
(venv) ~\Git\pyOCD [LinuxLibUSB ≡]> 
```

haneefdm, 2021-02-24:
To all: pyOCD is not ST specific. It is ARM Cortex-M specific - may support Cortex-A, I don't know, supports multiple cores very well, funded by ARM (I think). I lent a hand last year and the ARM people were very supportive. Yes, it still evolving but lot fewer bugs, very well written, modular.

If you want your device supported, talk to your MCU vendor. It probably already is.

Also, people using `Mbed OS` may want to use pyOCD, which is how I got sucked into it. Last thing I support is dropping pyOCD support. We can re-evaluate using `pyocd-gdbserver` because the startup python code between old and new his diverged a bit. We have to be carefull

haneefdm, 2021-02-25:
@newbrain Thanks

OpenOCD: I had to deal with Tcl scripts (config files), their C code. They don't even follow the proper gdb-sever protocol. Wrong on so many levels, but I got a working version for my company. Still not satisfied because these are generic problems that all devices can benefit from. Buffer overflows, I got that PR through. But there are so many more!?

patrislav1, 2021-02-25:
@haneefdm Fully agree, maybe except for the 'we have to be careful' bit.
Python is a fast-moving ecosystem, and I think it's unlikely that people run the latest and greatest cortex-debug version together with a pyocd version from three years ago. And - maybe I'm not getting the full picture here - but I think even those users can just `pip3 install --upgrade pyocd` and be fine, right? Also quite ironic that for OpenOCD, a [bleeding edge release candidate is recommended](https://github.com/Marus/cortex-debug/wiki) while for PyOCD an API is used that's deprecated since 2 1/2 years...

patrislav1, 2021-02-25:
By the way, I made this little [wrapper script](https://github.com/Marus/cortex-debug/issues/351#issuecomment-725528193) that fixes the issues with the current pyocd version. The changes necessary for cortex-debug should be minimal - I would open a PR myself, if I had any Typescript skills.

haneefdm, 2021-02-25:
@patrislav1 :-) being careful is my default mode. 2.5 years ago, things were different, now pretty much you have to upgrade pyOCD very frequently to get bug fixes and new device support

newbrain, 2021-02-25:
>  if I had any Typescript skills.

I understand TS as well as cuneiform, but I sometimes (after updates) just modify the `debugadapter.js` file to remove `-gdbserver` from the pyocd name (not even needed if you specify your own path) and instead add `gdbserver` to the array of arguments.
Unfortunately, the latter can't be done using "args" in the launch.json, as it would end up in the wrong place (pyocd commands must be the first argument).

I would expect similarly simple changes in the TS source.

So:
`[js gibberish]serverExecutable(){return this.args.serverpath?this.args.serverpath:"pyocd-gdbserver"}serverArguments(){const e=this.ports.gdbPort,t=this.ports.consolePort;let r=["--persist","--port",e.toString(),"--telnet-port",t.toString()][more js gibberish]`

Becomes:
`serverExecutable(){return this.args.serverpath?this.args.serverpath:"pyocd"}serverArguments(){const e=this.ports.gdbPort,t=this.ports.consolePort;let r=["gdbserver","--persist","--port",e.toString(),"--telnet-port",t.toString()]`
Ugly hack? Why would you say that...

flit, 2021-02-27:
(I'm the pyocd maintainer.)

Sorry I missed out on the conversation here, I was really busy with work the last week. I won't try to reply to everything (not that it would be possible 😄 ), but maybe can answer a few things.

### Me

I am indeed an Arm employee, working in the Research group. Mostly, pyocd is developed in my spare time, with a small amount on Arm time.

###  Pyocd and Arm

Pyocd was originally created and owned by the Mbed OS team at Arm. I was involved from almost the beginning, long before I joined Arm. In mid-2020, pyocd ownership was transferred to me so that it can become a true community project, and any community members can potentially become committers/maintainers. That's why the GitHub org changed to 'pyocd'.

### Raison d'être

The goal (mission statement if you want to call it that) is for pyocd to be an easy to use debug tool and gdbserver for Arm-based* devices. A major part of pyocd is the REPL interface with good support for CoreSight low level debug (e.g. chip bringup) and the Python API. A common use case aside from development is in CI and other automated uses.

### Device support

Currently pyocd only supports Cortex-M. There are plans to add AArch32 and eventually AArch64. All depends on available time. Help is welcomed! 😉 

It's true that technically "any" Cortex-M device with a CMSIS-Pack is supported. However, there are, of course, issues with certain devices, and plenty of bugs. And the packs themselves are sometimes incorrect. If you have an issue, please report it! It's hard for me to keep up with every issue, but I do my best to address problems and fix bugs.

For reference, the [target support docs](https://github.com/pyocd/pyOCD/blob/master/docs/target_support.md) show how to use CMSIS-Packs. I'm considering auto-install of packs, ie if you specify a part number for which a pack exists but is not installed (but unexpected, automatic network access is somewhat uncomfortable for me as a security researcher 😅 ).

### Other

The state of pyocd docs is sorely recognized as lacking. 😢 I'm in the process of building a website (slowly), and then will start writing more docs. If anyone that wants to help, I would be massively appreciative!

I have a *huge* todo list and set of features I want to add. It will keep getting better. 

If you have any other questions, probably should ask on via [pyocd discussions](https://github.com/pyocd/pyOCD/discussions).

* Now that pyocd is not Arm owned, there is room for other architectures (e.g., RISC-V). But I can't work on that directly aside from general architecture and plumbing, as it would conflict with my Arm employee agreement.

haneefdm, 2021-02-27:
@flit Thanks for chiming in Chris, and making things clearer

Chris, if I find something wrong, it is usually vendor code for programming, When I am able, I am willing to contribute. Especially in the core

jmc8, 2021-05-13:
Is anyone planning on driving this through to completion?
It's been an open ticket for months, with a lot of talk, but I don't see any commits regarding the issue.

@haneefdm : You say you already made the changes but searching the repo yielded no results. Can you please tell me the commit rev/number your work is at?

Is there any way to simply override Cortex-Debug's hardcodes and call pyocd with my own command line?

haneefdm, 2021-05-18:
See PR #435.

> Is there any way to simply override Cortex-Debug's hardcodes and call pyocd with my own command line

Sure, you can write your own shell/bat script named `pyocd-gdbserver` that will intercept and do what you want. Note that the path is not hardcoded. Just the **default** name of the executable

haneefdm, 2021-07-10:
To all, just upgraded the pyocd.
* no --persist
* Stop using `pyocd-gdbserver` and use `pyocd gdbserver`
* SWO support over TCP. We use the raw interface which is kinda new. Works just like any other server and have not found any problems yet. There was one tiny problem (memory regions layout for ST devices incomplete, so had to workaround that)

closed 2022-01-28 by haneefdm