Marus/cortex-debug #323: cortex-debug main problem

tdjastrzebski, 2020-08-03:
I believe currently the single, main cortex-debug problem is the lack of documentation.  

Sample configurations once probably available at
https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/
are no longer accessible since the website is down for months.  
The Wiki page https://github.com/Marus/cortex-debug/wiki does not contain much info.

Based on my experience, sample, typical configuration for OpenOCD, pyOCD and JTag (just those three) would save beginners like myself many hours of research and frustration.  

The most helpful information which allowed me to get things working I found here:
* https://higaski.at/vscode-arm-cortex-m-debugging/
* https://lonesometraveler.github.io/2020/03/27/debug.html
* https://blog.csdn.net/qq_40833810/article/details/106713462
* https://www.programmersought.com/article/45004258269/

My point is that cortex-debug is a great tool. I believe with just some documentation and sample configs it could become even more popular.

haneefdm, 2020-08-03:
I don't disagree. But we are also looking for volunteers to update out Wiki. It is just me and @Marus 

I can't even deal with the number of issues and no one else helps out. This is a problem where we can't even duplicate issues without the required HW/boards and an elf file. But, we guess and try to resolve things.

I hope people understand.

tdjastrzebski, 2020-08-04:
@haneefdm probably @Marus still has this website content with useful examples. Copy-pasting it to Wiki is probably a matter of minutes. Otherwise, or in the mean time, the list of blogs I posted above may help some people a lot.

haneefdm, 2020-08-04:
Yes, you can find it on the Wayback machine

https://web.archive.org/web/20190701000000*/https://marcelball.ca/projects/cortex-debug/

But be careful because some info may be out of date. Not much

tdjastrzebski, 2020-08-04:
Thanks, I will look at it. For now I just added list of blogs to the Wiki page.

malizd, 2020-12-16:
Hi,

Forgive me if this is not right place to ask the question.
How can I see program and ram memory dump in cortex-debug. Wiki says "view memory" command, but where? how?
Can someone point me or perhaps a screenshot.
Cheers,
Matt

PhilippHaefele, 2021-12-27:
@haneefdm As Links are in Wiki, all commands will be described in the wiki in the future and there's another issue for the old webpage being down (#277) we can close this one

closed 2023-04-14 by haneefdm