Marus/cortex-debug #325: The possibility of using this debugger for other ISA's

yunuscukran, 2020-08-05:
Hi,

I was wondering if there was any talk or examples of people using this debugger for other architectures. Given that this extension provides the ability to customize OpenOCD and GDB versions, it seems to me that it could be a great platform for building debuggers for other architectures as long as they're using the standard GDB + OpenOCD + JTAG stack.

In fact, I am currently trying to see if I can get it to work with a RISC-V chip. Just by changing the openOCD and GDB versions into their RISC-V counterparts, I was able to run the debugger, set breakpoints, view local and static variables.

I've had some problems getting the register values (as I've mentioned in https://github.com/riscv/riscv-openocd/issues/500), but that too seems to be an issue regarding GDB-MI and OpenOCD discrepancies, and not an ARM vs RISC-V problem.

Would love to hear your thoughts on this, or experiences, if anyone else tried doing this.

haneefdm, 2020-08-06:
Yes, I suppose you can use this extension to develop for other architectures including x86. There are a few assumptions built-in but they should not be serious, I hope

- Endian-ness. There are some areas where little-endian is assumed
- Only 32-bit. It would be a major challenge to support 64-bit as it is not natively supported by JavaScript
- We query gdb to get the register list (names and values). It is not hardcoded nor do we care. What is hardcoded however is decoding of registers named `XPSR`, `CPSR`, and `CONTROL` into their flags. But other than that it should work.
- Again, we expect that gdb does the right thing (along with openocd, jlink) when we query for register names and values. We have no magic. We expect that gdb follows the MI spec and so do we. We never interact with OpenOCD other than starting it and hooking gdb to OpenOCD. If gdb does not work, neither do we.
- Ability to read memory arbitrarily without alignment issues. Meaning we need to be able to ask gdb for a set of bytes at arbitrary locations. This applies to memory-window and peripheral-window
- Ability to get disassembly from gdb for a given function. We don't care what it returns. Just something legible for the user so we can display
- That you have a version of `objdump` that gives us symbol names properly. If it is a proper elf file then it should work
- I don't know if there are equivalents to SWO, SVD, etc., but if you don't use those features then not a problem

I may have missed a few things. My main problem is the lack of HW when things don't work properly and no one helps us. I already have this issue with ARM devices. We get so many issues reported to us where we do not have the HW setup to be able to debug. Takes a long time to keep guessing what could be going wrong.

Finally, the name of the extension can be a problem :-) I am sure a blog can cure that but it is unlikely we can rename the extension

haneefdm, 2020-08-06:
Oh, we HAVE to use the gdb MI interface. There is no other option because non-mi query/result keeps changing whereas the MI interface is a contract. If we are violating that contract in some way, please let us know.

I saw that issue https://github.com/riscv/riscv-openocd/issues/500 and yes, that register list is ginormous. I believe the register names actually come from gdb and not OpenOCD. I could be wrong. Looks like gdb and openocd have two versions of the truth with a mismatch. openocd seems right though.

Whoever configured gdb put everything including the kitchen sink rather than making specific for a given arch. or variant thereof.

yunuscukran, 2020-08-06:
Regarding the MI Interface, I don't know if it breaks the contract, but it seems that the cortex-debugger is calling **data-list-register-values x** in https://github.com/Marus/cortex-debug/blob/master/src/gdb.ts, and tries to get register values all in one go. If one of the register reads fails though, the function stops enumerating and does not return the previous successful reads. 

Do you think more graceful error handling there would be a useful feature? Then it wouldn't have to rely on the XML it's getting from GDB and could just work with registers that exist. That's my first idea on how to tackle my own RISC-V problem. The all or nothing behaviour seems baked into that GDB-MI command, and not just the calling code though, which makes it a bit harder to deal with.

haneefdm, 2020-08-06:
> If one of the register reads fails though, the function stops enumerating and does not return the previous successful reads.

Not quite true. GDB tells us which registers got updated in bulk. It does it by index/number and not by name and not every register is updated every time. Sparsely populated.

To query/update a register at a time will severely impact the user experience. It goes from milliseconds to multiple seconds. The round trips are pretty slow. I would not even consider it.

I believe this needs to be addressed by the RISC-V ecosystem for gdb support. `data-list-register-values` should never ever fail. There should not be phantom registers. Here is an example that what we see and parse according to the MI contract. This is where ARM spend a lot of time to get the GDB interface right for their processors

https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_223.html

```
(gdb)
-data-list-register-values x
^done,register-values=[{number="0",value="0xfe0043c8"},
{number="1",value="0x3fff88"},{number="2",value="0xfffffffe"},
{number="3",value="0x0"},{number="4",value="0xa"},
{number="5",value="0x3fff68"},{number="6",value="0x3fff58"},
{number="7",value="0xfe011e98"},{number="8",value="0x2"},
{number="9",value="0xfa202820"},{number="10",value="0xfa202808"},
{number="11",value="0x1"},{number="12",value="0x0"},
{number="13",value="0x4544"},{number="14",value="0xffdfffff"},
{number="15",value="0xffffffff"},{number="16",value="0xfffffeff"},
{number="17",value="0xefffffed"},{number="18",value="0xfffffffe"},
{number="19",value="0xffffffff"},{number="20",value="0xffffffff"},
{number="21",value="0xffffffff"},{number="22",value="0xfffffff7"},
{number="23",value="0xffffffff"},{number="24",value="0xffffffff"},
{number="25",value="0xffffffff"},{number="26",value="0xfffffffb"},
{number="27",value="0xffffffff"},{number="28",value="0xf7bfffff"},
{number="29",value="0x0"},{number="30",value="0xfe010000"},
{number="31",value="0x0"},{number="32",value="0x0"},
{number="33",value="0x0"},{number="34",value="0x0"},
{number="35",value="0x0"},{number="36",value="0x0"},
{number="37",value="0x0"},{number="38",value="0x0"},
{number="39",value="0x0"},{number="40",value="0x0"},
{number="41",value="0x0"},{number="42",value="0x0"},
{number="43",value="0x0"},{number="44",value="0x0"},
{number="45",value="0x0"},{number="46",value="0x0"},
{number="47",value="0x0"},{number="48",value="0x0"},
{number="49",value="0x0"},{number="50",value="0x0"},
{number="51",value="0x0"},{number="52",value="0x0"},
{number="53",value="0x0"},{number="54",value="0x0"},
{number="55",value="0x0"},{number="56",value="0x0"},
{number="57",value="0x0"},{number="58",value="0x0"},
{number="59",value="0x0"},{number="60",value="0x0"},
{number="61",value="0x0"},{number="62",value="0x0"},
{number="63",value="0x0"},{number="64",value="0xfe00a300"},
{number="65",value="0x29002"},{number="66",value="0x202f04b5"},
{number="67",value="0xfe0043b0"},{number="68",value="0xfe00b3e4"},
{number="69",value="0x20002b03"}]
```

yunuscukran, 2020-08-07:
I see, yes this seems to be a`data-list-register-values` problem.

When I try to get them all, this is what I get:
```
-data-list-register-values x 
  0000914010: Thread Warning: repl: eval. expression '-data-list-register-values x' with no thread context. Using default
0000914010: 29-thread-select 1
0000914010: 30-stack-select-frame 0
0000914010: 31-data-list-register-values x
0000914020: GDB -> App: {"token":29,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["new-thread-id","1"],["frame",[["level","0"],["addr","0x0000970e"],["func","??"],["args",[]]]]]}}
0000914020: GDB -> App: {"token":30,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000914021: GDB -> App: {"token":31,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Could not fetch register \"ft0\"; remote failure reply 'E0E'"]]}}
Could not fetch register "ft0"; remote failure reply 'E0E' (from data-list-register-values x)
```

Corresponding OpenOCD log is:

```
Debug: 45896 913969 gdb_server.c:3373 gdb_input_inner(): received packet: 'T1'
Debug: 45897 913969 gdb_server.c:404 gdb_put_packet_inner(): sending packet '$OK#9a'
Debug: 45898 913970 gdb_server.c:3373 gdb_input_inner(): received packet: 'p21'
Debug: 45899 913970 riscv.c:1672 riscv_get_gdb_reg_list_internal(): rtos_hartid=0, current_hartid=0, reg_class=0, read=0
Debug: 45900 913970 riscv.c:3051 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 45901 913970 riscv.c:3209 riscv_get_register_on_hart(): [riscv.tap.0]{0} fpr0 does not exist.
Debug: 45902 913970 gdb_server.c:1320 gdb_get_register_packet(): Couldn't get register ft0.
Debug: 45903 913970 gdb_server.c:1413 gdb_error(): Reporting -4 to GDB as generic error
Debug: 45904 913970 gdb_server.c:404 gdb_put_packet_inner(): sending packet '$E0E#ba'
```

Meanwhile, a single register read works.

```
-data-list-register-values r 2
0000136880: 44-thread-select 1
0000136881: 45-stack-select-frame 0
0000136881: 46-data-list-register-values r 2
.....
0000136884: GDB -> App: {"token":45,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000136884: GDB -> App: {"token":46,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","2"],["value","0x1000ff10"]]]]]}}
{"token":46,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","2"],["value","0x1000ff10"]]]]]}}

```
 I still want to note that `ft0`, where the fail happens, is the first register on that gigantic name list that doesn't exist on my machine or in the OpenOCD generated list. It may be related. I will raise the issue with GDB folks. It may also have something to do with me running the RISC-V as a simulator.

Thanks!

haneefdm, 2020-08-07:
I am going to close this issue as there is not really much I can do. I wish gdb could fetch whatever registers are available rather than erroring out. It is okay to produce a warning. Machine interface though is kinda rigid

closed 2020-08-07 by haneefdm

GitMensch, 2021-09-17:
Is the toolchain directly used when running with external/gdbserver or is the toolchain then used directly on the server side?

haneefdm, 2021-09-17:
@GitMensch Only two tools from the tool-chain. `gdb` and `objdump` will be required locally. It may work even if `objdump` does not exist (disassembly mode will silently not work)

GitMensch, 2021-09-17:
Can you please summarize why [`-data-disassemble`](https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI-Data-Manipulation.html), possibly with the `mode` option, is not used instead of `objdump`?
Should I file a new issue requesting that change?

haneefdm, 2021-09-17:
@GitMensch No, I can't summarise exactly. We do use `data-disassemble`. The original author (the awesome @Marus ) did that and disassembly was not a high priority for me. So, we don't use the disassembly from objdump. Just the function addresses (symbol table) to get a start/end of a function. The symbol table is also used to display Globals and Statics.

Having said that, we will probably re-do disassembly completely. So, you can do source mixed with disassembly -- the entire program, not just a function at a time. I worked with MS to create a spec for it and there is now a sample implementation in the MS cpptools debugger. It is still a bit rough on the edges and fresh out of the oven. I have an eye on it though.

haneefdm, 2021-09-17:
We use `data-disassemble` two different ways --- when we know which function we are in or use raw addresses as last resort. I have not used it that much myself

GitMensch, 2021-09-17:
> Having said that, we will probably re-do disassembly completely. So, you can do source mixed with disassembly -- the entire program, not just a function at a time.

The entire program... (I _think_ you mean process) that could be useful, but only for "small tin". Commonly I run processes with multiple megabytes of executable code, producing the disassembly for all of the shared objects will take a while, inspecting that would not be reasonable either. So: please leave "function" as an option. I'll create a "drop objdump" issue next.

haneefdm, 2021-09-17:
Yes, it will look like the entire program where we lazy fill stuff. And, no programs are usually not mult-megabytes or actual code. Only regions you see are fetched. I want to do something similar to memory views as well. This is how commercial debuggers do it.