Marus/cortex-debug #329: Debugger exits if breakpoint is hit in FreeRTOS task

sidprice, 2020-08-20:
I am debugging an Ambiq Apollo3 target (Cortex-M) that uses FreeRTOS, the debug interface is a JLink.. I find that if I run the code without breakpoints it appears to work. However, if I have a breakpoint set in a FreeRTOS task, when that breakpoint is hit the debugger exits. The following is displayed in the "Debug Output" window:

```
Breakpoint 2, LedTask (pvParameters=<optimized out>) at ./src/led_task.c:273
273	        if (bitSet != 0)
/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-775_20180622_1529687456/src/gdb/gdb/inline-frame.c:167: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? 
(y or n) [answered Y; input not from terminal]

```
The "Output" window shows:

```
Setting breakpoint @ address 0x0000C55A, Size = 2, BPHandle = 0x0003
Starting target CPU...
...Breakpoint reached @ address 0x0000C55A
Reading all registers
Removing breakpoint @ address 0x0000C55A, Size = 2
Read 4 bytes @ address 0x0000C55A (Data = 0x28004604)
Reading 64 bytes @ address 0x10003700
Read 4 bytes @ address 0x0000C970 (Data = 0x681B4B0A)
Reading register (MSP = 0x100022E0)
Reading register (PSP = 0x10003700)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       2)
Reading register (FPSCR = 0x 2000000)
Reading register (s0 = 0x       0)
Reading register (s1 = 0x       0)
Reading register (s2 = 0x       0)
Reading register (s3 = 0x       0)
Reading register (s4 = 0x       0)
Reading register (s5 = 0x       0)
Reading register (s6 = 0x       0)
Reading register (s7 = 0x       0)
Reading register (s8 = 0x       0)
Reading register (s9 = 0x       0)
Reading register (s10 = 0x       0)
Reading register (s11 = 0x       0)
Reading register (s12 = 0x       0)
Reading register (s13 = 0x       0)
Reading register (s14 = 0x       0)
Reading register (s15 = 0x       0)
Reading register (s16 = 0x       0)
Reading register (s17 = 0x       0)
Reading register (s18 = 0x       0)
Reading register (s19 = 0x       0)
Reading register (s20 = 0x       0)
Reading register (s21 = 0x       0)
Reading register (s22 = 0x       0)
Reading register (s23 = 0x       0)
Reading register (s24 = 0x       0)
Reading register (s25 = 0x       0)
Reading register (s26 = 0x       0)
Reading register (s27 = 0x       0)
Reading register (s28 = 0x       0)
Reading register (s29 = 0x       0)
Reading register (s30 = 0x       0)
Reading register (s31 = 0x       0)
Reading register (d0 = 0x       0)
Reading register (d1 = 0x       0)
Reading register (d2 = 0x       0)
Reading register (d3 = 0x       0)
Reading register (d4 = 0x       0)
Reading register (d5 = 0x       0)
Reading register (d6 = 0x       0)
Reading register (d7 = 0x       0)
Reading register (d8 = 0x       0)
Reading register (d9 = 0x       0)
Reading register (d10 = 0x       0)
Reading register (d11 = 0x       0)
Reading register (d12 = 0x       0)
Reading register (d13 = 0x       0)
Reading register (d14 = 0x       0)
Reading register (d15 = 0x       0)
Read 4 bytes @ address 0x0000C970 (Data = 0x681B4B0A)
GDB closed TCP/IP connection (Socket 856)
```
Could you help me to understand what is happening and perhaps how I can fix the problem.

sidprice, 2020-08-24:
Sorry to bother you about this, I don't expect to get a response anytime soon to my GDB ticket so I need to figure a way of being able to keep my project moving forward.

If you could point me to a reference for getting OpenOCD up, assuming it would work with the MCU I have (Ambiq Apollo3), I need to give it a try. I assume it bypasses the JLink GDBServer and uses the JLink probe directly?

Again, thanks for you time and help, it is much appreciated.

haneefdm, 2020-08-25:
I would actually contact SEGGER first. The issue may actually lie there first.

sidprice, 2020-08-27:
@haneefdm Thanks for the suggestion, so far no response from Segger.

I have downloaded and installed OpenOCD but I cannot get it to connect to my JLink. I tried changing the driver to WinUSB, but that did not help.

Are you familiar with using JLink with OpenOCD?

Here is the response I get with the Segger driver installed:
`PS C:\utils\xpack-openocd-0.10.0-14\bin> .\openocd.exe -f interface\jlink.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Error: No J-Link device found.
`

With WinUSB installed it simply says no JLink found.

haneefdm, 2020-08-28:
Sorry, @sidprice I have no answers. One last thing I can recommend is to contact your chip vendor. If you cannot use gdb from the command line, the buck stops from them.

sidprice, 2020-08-28:
@haneefdm  Thanks for the moral support, it is much appreciated. I am able to use GDB directly because I can avoid doing the stack info commands that Cortex-Debug always issues. It would be outstanding if I could optionally disable the part of Cortex-Debug that sends those commands. Using GDB directly is a little bit of a pain.

haneefdm, 2020-08-28:
@sidprice We can't disable the stack trace at all because none of the windows will update because it is all dependent on the stack frames. Debugger becomes useless and the whole VSCode infrastructure will fail

If you are using gdb manually for instance and you hit a breakpoint or exception, then what you can do is to run a `backtrace` command. That is pretty much what we do on the current thread except in machine interface

https://sourceware.org/gdb/onlinedocs/gdb/Backtrace.html

sidprice, 2020-08-28:
@haneefdm I tried the backtrace and got the same error Cortex-Debug gets:

```
(gdb) backtrace
#0  usfsmax_task (pvParameters=<optimized out>) at ./src/usfsmax_task.c:354
#1  0x0000d4a4 in xEventGroupSetBits (
/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/src/gdb/gdb/inline-frame.c:156: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n)
```
I am trying to get OpenOCD working, but so far have not been able to find a target configuration for the Apollo3. Ambiq technical support is a complete waste of time.

haneefdm, 2020-08-29:
Feel bad for you. _You have to push your chip vendor on this_. Even OpenOCD, the chip vendor has to provide proper config files. Not easy to create your own. I would say near impossible.

So, what does Ambiq recommend for a dev. env.? I could not even get to their website. It was down.

sidprice, 2020-08-29:
Ambiq recommends Keil, IAR as dev. environments. Their examples also have makefiles for GCC.

I think the problem here may be GDB because I have read other people having issues with FreeRTOS and GDB. Unfortunately, I don't find any solution, other than not using the stack information commands or backtrace. 

Provided I don't use those commands when using GDB directly I can reliably debug my project. However, as I am sure you can imagine, it is not as nice an experience as using Cortex-Debug in VSCode. I find myself switching between GDB and VSCode to browse my code and decide on my next debugging move. I do not find the lack of the backtrace too limiting, at least so far in the project.

Pressing Ambiq is and has been a lost cause. My last interaction with a "support" engineer ended with him saying something like, "we are a small company so producing documentation for the SDK API is something we are not able to do. You should read the source code."

Thanks.

gbingersoll, 2022-02-10:
I know this is a couple years old at this point, but I just ran into a similar problem. Not with this VS Code plugin, but an issue with GDB, Segger, FreeRTOS. I found that the GDB MI command `-stack-info-depth --thread 5` (the 5 is irrelevant) would completely hang GDB and never receive a response. However, I found that this only seems to happen if I compile FreeRTOS with `-O0` optimization. If I compile FreeRTOS with, e.g., `-Og`, and the rest of my code with `-O0`, it works fine. I suspect a bug in Segger's FreeRTOS plugin. At least changing the FreeRTOS compile optimization is a possible workaround for others Googling this problem.

Incidentally, I also happen to be using Ambiq Apollo3, but I would be surprised if it is specific to that chip.

gbingersoll, 2022-02-10:
You're right that if you use -O<n> then you also have to add -g to get debug info. My point is just that even with `-Og`, the compiler often over-optimizes so you cannot see variable values when you step through code. If you keep seeing, "optimized out" with `-Og`, you can go to `-O0` to fix this. However, compiling FreeRTOS with `-O0` seems to result in problems with the Segger FreeRTOS plugin (I suspect) and getting stack info. So I set my application code to compile with `-g -O0` to prevent "optimized out" and compile FreeRTOS with `-Og` so the debugger works as expected.