Marus/cortex-debug #332: Data breakpoints (watchpoints) support

bobatsar, 2020-09-14:
Is it possible to add data breakpoints (watchpoints) to this extension?

https://code.visualstudio.com/Docs/editor/debugging#_data-breakpoints
https://sourceware.org/gdb/current/onlinedocs/gdb/Set-Watchpoints.html

I'm using the jlink gdb server and manual setting the watchpoint in the debugwindow is working.
It would be nice to set a watchpoint from the variables view.

haneefdm, 2020-09-14:
Short answer no. All breakpoints are managed by VSCode and not by this extension. Maybe you can ping Microsoft. Even the Variables/Watch windows are controlled by VSCode

https://github.com/microsoft/vscode/issues/55931

go2sh, 2020-09-14:
Sorry to interrupt, but data breakepoints are supported. See https://microsoft.github.io/debug-adapter-protocol/specification. There is a capability and a few requests that needs to be handled.

bobatsar, 2020-09-14:
See also the release note when they introduced data breakpoints. They reference, that the extensions have to handle it.

https://code.visualstudio.com/updates/v1_38#_breaking-when-value-changes-data-breakpoints

please reopen

haneefdm, 2020-09-14:
I knew data breakpoints were supported in the debug protocol (DAP) which is not just for VSCode. I contributed many things to that protocol. Not all of the protocol is exposed by the VSCode GUI though.

I did not know they added the `Break When Value Changes` option. I have to figure out how it actually works and if it works at all

PhilippHaefele, 2021-12-27:
Just to mention it here too. Since version V0.4.8 there's a `Preliminary support for data watchpoints`. In general it works for me, i only had some issue with deleting a watchpoint so far.

PhilippHaefele, 2022-01-04:
@haneefdm Watchpoint deleting fails because of the following GDB result value on breakpoint creation:
The breakpoint number couldn't be parsed in https://github.com/Marus/cortex-debug/blob/master/src/backend/mi2/mi2.ts#L595 as the result from GDB doesn't have an element named `bkpt` instead it has a `hw-awpt` element:

[image]

So changeing the line
`const bkptNum = parseInt(result.result('hw-awpt.number'));` fixes the issue. Also line 596 doesn't make sense for watchpoints and can be removed.

For my tests i used GNU Arm Embedded Toolchain `10 2021.10` and `8 2019-q3-update` together with `J-Link 7.60a`.

Can you please confirm this with your setup?

Best regards
Philipp

PhilippHaefele, 2022-01-04:
Thanks for the quick response and fix.

> Interesting issue with global variables. If you set a watchpoint on a global, and you restart, the program may stop in some unknown (to most users) location. It is actually the C-runtime code where the variable is initialized (either to zero or to value specified at declaration). People won't expect that but it is the nature of the beast.

Yup. As we have our custom startup code (written in C) it‘s more clear on our side. But of course it his can be very confusing if a user is not aware of that something like that exists.

PhilippHaefele, 2022-01-04:
@haneefdm I also think we can close that now or do you want to wait until the final release?

haneefdm, 2022-01-04:
I will close this. Thanks for your help @PhilippHaefele.

It will be in the next release (preview and final)

closed 2022-01-04 by haneefdm