Marus/cortex-debug #333: ST-Link Gdb Server wrong startup arguments

CoffeeCrew, 2020-09-14:
So I have set up a ST-Link Gdb server which connects to my STM32F303RE board just fine. I can also connect to the server for debugging with the Gdb client in the arm-embedded toolchain package.

Now i try to setup Cortex-debug for debugging inside VS-Code by adding the ARM-Embedded toolchain path and the ST-Link server path to the plugin settings.
When i click Start debugging, Cortex-Debug trys to start the ST-Link Server as follows:
``
C:/ST/STM32CubeIDE_1.4.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.4.0.202007081208/tools/bin/ST-LINK_gdbserver.exe" "-p" "50000" "-v" "--no-reset" "-d" "-cp " "C:/ST/STM32CubeIDE_1.4.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_1.4.0.202007081208/tools/bin
``
The problem is that the ST-Link server does not know the "--no-reset" parameter.
``
PARSE ERROR: Argument: --no-reset
             Couldn't find match for argument
``
Can anyone explain how the start parameters for the ST-Link server can be changed ? How do i tell Cortex-Debug which command to use for starting up the Gdb server ?

haneefdm, 2020-09-14:
We do not have support for ST-Link Gdb Server. What we do have is support for this

https://github.com/stlink-org/stlink

It has been on my TODO list to officially support ST-Link Gdb Server but doesn't look like I will be able to until Nov.

Also, let me know your recipe for how you are doing this on the command-line and post what the output of the gdbserver looks like?

CoffeeCrew, 2020-09-14:
For starting it on the command line i followed the instructions from the official ST user manual: 
https://www.st.com/resource/en/user_manual/dm00613038-stm32cubeide-stlink-gdb-server-stmicroelectronics.pdf

The output of the GDBserver if started correctly is:

``
STMicroelectronics ST-LINK GDB server. Version 5.6.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Enabled
        SWD Debug                  : Enabled

Target connection mode: Default
Hardware watchpoint supported by the target
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J37M26
Device ID: 0x446
PC: 0x8000c3c
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.24 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Waiting for debugger connection...
Waiting for connection on port 61234...
Accepted connection on port 61234...
Debugger connected
``

The output of the Gdb Server (if tried to start with the Stutil setting)

``
PARSE ERROR: Argument: --no-reset
             Couldn't find match for argument

Brief USAGE: 
   C:/ST/STM32CubeIDE_1.4.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.e
                                        xternaltools.stlink-gdb-server.win3
                                        2_1.4.0.202007081208/tools/bin/ST-L
                                        INK_gdbserver.exe 
                                        [--external-init]
                                        [--pend-halt-timeout <Pending halt
                                        timeout>] [--halt] [-c <config
                                        file>] [-e] [-f <log file>] [-l
                                        <log level>] [-p <port number>]
                                        [-v] [-r <refresh delay>] [-s] [-d]
                                        [-z <port number>] [-a <cpu clock>]
                                        [-b <SWO CLOCKDIV>] [-k] [-q] [-i
                                        <ST-LINK S/N>] [-m <apID>] [-g]
                                        [-t] [--erase-all] [--memory-map
                                        <device id>] [--ext-memory-loaders]
                                        [-el <file_path>] [-cp <path>]
                                        [--temp-path <path>]
                                        [--preserve-temps] [--frequency
                                        <max freq kHz>] [--licenses] [--]
                                        [--version] [-h]

For complete USAGE and HELP type: 
   C:/ST/STM32CubeIDE_1.4.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.4.0.202007081208/tools/bin/ST-LINK_gdbserver.exe --help
``
(But i guess that irrelevant)

Thank you for your reply! I'm very new to this and didn't even know that there is a difference between stlink and the official st-link gdb server. For me it looked all the same. 

Just one more questions: Do you happen to know if there is a way to debug projects with ST-Link in Visual Studio IDE or Visual Studio code ?

haneefdm, 2020-09-14:
Thank you so much. And, thanks for the link to ST's doc

After the `-no-reset` problem, you would have run into another one but there is a workaround for that. Okay, the `-cp` is problematic. We didn't have to such a thing for any other gdb-server.

Let's say I add official STLink support. Are you willing to test for me? I can do it next week. I can test on a Mac and you could do it on Windows?

Do you use SWO in your project?

Don't know about Visual Studio, but yes, it is possible to debug using VSCode cpptools. It works well for basic stuff but no registers, peripherals, disassembly, memory, etc.

CoffeeCrew, 2020-09-15:
Thanks you very much for your support and your help.
The problem is that the company i am working for, has changed their goal now (so no debugging needed anymore).

But I'm also a student and still interested in debugging my private embedded projects with VSCode so i can test your code in Windows and give you feedback if it works. 

Regarding SWO I have to admit I'm not really sure. I think I am **not** using it. I develop code for a STM32 Nucleo board with build in  ST-LINK/V2-1. It's connected to my PC via USB and I've never set up SWO explicitly.

If you need some more information just write me and i will send you them :)

closed 2020-09-18 by CoffeeCrew

MrMuMu, 2020-10-26:
I would also really appreciate official st-link gdb server, I can help to test on windows.

Also maybe check [this](https://www.st.com/en/development-tools/st-link-server.html#overview). I have not find much documentation but maybe it could give some ideas.