Marus/cortex-debug #334: "FreeRTOS" breaks debug session - JLink

BarrettLowe, 2020-09-15:
Looks like #35 but with JLink.

If I set a breakpoint withOUT "rtos":"FreeRTOS" in my launch config, everything works great! When I enable free rtos, however, a few things happen:

1) Upon launching the program, vscode gives me a popup error in the bottom left "Unable to read registers: The current thread has terminated (from data-list-register-values N)."
2) If I hit continue to actually run the program, the breakpoint never hits, infact the program doesn't seem to be running
3) I am unable to pause execution (presumably because there is no program running)

Here's the weird thing (at least to me). Once it's failing, I can't just comment out the "rtos:FreeRTOS" line in my launch config to get it working again, I must also change something (add a dummy printf) and rebuild. At that point, the program runs and I can pause execution (though of course without rtos support).

Relevant launch config:
```
            ...
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "MIMXRT1064DVL6A",
            "interface": "swd",
            "runToMain": false,
            "rtos":"FreeRTOS",
            "showDevDebugOutput": true,
```

Debug console output:
```
T1064/hoists/P2/bootloader/P2_Bootloader.elf'
Finished reading symbols
Please check OUTPUT tab (Adapter Output) for output from C:\Program Files (x86)\SEGGER\JLink\JLinkGDBServerCL.exe
Launching server: "C:\Program Files (x86)\SEGGER\JLink\JLinkGDBServerCL.exe" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MIMXRT1064DVL6A" "-rtos" "GDBServer/RTOSPlugin_FreeRTOS"
Launching GDB: "C:\Program Files (x86)\GNU ARM Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-gdb.exe" "-q" "--interpreter=mi2"
1-gdb-set target-async on
2-interpreter-exec console "source c:/Users/blowe/.vscode/extensions/marus25.cortex-debug-0.3.7/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
8-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from C:\\p4\\blowe_UTB/cmakebuild/debug_iMXRT1064/hoists/P2/bootloader/P2_Bootloader.elf...\n"}]}
Reading symbols from C:\p4\blowe_UTB/cmakebuild/debug_iMXRT1064/hoists/P2/bootloader/P2_Bootloader.elf...
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"BootloaderInit () at C:\\p4\\blowe_UTB\\hoists\\P2\\bootloader\\src\\BootloaderStart.cpp:43\n"}]}
BootloaderInit () at C:\p4\blowe_UTB\hoists\P2\bootloader\src\BootloaderStart.cpp:43
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"43\t}\n"}]}
43	}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["frame",[["addr","0x7000a54a"],["func","BootloaderInit"],["args",[]],["file","C:\\p4\\blowe_UTB\\hoists\\P2\\bootloader\\src\\BootloaderStart.cpp"],["fullname","C:\\p4\\blowe_UTB\\hoists\\P2\\bootloader\\src\\BootloaderStart.cpp"],["line","43"],["arch","armv7e-m"]]],["thread-id","1"],["stopped-threads","all"]]}]}
Not implemented stop reason (assuming exception): undefined
9-thread-list-ids
GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"connected","results":[]}}
GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
10-thread-list-ids
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
Resetting target
:[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":6,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["address","0x7000b00c"],["load-size","37812"],["transfer-rate","100832000"],["write-rate","5401"]]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
Resetting target
GDB -> App: {"token":7,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":8,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
11-break-insert "c:\\p4\\blowe_UTB\\hoists\\P2\\bootloader\\src\\BootloaderStart.cpp:39"
[... 29 lines cut ...]
GDB -> App: {"token":21,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","2"],["target-id","Remote target"],["frame",[["level","0"],["addr","0x00202090"],["func","??"],["args",[]],["arch","armv7e-m"]]],["state","stopped"]]]]]}}
GDB -> App: {"token":22,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","2"]]],["current-thread-id","2"],["number-of-threads","1"]]}}
GDB -> App: {"token":23,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","0"],["value","0"]],[["number","1"],["value","0"]],[["number","2"],["value","0"]],[["number","3"],["value","0"]],[["number","4"],["value","0"]],[["number","5"],["value","0"]],[["number","6"],["value","0"]],[["number","7"],["value","0"]],[["number","8"],["value","0"]],[["number","9"],["value","0"]],[["number","10"],["value","0"]],[["number","11"],["value","0"]],[["number","12"],["value","0"]],[["number","13"],["value","0x20201000"]],[["number","14"],["value","4294967295"]],[["number","15"],["value","0x202090"]],[["number","25"],["value","16777216"]],[["number","91"],["value","538972160"]],[["number","92"],["value","0"]],[["number","93"],["value","0"]],[["number","94"],["value","0"]],[["number","95"],["value","0"]],[["number","96"],["value","0"]],[["number","97"],["value","0"]],[["number","98"],["value","0"]],[["number","99"],["value","0"]],[["number","100"],["value","0"]],[["number","101"],["value","0"]],[["number","102"],["value","0"]],[["number","103"],["value","0"]],[["number","104"],["value","0"]],[["number","105"],["value","0"]],[["number","106"],["value","0"]],[["number","107"],["value","0"]],[["number","108"],["value","0"]],[["number","109"],["value","0"]],[["number","110"],["value","0"]],[["number","111"],["value","0"]],[["number","112"],["value","0"]],[["number","113"],["value","-nan(0x7fffff)"]],[["number","114"],["value","0"]],[["number","115"],["value","0"]],[["number","116"],["value","0"]],[["number","117"],["value","0"]],[["number","118"],["value","0"]],[["number","119"],["value","0"]],[["number","120"],["value","0"]],[["number","121"],["value","0"]],[["number","122"],["value","0"]],[["number","123"],["value","0"]],[["number","124"],["value","0"]],[["number","125"],["value","0"]],[["number","126"],["value","0"]],[["number","127"],["value","0"]],[["number","128"],["value","0"]],[["number","129"],["value","-nan(0x7fffff)"]],[["number","130"],["value","0"]],[["number","131"],["value","0"]],[["number","132"],["value","0"]],[["number","133"],["value","0"]],[["number","134"],["value","0"]],[["number","135"],["value","0"]],[["number","136"],["value","0"]],[["number","137"],["value","-nan(0xfffff00000000)"]],[["number","138"],["value","0"]],[["number","139"],["value","0"]],[["number","140"],["value","0"]],[["number","141"],["value","0"]],[["number","142"],["value","0"]],[["number","143"],["v [... cut ...]
GDB -> App: {"token":32,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x00202090"],["func","??"],["arch","armv7e-m"]]],["frame",[["level","1"],["addr","0xfffffffe"],["func","??"],["arch","armv7e-m"]]]]]]}}
GDB -> App: {"token":33,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["depth","2"]]}}
34-stack-list-frames --thread 2 0 1
GDB -> App: {"token":34,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x00202090"],["func","??"],["arch","armv7e-m"]]],["frame",[["level","1"],["addr","0xfffffffe"],["func","??"],["arch","armv7e-m"]]]]]]}}
```

haneefdm, 2020-09-15:
Could you attach the output from JLink? It is the OUTPUT tab in the "Debug Adapter" secdtion.

JLink is known to give bogus stack information at the very start when RTOS is enabled

BarrettLowe, 2020-09-15:
Sure! 

```
SEGGER J-Link GDB Server V6.80a Command Line Version

JLinkARM.dll V6.80a (DLL compiled May 29 2020 16:27:00)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1064DVL6A -rtos GDBServer/RTOSPlugin_FreeRTOS
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 MIMXRT1064DVL6A
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Apr 23 2020 16:49:14
Hardware: V10.10
S/N: 850101344
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.33 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 2 bytes @ address 0x7000A048 (Data = 0xB580)
Read 4 bytes @ address 0x7000BB36 (Data = 0xF8B1F7FF)
Reading 64 bytes @ address 0x2020AC00
Read 4 bytes @ address 0x7000ACEE (Data = 0xD1F52800)
Loading RTOS plugin: GDBServer/RTOSPlugin_FreeRTOS...
RTOS plugin (API v1.0) loaded successfully
RTOS plugin: Loaded
Received symbol: pxCurrentTCB (0x20000128)
Received symbol: pxReadyTasksLists (0x2000012C)
Received symbol: xDelayedTaskList1 (0x20000294)
Received symbol: xDelayedTaskList2 (0x200002A8)
Received symbol: pxDelayedTaskList (0x200002BC)
Received symbol: pxOverflowDelayedTaskList (0x200002C0)
Received symbol: xPendingReadyList (0x200002C4)
Received symbol: xTasksWaitingTermination (0x200002D8)
Received symbol: xSuspendedTaskList (0x200002F0)
Received symbol: uxCurrentNumberOfTasks (0x20000304)
Received symbol: uxTopUsedPriority (0x00000000)
Received symbol: uxTopReadyPriority (0x2000030C)
Received symbol: vPortEnableVFP (0x70006898)
Received symbol: FreeRTOSDebugConfig (0x00000000)
All mandatory symbols successfully loaded.
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x7000BB36)
Received monitor command: reset
Resetting target
[... 77 lines cut ...]
Reading register (s24 = 0x       0)
Reading register (s25 = 0x       0)
Reading register (s26 = 0x       0)
Reading register (s27 = 0x       0)
Reading register (s28 = 0x       0)
Reading register (s29 = 0x       0)
Reading register (s30 = 0x       0)
Reading register (s31 = 0xFFFFFFFF)
Reading register (d0 = 0x       0)
Reading register (d1 = 0x       0)
Reading register (d2 = 0x       0)
Reading register (d3 = 0x       0)
Reading register (d4 = 0x       0)
Reading register (d5 = 0x       0)
Reading register (d6 = 0x       0)
Reading register (d7 = 0x       0)
Reading register (d8 = 0x       0)
Reading register (d9 = 0x       0)
Reading register (d10 = 0x       0)
Reading register (d11 = 0x       0)
Reading register (d12 = 0x       0)
Reading register (d13 = 0x       0)
Reading register (d14 = 0x       0)
Reading register (d15 = 0x       0)
```

haneefdm, 2020-09-15:
Can you try to enable `runToMain` and see how it behaves?

BarrettLowe, 2020-09-15:
Looks like there isn't much different.

I CAN however hit a breakpoint inside main() but nothing in a thread (after vTaskStartScheduler is called).

haneefdm, 2020-09-15:
We don't do anything special with JLink. We only start it and connect gdb to the server.

After that, we only deal with gdb rely on what gdb reports. Gdb in turn talks to the JLinkGDBServer. We never talk to the server directly.

You should be able to duplicate the problem even with command-line tools

BarrettLowe, 2020-09-17:
Ah - indeed. I'll search elsewhere. Thanks

closed 2020-09-17 by BarrettLowe

BarrettLowe, 2020-09-17:
Okay after doing some poking around in different places, it was that I had configUSE_PORT_OPTIMISED_TASK_SELECTION set to 1 which seems to wreck thread awareness.