Marus/cortex-debug #335: "Watch" variable radix

stephendpmurphy, 2020-09-22:
With the latest update of VScode (08-2020), I see mention of what they are calling "Contributable context menu for Variables view". It looks like they are finally enabling support for changing the radix on variables. I remember seeing some rationale for why Cortex-Debug doesn't support it currently, but I can't remember the reasoning (I believe it was a limitation of VScode and not this extension)

Does this latest VScode release enable us to finally get this feature when using Cortex-Debug?

haneefdm, 2020-09-23:
I will look into it.

FYI: You can do this in Cortex-Debug by adding a `,x` or `,b`, etc. to a watch expression.

haneefdm, 2020-09-23:
Hmm, I don't see how it can work. Every time the debugger stops, the stack is refreshed and the variables window is refreshed based on the stack. The thread/frame/variable ids are transient.

You can have a variable `blah`, it could exist in many threads/frames. When you set a radix initially we know what the context is. But after a continue/step all that context is lost and rebuilt.

However, it can work for Global/Static variables because they are not dependent on the stack. But you can already do that using the Watch window and is much more efficient that way

haneefdm, 2020-09-23:
I want to see MS implement this in cpptools before I proceed.

stephendpmurphy, 2020-09-23:
Thanks for the response @haneefdm. Just thought it ask the question, and I appreciate the tip about using `,x` or `,b`. I did not know that was a thing, and it should help a lot.

Closing this since my real question (Adjusting radix on watch variables), as has been answered.

closed 2020-09-23 by stephendpmurphy

haneefdm, 2020-09-23:
Here is the full set in ChangeLog for V0.3.0
```
Added support for formatting watch values; add the following format strings:

b - format in binary
h or x - format in hexadecimal
d - format in decimal
o - format in octal
These format sepecifiers are appended to the end of the watch expression separated by a , - eg. *(unsigned int *)(0x40011004),b would display the contents at address 0x40011004 in binary.
```