Marus/cortex-debug #337: ubuntu 20.04 ships only gdb-multiarch

WoodyAtHome, 2020-09-30:
Hi,

I'm using ubuntu 20.04. It ships binutils-arm-none-eabi (contains arm-none-eabi-objdump) but no arm-none-eabi-gdb.
There is only a gdb-multiarch package. The gdb-multiarch debugger works fine at commandline together with openocd.

Is it possible to use gdb-multiarch together with cortex-debug or even support it in future?

haneefdm, 2020-10-01:
We may but don't have the time/resources to experiment.

For now, you can use a symlink(s) an any other Linux trick

closed 2020-10-01 by haneefdm

Marus, 2021-02-12:
@WoodyAtHome - just as an FYI - one of the recent updates added the ability to set the GDB path including the executable name, so with that you should be able to use gdb-multiarch