Marus/cortex-debug #338: Additional launch configurations needed

RogFedex, 2020-09-30:
Issue Type: <b>Feature Request</b>

Iam using JLink and 
i dont want to load executables but just load symbols and i dont want to reset/restart the target after debugger is connected. 
I dont want GdbServer to verify download or initialize registers
I would like to select GDB Server and client configurations including port/swo port/Telbet port
It will be great if above configurations are available/enabled.

Extension version: 0.3.7
VS Code version: Code 1.49.2 (e5e9e69aed6e1984f7499b7af85b3d05f9a6883a, 2020-09-24T16:29:41.983Z)
OS version: Windows_NT x64 10.0.18363

haneefdm, 2020-10-01:
A lot of what you want is already possible. IntelliSense in launch.json is your friend. There are ways to override pretty much everything Cortex-Debug does for you automatically.

Beyond starting the gdb-server, we do not talk to the gdb-server.

You can use the "external" gdb-server and do what you want -- including port selection.

Let us know if something does not work

closed 2020-10-01 by haneefdm

RogFedex, 2020-10-01:
Ok.. One quick question on this topic. how do we select port number and host name on gdb client side ?
One issue iam observing is that the debugger is taking minutes to read variables or call stack after hitting a break point.

haneefdm, 2020-10-01:
There is a launch.json option for "gdbTarget" I think where you have to specify `"gdbTarget": "host:1234"`

`host` can be any IP host-name resolvable on your network. Usually "localhost", `1234` is the port that gdb needs to use.

RogFedex, 2020-10-01:
Thanks to the team for the debug extension and for your quick reply.
"serverArgs": ["-port","2331", "-swoport","2332", "-telnetport","2333",
"-speed", "2000"],
"gdbTarget": "localhost:2331"

In Debug terminal i get below message. I notice -port is assigned two
values here (50000 and 2331)
Please check OUTPUT tab (Adapter Output) for output from C:/Program Files
(x86)/SEGGER/JLink_V610n/JLinkGDBServerCL.exe Launching server: "C:/Program
Files (x86)/SEGGER/JLink_V610n/JLinkGDBServerCL.exe" "-if" "swd" "-port"
"50000" "-swoport" "50001" "-telnetport" "50002" "-device" "Cortex-M0+"
"-port" "2331" "-swoport" "2332" "-telnetport" "2333" "-speed" "2000"

Secondly, iam getting error message. Iam not sure if iam setting gdbTarget
correctly.
Failed to launch GDB: putpkt: write failed: No error. (from target-select
extended-remote localhost:50000)

If i dont set "serverArgs", the debugger is getting connected (dont see the
above error message). However after hitting the breakpoint in the code it
is taking too long time (about 2 mins) to get resume and other buttons to
get enabled. After 2 mins if i step in or step out then it takes again
another 2 mins to move to next instruction. what is actually missing ?

haneefdm, 2020-10-02:
Post your entire launch.json. I don't think you set the server to "external" and I don't think you even started the server manually.

RogFedex, 2020-10-02:
Here is the launch.json. I just tried with "external" and starting
Jlinkserver manually but i still seeing the issue of long wait after
hitting breakpoint.
I tried with and without the preAttachcommands but there is no difference.
Issue:
 after hitting the breakpoint in the code it is taking too long time (about
2 mins) to get resume and other buttons to get enabled. After 2 mins if i
step in or step out then it takes again another 2 mins to move to next
instruction

{
   // Use IntelliSense to learn about possible attributes.
   // Hover to view descriptions of existing attributes.
   // For more information, visit:
https://go.microsoft.com/fwlink/?linkid=830387
   "version": "0.2.0",
   "configurations": [
      {
         "cwd": "${workspaceRoot}",
         "executable": "./PN547/build/pn547.axf",
         "name": "Debug J-Link",
         "request": "attach",
         "type": "cortex-debug",
         "device": "Cortex-M0+",
         "servertype": "external",
//         "servertype": "jlink",
//         "serverpath": "C:/Program Files
(x86)/SEGGER/JLink_V610n/JLinkGDBServerCL.exe",
         "interface": "swd",
//         "serverArgs": ["-port","2331", "-swoport","2332",
"-telnetport","2333","-speed", "2000"],
         "toolchainPrefix": "arm-none-eabi",
         "armToolchainPath": "C:\\Users\\nxf25845\\eclipse\\cpp-2020-09\\
eclipse\\gnu\\bin",
//         "armToolchainPath": "C:\\Program Files (x86)\\GNU Arm
Embedded Toolchain\\9 2020-q2-update\\bin",
         "gdbTarget": "localhost:2331",
//         "overrideAttachCommands": "set mem inaccessible-by-default off",
         "preAttachCommands": ["set mem inaccessible-by-default off"]

      }
   ]
}

haneefdm, 2020-10-02:
Ouch, 2 mins!?!? Don't know what is going on. In this situation, there is only one thing to do.

Launch the gdb-server and gdb at the command-line and do exactly what Cortex-Debug is trying to do. Please enable debug mode `"showDevDebugOutput": true` and you will see the exact commands being sent to gdb in the 'Debug Console'. You don't have to use the machine interface but do something similar.

If you cannot get the command-line to work, there is no hope of Cortex-Debug working. See what is going on on the gdb-server stdout/stderr. If it is taking forever to respond, things time-out and you may have to contact SEGGER/chip-vendor

RogFedex, 2020-10-02:
If i use Eclipse, i do not see this issue.. so something missing here.
With showDebugOutput, i see the below message right after break point is hit
53-stack-info-depth --thread 1 10000
54-stack-info-depth --thread 1 10000

And then after almost 2 mins later, lot of messages are poring out starting
from...
GDB -> App:
{"token":53,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["depth","10000"]]}}
55-stack-list-frames --thread 1 0 19
.....

Not sure if this information is helpful but just sharing.

As i mentioned, if i use eclipse for debugger client and for starting
debugger server i dont see the 2 mins hung after BP is hit..

Thanks for your support.. if you get any clue now or in the future let me
know. It will awesome to use the debugger with vscode, cant really wait for
the same.

RogFedex, 2020-10-03:
Just FYI: During the 2 mins, i notice in the Debug output window the
debugger is continuously reading a a particular address in memory
.....
.....
Read 4 bytes @ address 0x0020E36E (Data = 0xA801E7DC)
Read 4 bytes @ address 0x0020E36E (Data = 0xA801E7DC)
Read 4 bytes @ address 0x0020E36E (Data = 0xA801E7DC)
Read 4 bytes @ address 0x0020E36E (Data = 0xA801E7DC)
Read 4 bytes @ address 0x0020E36E (Data = 0xA801E7DC)
.....
.....

Not sure why debugger is continuously reading the memory

haneefdm, 2020-10-05:
I doubt it is Cortex-Debug doing that. Do you have anything in the Watch window?

0x0020E36E address is code/flash so there should be nothing interesting to read there. Gdb should already know what is at that address. Do you know from the map file what could be there? I believe (not 100% sure) that it is gdb making those requests. All we asked for was the current stack information.

RogFedex, 2020-10-07:
I tried installing the latest version of Segger Jlink gdbserver and Armtool
and it seems working fine. The 2 mins hung is not observed today but i may
have to work little more to confirm on the same.
On quick look, it feels far better and convenient to use the debugger with
Vscode compared to eclipse.. All credit to your team !!!

RogFedex, 2020-10-08:
The 2 mins hung issue have shown up again today..

Read 4 bytes @ address 0x0020E386 (Data = 0xA801E7D4)
Read 4 bytes @ address 0x0020E386 (Data = 0xA801E7D4)
Read 4 bytes @ address 0x0020E386 (Data = 0xA801E7D4)
Read 4 bytes @ address 0x0020E386 (Data = 0xA801E7D4)
......

Yes the address (0x0020e386) is in the code area and not related to the
breakpoint set - however in the call_Stack i see the address showing up
(incorrectly). Line 474 in  phKernel_Process.c is at address - 0x0020e8d6.
None of the addresses in the call_stack seems correct anyways.

phNci_System_SetPwrMode@0x0021c3b0
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\phNci\src\phNci_System.c:200)
phHostFe_Process_TypeData@0x00211ed8
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\phFrontEnd\src\phHostFe.c:1948)
phHostFe_Process@0x00211f1e
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\phFrontEnd\src\phHostFe.c:1109)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)
phKernel_Process_Task@0x0020e386
(c:\_ddm\Venus_Ceres_Maintenance\PN547\comps\phKernel\src\phKernel_Process.c:474)

RogFedex, 2020-10-10:
If debugger is disconnected at this hung time, the below error message is
noticed

Failed to get Stack Trace: Invalid thread id: 1 (from stack-info-depth
--thread 1 10000)

RogFedex, 2020-10-15:
limiting backtrace helped and not observing the hung with
"preAttachCommands": ["set backtrace limit 6"],