Marus/cortex-debug #339: Debugging dual cores with a single debug configuration

RolfNoot, 2020-10-03:
Hi,

Currently I am using a dual core micro (PSoC6).

In order to debug simultaneously on two cores I have to start two separate debugging configuration sessions:
1- CM0 launch: starts OpenOCD, programs flash and starts a debugging session on CM0
2- CM4 attach: connects to OpenOCD (external server, localhost:50001) and attaches to CM4 core

Is it possible to start both sessions within a single configuration (single click on the green "Start Debugging" button)?

Best,
Rolf

haneefdm, 2020-10-04:
Unfortunately no. While PSoC6 is dual-core, it is asymmetric. Each core is running a completely different program. You will need two instances of VSCode each running a different debugger session. But, this creates a problem with OpenOCD.

You can start openocd from the command-line and use the server type as `"external"` and use `gdbTarget` to specify the TCP/IP port. There are other ways too but hard to explain

RolfNoot, 2020-10-04:
I am debugging the two cores simultaneously using one VS Code instance, which works well. I’ve got a setup with two projects merged into one .elf file. Each GDB session is using the same image but with stripped out symbols of the non-debugging core. This way, debugging the two cores inside one VS Code instance works well.

The only issue is that I need to start the two debug sessions separately. It would be handy if both sessions could be started with one click.

haneefdm, 2020-10-05:
There is a mechanism in VSCode where you can set up a session for compound debugging (more than one debug session) in launch.json Please let us know if that works.

https://code.visualstudio.com/docs/editor/debugging#_compound-launch-configurations

I have a feeling it won't work because of multiple gdb-servers but with an external gdb server, it might just work

RolfNoot, 2020-10-07:
Thanks, it works perfectly! I only had to add a delay (preLaunchTask) before the CM4 GDB instance connected to OpenOCD in order to have OpenOCD finished programming.

It's all working right now, setting breakpoint(s) at whatever code and the right core will halt, able to view ram/flash/variables etc.

One other question (or should I open another issue): can I limit the amount of breakpoints for a certain debug session? The maximum amount for CM0p is 4 and if I set more, all wil fail.

Best,
Rolf

[image]

haneefdm, 2020-10-07:
I am glad you figured it out.

May I see you `launch.json` and `openocd.tcl` files? I can have these changes incorporated into MTB's next release or even sooner as a patch. Your entire recipe really. Did you start by running `make vscode` to create the initial files?

Unfortunately, Cortex-Debug cannot check or verify the max number of break/data points. Idea is that it should fail gracefully when you run out.

haneefdm, 2020-10-07:
And, yes adding a delay makes sense before an attach

RolfNoot, 2020-10-07:
### 1
I use cmake to generate the make files. This allows me to add / remove files to the project without having to edit the makefiles. Both projects (CM0p and CM4) are compiled and build using each project's cmake files. Finally the two separate build outputs are merged with the cymcuelf tool to generate two outputs for each project with debug symbols accordingly (you probably don't want the CM0p signed with a private key, so that line should be changed):

```
add_custom_target(
    ${MAIN_PRJ_NAME} ALL
    # Sign the CM0p image with the private key
    COMMAND "${CYPRESS_ELF_TOOL}" --sign "source/CortexM0/${M0_PRJ_NAME}.elf" SHA256 --encrypt RSASSA-PKCS --key "${PRIVATE_KEY}" --output "source/CortexM0/${M0_PRJ_NAME}_signed.elf" --hex "source/CortexM0/${M0_PRJ_NAME}_signed.hex"
    # Sign the CM4 image
    COMMAND "${CYPRESS_ELF_TOOL}" --sign "source/CortexM4/${M4_PRJ_NAME}.elf" --output "source/CortexM4/${M4_PRJ_NAME}_signed.elf"
    # Merge the CM0p image with the CM4 image and output two .elf files with symbols for each project
    COMMAND "${CYPRESS_ELF_TOOL}" --merge "source/CortexM4/${M4_PRJ_NAME}_signed.elf" "source/CortexM0/${M0_PRJ_NAME}_signed.elf" --output "${MAIN_PRJ_NAME}_M4debug.elf"
    COMMAND "${CYPRESS_ELF_TOOL}" --merge "source/CortexM0/${M0_PRJ_NAME}_signed.elf" "source/CortexM4/${M4_PRJ_NAME}_signed.elf" --output "${MAIN_PRJ_NAME}_M0debug.elf" --hex "${MAIN_PRJ_NAME}.hex"

    # Output some usefull information
    COMMAND echo "------- M0 memory usage -------"
    COMMAND "${READELF_TOOL}" -Sl source/CortexM0/${M0_PRJ_NAME}.elf > source/CortexM0/${M0_PRJ_NAME}.readelf
    COMMAND "${BASH_TOOL}" ${CONFIG_DIR}/scripts/memcalc.bash source/CortexM0/${M0_PRJ_NAME}.readelf "131072" "32768" "269352960" "134479872" #Flash 0x020000   SRAM 0x8000   Flash start 0x100E0000     SRAM start 0x08038000
    COMMAND echo "------- M4 memory usage -------"
    COMMAND "${READELF_TOOL}" -Sl source/CortexM4/${M4_PRJ_NAME}.elf > source/CortexM4/${M4_PRJ_NAME}.readelf
    COMMAND "${BASH_TOOL}" ${CONFIG_DIR}/scripts/memcalc.bash source/CortexM4/${M4_PRJ_NAME}.readelf "917504" "229376" "268435456" "134217728" #Flash 0x0e0000   SRAM 0x38000   Flash start 0x10000000   SRAM start 0x08000000
    DEPENDS "source/CortexM0/${M0_PRJ_NAME}.elf" "source/CortexM4/${M4_PRJ_NAME}.elf"
)
```

RolfNoot, 2020-10-07:
### 2
The first CM0p launch will program the entire flash with the merged output from above (with the CM0p debugging symbols). The launch also uses some folders which are globally set in the environment variables of the system (the location of openocd etc.) 

```
        {
            "name": "Launch CM0 (KitProg)",
            "type": "cortex-debug",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/${workspaceFolderBasename}_M0debug.elf",
            "servertype": "openocd",
            "serverpath": "${env:ONETHINX_TOOLS_LOC}/openocd/bin/openocd",
            "searchDir": [ "${env:ONETHINX_TOOLS_LOC}/openocd/scripts" ],
            "numberOfProcessors": 2,
            "targetProcessor": 0,
            "openOCDPreConfigLaunchCommands": [
                "set PROGRAMMER kitprog3",
                "set ENABLE_ACQUIRE 1",
            ],
            "openOCDLaunchCommands": [
                "tcl_port disabled",
                "telnet_port disabled",
                //"cmsis_dap_serial 1B1D0F8B02237400",            // Enter the serial of the Miniprog in case more programmers are connected
                "kitprog3 power_config on 3300"
            ],
            "configFiles": [
                "${workspaceRoot}/dependencies/scripts/openocd.tcl"
            ],
            "overrideLaunchCommands": [
                "set mem inaccessible-by-default off",
                "mon psoc6 sflash_restrictions 0",                  // SFLASH prohibited
                //"mon psoc6 sflash_restrictions 1",               // USER + KEY + TOC2
                // "mon flash erase_address 0x16000800 0x800",      // USER
                // "mon flash erase_address 0x16005A00 0xC00",      // KEY
                // "mon flash erase_address 0x16007C00 0x400",      // TOC2
                "mon flash erase_sector 0 0x700 0x7FF",             // BANK0 = FLASH SECTOR 700-800 = 0x100E0000 - 0x100FFFFF
                "-enable-pretty-printing",
                "monitor reset init",
                "-target-download",                                 // Comment this out if you don't want to reload program
                //"set output-radix 16",                            // Comment this out if you want decimal output instead of hexadecimal
                "monitor reset run",
                "monitor sleep 200",
                "monitor psoc6 reset_halt sysresetreq"
            ],
            "postStartSessionCommands": [       // Needed if runToMain is false
                /*
                // Following two commands are needed to get gdb and openocd and HW all in sync.
                // Or, execution context (PC, stack, registers, etc.) look like they are from before reset.
                // The stepi, is a pretend instruction that does not actually do a stepi, but MUST be done
                // Its a documented workaround in openocd. Do a 'monitor help' to see more info
                //
                // An alternative command to use is "continue" instead of the following two
                */
                "monitor gdb_sync",
                "stepi"
            ],
            "overrideRestartCommands": [
                "monitor reset init",
                "monitor reset run",
                "monitor sleep 200",
                "monitor psoc6 reset_halt sysresetreq"
            ],
            "postRestartSessionCommands": [
                "monitor gdb_sync",
                "stepi",
                "tbreak main"
            ],
            // svdFile is optional, it can be very large.
            "svdFile": "${env:ONETHINX_SVD_LOC}/psoc6_01.svd",
            "runToMain": true,                  // if true, program will halt at main. Not used for a restart     
           // "preLaunchTask": "Check build",     // Set this to run a task from tasks.json before starting a debug session
            "showDevDebugOutput": false,        // Shows output of GDB, helpful when something is not working right
        }
```
This is the tcl file I use, which uses some variables I added to the session's openOCDPreConfigLaunchCommands.
```
source [find interface/${PROGRAMMER}.cfg]
transport select swd
source [find target/psoc6.cfg]
#if {$PROGRAMMER == "kitprog3"} { kitprog3 power_config on 3300 }
if {[info exists ENABLE_CM0] && ${ENABLE_CM0} == 0 } {
    #${TARGET}.cm0 configure -gdb-port disabled
}
if {[info exists ENABLE_CM4] && ${ENABLE_CM4} == 0 } {
   #${TARGET}.cm4 configure -gdb-port disabled
} else {
    ${TARGET}.cm4 configure -rtos auto -rtos-wipe-on-reset-halt 1
}
```

RolfNoot, 2020-10-07:
### 3
The CM4 session uses the attach procedure to connect to the CM4. As OpenOCD is already running 'externally', it doesn't need an OpenOCD script.
```
        {
            "name": "Attach CM4 (KitProg)",
            "type": "cortex-debug",
            "request": "attach",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/${workspaceFolderBasename}_M4debug.elf",
            "servertype": "external",
            "gdbTarget": "localhost:50001",
            "searchDir": [ "${env:ONETHINX_TOOLS_LOC}/openocd/scripts" ],
            "numberOfProcessors": 2,
            "targetProcessor": 0,
            "openOCDPreConfigLaunchCommands": [
                "set PROGRAMMER kitprog3",
                "set ENABLE_ACQUIRE 0",
                //"set ENABLE_CM0 0"
            ],
            "overrideAttachCommands": [
                "set mem inaccessible-by-default off",
                "-enable-pretty-printing",
                "set output-radix 16",                  // Comment this out if you want decimal output instead of hexadecimal
                "monitor halt"
            ],
            "overrideRestartCommands": [
                "monitor reset init",
                "monitor reset run",
                "monitor sleep 200",
                "monitor psoc6 reset_halt sysresetreq"
            ],
            "postRestartSessionCommands": [
                "monitor gdb_sync",
                "stepi",
                //"tbreak main"
            ],
            // svdFile is optional, it can be very large.
            "svdFile": "${env:ONETHINX_SVD_LOC}/psoc6_01.svd",
            "preLaunchTask": "Sleepdelay",     // Set this to run a task from tasks.json before starting a debug session
            "showDevDebugOutput": false,// Shows output of GDB, helpful when something is not working right
        }
```

Note that the CM4 session uses the prelaunchtask to prevent it getting started before the CM0 is running.
```
        {
            "label": "Sleepdelay",
            "type": "shell",
            "command": "sleep 10",
            "windows": {
                "command": "ping 127.0.0.1 -n 10 > nul"
            },
            "group": "none",
            "presentation": {
                "reveal": "silent",
                "panel": "new"
            }
        }

```

RolfNoot, 2020-10-07:
Also the delay should be long enough to be sure the flash is programmed and OpenOCD is ready to accept GDB input.

The files above are in an 'experimental' state, and adjusted for a specific project (SFLASH commands) but should provide a good starting point.

I would love to see the more 'polished' and maybe enhanced versions which are going into MTB!

Regarding the breakpoints, they fail 'gracefully' and rather all breakpoints are failing if more than 4 are set :-(
I also see problems sometimes with clearing breakpoints (VSC fails to clear), having to type 'd' into the GDB console to clear all breakpoints during a session.

RolfNoot, 2020-10-07:
And one note on usage: the CM0p is breaking at main. So before hitting the CM4 'start debug' button, the CM0p should be already started and having the CM4 enabled (`Cy_SysEnableCM4(..);`).

RolfNoot, 2020-10-07:
FYI, the 'compound:
```
    "compounds": [
        {
            "name": "Dual",
            "configurations": ["Launch CM0 (KitProg)", "Attach CM4 (KitProg)"],
            "preLaunchTask": "Check build",     // Set this to run a task from tasks.json before starting a debug session
        }
    ]

```

PhilippHaefele, 2022-01-04:
@haneefdm As this seems to be very similar to #529 for me. We mostlikely can close this one here

haneefdm, 2022-01-04:
Yes, and this very example with this device is working with the preview releases

closed 2022-01-04 by haneefdm