Marus/cortex-debug #345: [ Feature request ]: Edit variables from the watch panel

ubidefeo, 2020-10-09:
Every time you step forward the expanded variables group closes and you have to go dig a variable again.
It could be very useful if once a variable is being watch we could double-click it and change its value.
Right now you edit the name of the watched item rather than its value

thanks
ubi

haneefdm, 2020-10-09:
The UI is controlled by VSCode and not this extension. And it keeps changing. You can always submit a request on the core VSCode github

About watch variables. They are not variables but rather expressions. Any C/C++ expression is valid and hence there is no set value.

Btw, you can use the command line gdb methods to change variables from the Debug Console

closed 2020-10-09 by haneefdm

noetzold, 2025-01-03:
Should now possible: https://github.com/microsoft/vscode/issues/130170
https://user-images.githubusercontent.com/1898161/128382301-dd415efa-a373-4f8e-b51c-04e9d457093c.gif

haneefdm, 2025-01-17:
It is rumored that MS has update the API to be able to do this. Researching...