Marus/cortex-debug #346: NIIET SVD does not work

ontuo, 2020-10-14:
Hi,

When creating a project, I try to add an svd config file and get the error:
`Unable to parse SVD file: TypeError: Cannot read property 'map' of undefined`.
Project params are:

```
"configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "build/led_run.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",

            "device": "K1921VK01T",
            "svdFile": "K1921VK01T.svd",
            "runToMain": true,
            "configFiles": [
                "interface/stlink.cfg",
                "target/k1921vk01t.cfg"                
            ]
        }
    ]
```

I thought that problem is with the svd file itself, but I checked it with cmsis checker and get this:
`Found 0 Error(s) and 88 Warning(s).`
And this [one](https://github.com/posborne/cmsis-svd/tree/master/python) parsed it right.
[Link](https://bitbucket.org/niietcm4/k1921vkx_sdk/src/master/tools/svd/K1921VK01T.svd) to svd file. 

Thanks,
tuo

xtal-ball, 2020-12-11:
I see the same error with a different SVD file. GDB parses it without any problems but cortex-debug throws that error. The root cause is that it doesn't handle the case of an empty fields section. @ontuo you should be able to work around the error by deleting lines 10823 and 10824 of your SVD.

Marus, 2021-01-20:
Will look at fixing the paring here - thanks for the example.

ontuo, 2021-01-20:
> I see the same error with a different SVD file. GDB parses it without any problems but cortex-debug throws that error. The root cause is that it doesn't handle the case of an empty fields section. @ontuo you should be able to work around the error by deleting lines 10823 and 10824 of your SVD.

Thank you! Worked for me.

Marus, 2021-01-22:
While good that you've found a work-around, and I do encourage you to use it for the time being, it is something that the parser should handle - for that reason I'm going to re-open the issue to track this particular parsing bug.

haneefdm, 2021-07-10:
Looks like this has been fixed a long time ago by PR #330 thanks to @alexclewontin. I remember merging it

I checked with the latest build, it works as intended.

closed 2021-07-10 by haneefdm