Marus/cortex-debug #348: Debugger requested to halt target on Detach instead of exit

RogFedex, 2020-10-15:
Is there configuration to overrideDetachCommands to let debugger just exit gdb instead of target being halted

In output window the below message is indicating the debugger is killed and not detached silently
Debugger requested to halt target...

haneefdm, 2020-10-16:
Not sure what is going on, Behavior should be to resume and detach and leave target running. But it depends on gdb servers, openocd does not behave. Do you know how to do it from the command line?

haneefdm, 2021-07-10:
Closing issue as there is no activity. Yes, the debug process lets the process continue but if your gdb-server is halting the processor (OpenOCD does that).

In my version of the OpenOCD, I had to fix it myself to resume on a detach (after removing any breakpoints)

closed 2021-07-10 by haneefdm