Marus/cortex-debug #349: Debug a STM32L432 using J-Link with a rest vector not at 0x08000000

gmgunderground, 2020-10-19:
Hi, I can't understand the problem, if I link my application at 0x08000000 all is working fine, but if I  link my application at 0x08004000 at 1st step the debug doesn't work. Can someone point me in the right direction? The debug window set the cursor in the right place, PC from regs seams to be right, but executing
  ldr   sp, =_estack    /* Set stack pointer */
generates a wrong PC.
What is the mistake?

Best regards

RisinT96, 2020-10-19:
The flash base is at `0x08000000`, usually that's where the vector table should be placed, the core then finds the reset handler from this table.
If you link your program to a different location, something's gotta tell the core where the vector table is.

Since you've got not bootloader, nothing does that.

gmgunderground, 2020-10-19:
Yes at 0x08000000 there is a bootloader, and the application vector table
is at 0x08004000, but doesn't work. Have I to set something to J-Link?

gmgunderground, 2020-10-19:
I have find the problem, for some reasons what Visual Studio Code displays is not right
In the following image Visual Studio Code displays the right PC value bat is not the value of R15 register
[image]
If I check with J-Link tools R15 has a different value! 
[image]
If I change the value of R15 from J-Link tool and I come back to visual studio, I can debug correctly.
How can I solve the issue?

RisinT96, 2020-10-20:
I'm having no problem debugging code that runs on top of a custom bootloader.

Are you sure your bootloader updates the Vector Table Offset Register (VTOR)  to the correct address of the program you're trying to debug?

Judging by the value of the PC, your microcontroller entered lockup mode, which means you got a HardFault and another fault after that.
That would make sense if the cpu was looking for the vector table at the wrong place, it would then try to handle interrupts with some gibberish.

closed 2020-10-21 by gmgunderground