Marus/cortex-debug #350: Error: gdb sent 32 bits for a 64-bit register (d2)

ChrisOwenIMG, 2020-10-28:
Hi there, I am trying out cortex-debug to work with an STM32-H407board, this has a 32-bit STM32F4x CPU.  I am using openocd with an olimex jtag probe, on Windows 10.  I can run code on the target OK by running openocd and gdb manually.

When I try with cortex-debug I get the following in the debug console:

[image]
and the following in the Output panel:
[image]

This is my launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/main.out",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": ["C:\\cortex-debug_investigation\\xpack-openocd-0.10.0-15\\scripts\\project\\STM32-H407_blink_FLASH.cfg"],
            "preRestartCommands": [
                "file ./main.out",
                "load"
              ]
        }
    ]
}
```
Any idea why this is failing?
Thanks a lot, Chris

ChrisOwenIMG, 2020-10-30:
Additional info: D2 is indeed a 64-bit register, associated with the FPU.  So something writing 32-bits to this register is the problem.

I got the cortex-debug source and instrumented it a bit to try and see what's going on.  I added the -d3 flag to the call to openocd, which gives the debug output in the attached file: [openocd_d3_out.txt](https://github.com/Marus/cortex-debug/files/5464483/openocd_d3_out.txt)

I also believe that the setValue method of RegisterNode class never seems to get called, at least not from within the extension, so I am none the wiser about where this register write has come from.

Thanks
Chris

haneefdm, 2022-06-17:
Cleaning up old issues. This is not a problem anymore. Registers are now hosted in the Variables window where you can set values. I was able to change dx registers.

closed 2022-06-17 by haneefdm