Marus/cortex-debug #351: pyocd gdbserver's --persist flag prevents debugging multiple times

DrTobe, 2020-10-29:
When debugging with pyocd for the first time, it is started with
```
"pyocd-gdbserver" "--persist" "--port" "50000" "--telnet-port" "50001" "--target" "stm32l432kcux"
```
When I stop debugging and try to do the same a second time, it is started with
```
"pyocd-gdbserver" "--persist" "--port" "50002" "--telnet-port" "50003" "--target" "stm32l432kcux"
```
Most probably, cortex-debug just properly sees that port 50000 is already used, so it tells pyocd to use the next available port? Unfortunately, the second pyocd instance will not be able to connect because the device is already connected to the first pyocd instance. 

https://github.com/Marus/cortex-debug/blob/master/src/pyocd.ts#L106 adds the `--persist` flag which prevents pyocd from shutting down after the first debug session. It is already commented with "not sure we need this anymore" so I would like to suggest it is removed.

Microsparky, 2020-11-08:
I have the same issue `--persist` flag is intended to "Keep GDB server running even after remote has detached." however it can only be reconnected on the same port 50000. Instead, on reconnect cortex-debug tries to connect on 50002 (which is the next available port) this times out because the target is already connected to the previous instance.

`--persist` flag does not need to be set.

@DrTobe, I found a manual workaround: before pressing stop, enter `monitor $quit()` in the debug console.

Microsparky, 2020-11-09:
I have found a workaround using the post launch task to send `monitor $quit()` to the gdb server. I thought I would share here, hopefully it helps someone!

This hack will use a `"postDebugTask"` to run a python script which closes the gdbserver. It uses [pygdbmi](https://pypi.org/project/pygdbmi/) under python3 in WSL to send  `monitor $quit()` to the gdb server on port 50000. 
Note: You may need/want to modify this to correct paths depending on your setup and/or run in Windows instead of WSL.

Install pygdbmi in WSL:
`pip3 install pygdbmi`

Create a python script in `${workspaceRoot}` called `stop-pyocd-gdbserver.py` with the code below. This will connect to the gdb server and send `monitor $quit()`.
```
from pygdbmi.gdbcontroller import GdbController
from pprint import pprint

gdbmi = GdbController()   # Start gdb process

response = gdbmi.write('target remote localhost:50000')   # Connect to remote target on port 50000
pprint(response)
response = gdbmi.write('monitor $quit()')   # Connect to remote target on port 50000
pprint(response)
response = gdbmi.exit()   # Connect to remote target on port 50000
pprint(response)
```

Add a task to `tasks.json` called `"QuitPyOCD"` which will run the python script
```
{   // Stop the PyOCD GDB server (persists due to --persist flag)
            "label": "QuitPyOCD",
            "type": "shell",
            "command": "python3",
            "args": [
                "../stop-pyocd-gdbserver.py"
            ],
            "problemMatcher": [],
        }
```
Add a `"postDebugTask"` to `launch.json` which will run the `"QuitPyOCD"` task
```
"postDebugTask": "QuitPyOCD"   // Stop the PyOCD GDB server (persists due to --persist flag)
```

patrislav1, 2020-11-11:
I found another, more lightweight approach.
Save this little Python wrapper somewhere and let cortex-debug call it by changing `serverpath` in the `launch.json`.
Should also fix the [issue about the deprecated pyocd-gdbserver command](https://github.com/Marus/cortex-debug/issues/322).

```
#!/usr/bin/python3

from pyocd.__main__ import PyOCDTool
import sys

# Workaround wrapper for these cortex-debug issues:
# https://github.com/Marus/cortex-debug/issues/351
# https://github.com/Marus/cortex-debug/issues/322
#
# Add 'gdbserver' arg and remove '--persist'

args = ['gdbserver']
args.extend([arg for arg in sys.argv[1:] if arg != '--persist'])

sys.exit(PyOCDTool().run(args=args))
```

Marus, 2021-01-20:
Will look at this - the extension should have been killing the server after the disconnect - allowing the new server to get launched.

haneefdm, 2021-02-25:
@Marus, I think we do ask pyOCD to quit but the "--persist" may be interfering with recent updates to pyOCD. A year ago, "--persist" did not hurt or help, See my comment below

```typescript
        let serverargs = [
            '--persist',        // Not sure we need this anymore
            '--port', gdbport.toString(),
            '--telnet-port', telnetport.toString()
        ];
```

haneefdm, 2021-02-25:
https://github.com/Marus/cortex-debug/blob/35f42186726063bcb150f2e4132556a7528eea60/src/pyocd.ts#L106

haneefdm, 2021-05-18:
See PR #435. Fixed in next release

haneefdm, 2021-07-10:
The `--persist` is no longer used, and we have transitioned to using the `pyocd gdbserver` method of launching pyocd rather than the old `pyocd-gdbserver`. Changes have been merged into master and will be available in the next release

closed 2021-07-10 by haneefdm