Marus/cortex-debug #352: Feature request: connect to already running JLinkGDBServer

danielschenk, 2020-11-08:
I sometimes need to do remote debugging, for which I connected a J-Link to a Raspberry Pi which is running the ARM version of JLinkGDBServer. I would like to be able to create a launch config which does not start a local instance of the server, but instead takes an IP address and passes that to gdb instead.

Marus, 2021-01-20:
I think this is probably a bit too niche of a feature to be a priority - that said - I think you can probably use a pre-launch task to do the launching of the GDB server and then use the `external` server type to connect to that one.

tpilt, 2021-02-28:
@danielschenk , are you aware of this:
https://github.com/Marus/cortex-debug/issues/361#issuecomment-763862881
?