Marus/cortex-debug #353: Keil EventRecorder + SCVD files parsing

jankap, 2020-11-09:
Hi and thanks for this great project.

I'm wondering if there are any plans to build in a parser for ARM/Keil's [SCVD file format](https://www.keil.com/pack/doc/compiler/EventRecorder/html/SCVD_Format.html) which is in fact a XML file used in Keil's [component viewer](https://www.keil.com/support/man/docs/uv4/uv4_db_dbg_scvd_viewer.htm). 

In fact, the component viewer is like a variable/register view but does not visualize register but variables/memory locations in a structured format, i.e. see
https://www.keil.com/support/man/docs/uv4/uv4_dbg_scvd_viewer0.png

I have the feeling this should not be too hard to implement since memory view already works and VSCode has good support for these hierarchical menus. However, I don't know how complicated the XML parsing is.

Question 1: is there any plan to implement SCVD parsing to do something like component viewer?

A related concept is the so called [EventRecorder](https://www.keil.com/pack/doc/compiler/EventRecorder/html/index.html). This seems to be similar to[ Segger's RTT](https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/). The idea is to have a dedicated memory area in the µC RAM that is read out or written to periodically and very fast by the debug probe. There is no ITM/ETM trace involved.
Keil uVision uses the SCVD to format their EventRecorder output, too.

Question 2: is there any plan to implement a EventRecorder output?

Thanks!

haneefdm, 2020-11-09:
Isn't this specific to the uVision debugger? VSCode only supports gdb as the debugger.

The only thing I can think of is the SWO output to record events which is already supported.

jankap, 2020-11-20:
Honestly, I'm not sure if this is only specific to the uVision debugger.

There is a software component called [ARM Compiler Extensions](https://www.keil.com/dd2/pack/): 
[image]

So this is a freely available component, written in plain C. In case you're interested, just download the shown component and unzip. It only contains 2 source files.

What the EventRecorder does is to define a RAM area where debug data, timing information and so on can be stored via relatively simple functions, I guess most of the work is to ensure it's thread save. This memory area is read out by the debugger (any probe is working, ULink, ST-Link, J-Link etc). 

That said, we should "just" read out the specified memory area continously and parse the data. I guess this is already working, since the debugger can read out any memory location? Please correct me if I'm wrong here :)

However, I don't know whether the proposed usage is allowed. The header just states this:
[image]

There no additional license file or so. Does anybody know if this component is only allowed to be used in combination with uVision?

In the mentioned module, the [SCVD ](https://www.keil.com/pack/doc/compiler/EventRecorder/html/SCVD_Format.html)file format (mentioned above) is specified, too. This not only belongs to the EventRecorder, but also to the ComponentView as described above.

I still think this all is more or less already implemented in the cortex-debug extension, since it can read out arbitrary memory locations and imo that's all we need. If that's not true I'd be very thankful if you could explain, why this is not all we need ;)

Thanks.

PS, to get an idea how this is working, please see https://www.youtube.com/watch?v=_z7hquSN31I

haneefdm, 2021-07-10:
SEGGERS RTT is definitely free. The firmware ships with the install of JLink and you can see the Copyright in there clearly says it is free to distribute and even modify so long as the copyright is preserved.

OpenOCD now supports RTT and we are in the process of supporting that in VSCode. I don't think I know how to talk to uVision folks, Since I cannot make much progress with SVCD, I am closing this issue. I am trying to cleanup our issue list. If you have further info wrt. licensing please re-open.

```
*       SEGGER RTT * Real Time Transfer for embedded targets         *
*                                                                    *
**********************************************************************
*                                                                    *
* All rights reserved.                                               *
*                                                                    *
* SEGGER strongly recommends to not make any changes                 *
* to or modify the source code of this software in order to stay     *
* compatible with the RTT protocol and J-Link.                       *
*                                                                    *
* Redistribution and use in source and binary forms, with or         *
* without modification, are permitted provided that the following    *
* conditions are met:                                                *
*                                                                    *
* o Redistributions of source code must retain the above copyright   *
*   notice, this list of conditions and the following disclaimer.    *
*                                                                    *
* o Redistributions in binary form must reproduce the above          *
*   copyright notice, this list of conditions and the following      *
*   disclaimer in the documentation and/or other materials provided  *
*   with the distribution.                                           *
*                                                                    *
* o Neither the name of SEGGER Microcontroller GmbH                  *
*   nor the names of its contributors may be used to endorse or      *
*   promote products derived from this software without specific     *
*   prior written permission.                                        *
*                                                                    *
* THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND             *
* CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES,        *
* INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF           *
* MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE           *
* DISCLAIMED. IN NO EVENT SHALL SEGGER Microcontroller BE LIABLE FOR *
* ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR           *
* CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT  *
* OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS;    *
* OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF      *
* LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT          *
* (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE  *
* USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH   *
* DAMAGE.                                                            *
*                                                                    *
**********************************************************************
```

closed 2021-07-10 by haneefdm