Marus/cortex-debug #354: Official ST-LINK gdbserver support

resonantworks, 2020-11-10:
ST's ST-LINK gdbserver from STM32CubeIDE seems to be a lot more stable than OpenOCD for the devices I have been using.

[ST-LINK gdbserver manual](https://www.st.com/resource/en/user_manual/dm00613038-stm32cubeide-stlink-gdb-server-stmicroelectronics.pdf)

On Windows, I run the st-link gdbserver in a separate command line via:
```cmd
C:\Apps\stm32\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.4.0.202007081208\tools\bin\ST-LINK_gdbserver.exe -p 3333 -l 1 -d -s -i 066CFF495652716587091651 -e -cp C:\Apps\stm32\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_1.4.0.202007081208\tools\bin -m 0 -k
```
Param|Option
-- | --
-l | logging verbosity
-d | SWD mode
-s | Flash verify
-i | ST-LINK serial number (optional - will choose the first found when left out)
-e | Persistent (do not quit - most likely do not want this when launching within vscode)
-cp | CubeProgrammer plugin bin path
-m | 'AppID to debug' (this was grabbed straight from the params STM32CubeIDE supplied launching st-link gdbserver)
-k | Initialise under reset

It also lists options for SWO (which I have yet to get around to trying):
Param|Option
-- | --
-z | SWO TCP port
-a | MCU clock speed
-b | Clock divider

I use something along the following in the launch.json settings:
```json
{
    "name": "stlink-gdbserver F401RE",
    "type": "cortex-debug",
    "request": "launch",
    "runToMain": true,
    "cwd": "${workspaceRoot}",
    "executable": "${command:cmake.launchTargetPath}",
    "servertype": "external",
    "gdbTarget": ":3333",
    "device": "STM32F401RETx",
    "svdFile": "${workspaceRoot}/Environment/svd/STM32F401.svd",
}
```

If Cortex-Debug had a new config setting for the st-link gdbserver path, then I'm sure the rest could be pattern matched from that. So given:  
`C:\Apps\stm32\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.4.0.202007081208\tools\bin\ST-LINK_gdbserver.exe`

The plugin path, platform and version could be extracted:   
plugin path: `C:\Apps\stm32\STM32CubeIDE\plugins`
OS and version: `win32_1.4.0.202007081208`

With those the `-cp` argument (CubeProgrammer plugin bin path) could be built:
`C:\Apps\stm32\STM32CubeIDE\plugins` + `\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.` + `win32_1.4.0.202007081208` + `\tools\bin`

I would add this support myself if I knew enough about hacking vscode as I feel it's such an important option where J-Link hardware is not available. And since STM32CubeIDE is cross-platform a lot of people would benefit from something like this.

If there is absolutely anything I can help out with then don't hesitate to ask!
(related: #333, @haneefdm)

resonantworks, 2020-11-10:
Also, this seems to work for attaching to the target without downloading:
```json
{
    "name": "stlink-gdbserver F401RE (attach)",
    "type": "cortex-debug",
    "request": "attach",
    "cwd": "${workspaceRoot}",
    "executable": "${command:cmake.launchTargetPath}",
    "servertype": "external",
    "gdbTarget": ":3333",
    "overrideAttachCommands": [],
    "device": "STM32F401RETx",
    "svdFile": "${workspaceRoot}/Environment/svd/STM32F401.svd",
}
```

hwmaier, 2020-12-01:
That would be so cool to support the official ST-LINK gdbserver. There is no practical alternative to it, I am currently using it by launching it externally ("servertype": "external"). But  if it could be supported as "servertype: "st-link" with automatic launch, that would make the debugging experience so much smoother.

hwmaier, 2020-12-02:
I added that feature last night, refer to https://github.com/Marus/cortex-debug/pull/365

Marus, 2021-01-18:
I'm working on this now, thanks to the PR from @hwmaier - if you want to you can try out manually installing the v0.3.8-pre2

I expect this may work now on windows and linux if you have properly working installation of the server; it may also work on macOS - although thus far I have not been able to get the official ST-Link server running on my macOS computer (it seems unable to find one of the dynamic libraries it needs as a driver for the ST-Link devices). I'll be trying to do more testing of this change tomorrow - with hopefully a release by Tuesday or Wednesday assuming there are no issues.

hwmaier, 2021-01-18:
@Marus 
> it seems unable to find one of the dynamic libraries it needs as a driver for the ST-Link devices

It seems not that easy to resolve the shared libraries paths on OSX. It seems to be same issue as with Linux. Refer to my discussion with Matthew here: https://github.com/Marus/cortex-debug/pull/365#issuecomment-743797231

PhilippHaefele, 2022-01-04:
@haneefdm As the shared library path issue has been resolved now, this can also be closed

haneefdm, 2022-01-04:
Yes, thanks. I close one earlier related to this.

closed 2022-01-04 by haneefdm