Marus/cortex-debug #355: Feature request: "socket" option for BMP's SWO

aam335, 2020-11-10:
I wrote usb<->tcp bridge for BMP's SWO usb device. This bridge makes tcp server and copy all _full_ swo packets from SWO BMP interface to all connected clients.

It is a simplest way to solve this:
`Unable to load Serial Port Module. A recent Visual Studio Code update has likely broken compatibility with the serial module. Please visit https://github.com/Marus/cortex-debug for more information.`

haneefdm, 2021-08-16:
This keeps happening every time VSCode's Electron release changes and we can't keep up. There is a script in the extension install directory you can run yourself. I should update the message by just asking the user to run the tool themselves instead of waiting for a new Cortex-Debug release.

```
serial-port-build.sh
```

But, can I get more information on the socket-based approach? Is this generic - as in for all platforms? Can you ask BMP to integrate it and support it?

haneefdm, 2023-03-10:
We no longer ship with the serial port. It can work without us shipping it and we have provided build instructions as well. If we see the serial port missing then you will be prompted on what to do.

I cannot do anything with BMP as I do not have any HW that I can use.

closed 2023-03-10 by haneefdm