Marus/cortex-debug #358: support for other ISAs

mttbx, 2020-11-17:
Can you extract the same part to support other ISAs? Maybe make a new extension. It would be helpful!

Marus, 2021-01-18:
Sorry for the slow response, it's unfortunately been a long time since I've had any time to put into the extension.

I am considering attempting to add support for other targets than ARM Cortex devices - although don't really have a timeline here. I expect my first additional target would be ESP32 devices (mostly because I've been using them in some of my personal projects). Are there particular other platforms you are interested in? Could be something to consider for the future.

mttbx, 2021-01-18:
Yeah, I'm interested in gd32vf103 (not gd32f103) chip which is riscv arch.

Marus, 2021-01-19:
Well - it's something that may be considered - can you send me links to information about the toolchain, GDB server (is it some OpenOCD variant), etc you are using and stuff - it may even be possible to get the basics running with some clever configuration properties - but no guarantee (and of course some features like SWO output simply wouldn't be applicable).

mttbx, 2021-01-19:
I can already debug it using your extension, but more support would be nice!

gdb: use toolchain or gdb-multiarch

gdbserver: riscv-openocd

`https://github.com/riscv/riscv-openocd`

toolchain: 

`https://github.com/sifive/freedom-tools/releases or platformio in vscode`

If you want to use gd32vf103, you may need its firmware library:

`https://github.com/riscv-mcu/GD32VF103_Firmware_Library`

I don't think it may need a lot of affort, because most of the jobs were done by openocd.

mttbx, 2021-01-19:
A good way to support different gdbserver would be to add interface with gdb. So we can configure it before we start to debug our program. For example add some cmd to load program or select swd/jtag protocol. Some debugger may need that.

I think this extension may need to focus on the gdb remote debugging. This is the part valuable and need flexibility. The implementation of debugging is normally the same. And all of these can be applied to all the ISAs.

closed 2021-01-23 by mttbx