Marus/cortex-debug #361: JLinkGDBServerExe: cannot connect to X server

4a-42, 2020-11-27:
I'm trying to debug a project using VS Code + WSL2 Ubuntu 20.04 LTS

The only thing I can think of is that maybe I should try uninstalling+reinstalling JLink. I remember having issues installing it and maybe I somehow broke something?

When I google search the issue, I end up here, which says it might be some sort of permissions issue:
https://askubuntu.com/questions/175611/cannot-connect-to-x-server-when-running-app-with-sudo

I'm not sure what to make of the idea that it's a permissions issue.

When I try to debug, I get this error message in the `DEBUG CONSOLE`:

```txt
Please check OUTPUT tab (Adapter Output) for output from /usr/bin/JLinkGDBServerExe
Launching server: "/usr/bin/JLinkGDBServerExe" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "nRF52840_xxAA"
```

I get this error message in the `Adapter Output`:

```txt
JLinkGDBServerExe: cannot connect to X server 
```

Here are my settings:

settings.json

```json
{
    "cortex-debug.JLinkGDBServerPath": "/usr/bin/JLinkGDBServerExe", // Linux
    "cortex-debug.armToolchainPath": "/home/4a42/gcc-arm-none-eabi-9-2020-q2-update/bin/", // Linux
    "C_Cpp.default.cppStandard": "c++17",
    "C_Cpp.default.cStandard": "c11",
    "C_Cpp.default.intelliSenseMode": "clang-arm",
    "C_Cpp.updateChannel": "Insiders",
    "cmake.cmakePath": "/usr/local/bin/cmake", // Linux
    "cmake.configureOnOpen": false,
    "cmake.generator": "Ninja Multi-Config",
    "cmake.configureArgs": [
        "-DENABLE_CLANG_TIDY:BOOL=TRUE"
    ],
    "clang-tidy.lintOnSave": false,
    "clang-tidy.buildPath": "${workspaceFolder}/build"
}
```

launch.json

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Launch Debug",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "executable": "${command:cmake.launchTargetPath}",
            "cwd": "${workspaceRoot}",
            "device": "nRF52840_xxAA",
            "svdFile": "${workspaceRoot}/lib/nRF5-SDK/modules/nrfx/mdk/nrf52840.svd"
        }
    ]
}
```

```txt
Version: 1.51.1 (system setup)
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-10T23:34:32.027Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041
```

```txt
marus25.cortex-debug v0.3.7
```

RisinT96, 2020-11-27:
Working with WSL can be tricky, as there are multiple problems:

1. `JLinkGDBServerExe` tries to open a GUI box with progress bars when it programs/erases the device, since WSL doesnt have a GUI (no X Server), the jlink software fails.
The solution is to use `JLinkGDBServerCLExe`, it's a CLI only version.
2. WSL doesn't have USB passthrough, meaning that even if you manage to launch `JLinkGDBServerCLExe`, you still won't be able to connect to your device.
The solution is to launch a `J-link Remote Server` on your windows host, and to launch `JLinkGDBServerCLExe` with a `-ip` `xxx.xxx.xxx.xxx` argument. 

As a side note, Microsoft recently announced that they will be adding GUI support to WSL: https://devblogs.microsoft.com/commandline/whats-new-in-the-windows-subsystem-for-linux-september-2020/#gui-apps.

4a-42, 2020-11-27:
That is very insightful for me. Thanks. 

I played around with your suggestion of running `JLinkRemoteServer.exe` and connecting with `JLinkGDBServerCLExe`; it looks like `JLinkGDBServerCLExe` is supposed to talk to gdb, not `JLinkRemoteServer.exe`. I couldn't really find anything here that suggested otherwise <https://wiki.segger.com/J-Link_GDB_Server>

Setting up gdb to talk to `JLinkGDBServerCL.exe` while working nicely with vscode without having an extension that has it all figured out is a rabbit hole I don't think I want to go into right now, so I'm going to continue using this extension with the windows arm embedded toolchian.

4a-42, 2020-11-27:
By the way, I get this error if I try setting `"cortex-debug.JLinkGDBServerPath": "/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe"`

[image]

Here's "Debug Console"

```txt
Please check OUTPUT tab (Adapter Output) for output from /mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe
Launching server: "/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "nRF52840_xxAA"
Launching GDB: "/home/4a42/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /mnt/c/blahblah/build/Debug/blinky.elf...
```

Here's "Adapter Output":

```txt
SEGGER J-Link GDB Server V6.80a Command Line Version

JLinkARM.dll V6.80a (DLL compiled May 29 2020 16:27:00)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nRF52840_xxAA
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nRF52840_xxAA
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Mar 17 2020 14:43:00
Hardware: V1.00
S/N: xxxxxxxxx
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
```

RisinT96, 2021-01-20:
All you have to do is to launch the `JLinkRemoteServer` on your windows side, and launch `JLinkGDBServerCLExe` with the argument `-ip {insert windows host up here}`

As documented here:
https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/

Cortex debug actually already exposes that functionality via the `ip` configuration in the `launch.json`

I can confirm that it works for me (quite well actually), even with the normal GDB server, it does complain about not being able to connect to the x server but work nonetheless.

tpilt, 2021-02-28:
> I can confirm that it works for me (quite well actually), 
me too

>it does complain about not being able to connect to the x server but work nonetheless.
Yes, the 'CL' version actually paints a gui if you start an x-server, e.g. VcXsrv.
I have no idea how that mades sense ...

hwmaier, 2021-08-09:
You can disable the progress bar and probably prevent this issue if you don't run a GUI by passing `serverArgs": ["-nogui"] in your  launch.json. That prevents the JLink DLL to show a GUI progress window.

haneefdm, 2021-09-09:
In the next release, there will be no more graphical display with JLink. If you want to enable it, you have to use `-gui` in `serverArgs`. It is unnatural for a command-line tool to be running a GUI as this makes many things complicated.

bmcdonnell-fb, 2021-10-04:
@haneefdm,

> no more graphical display with JLink. If you want to enable it, you have to use `-gui` in `serverArgs`.

Is this documented somewhere? If not, can you add it?

I don't see it in the [J-Link-Specific-Configuration wiki page](https://github.com/Marus/cortex-debug/wiki/J-Link-Specific-Configuration), which seems like the right place.

haneefdm, 2021-10-04:
You are free to edit that page. It is a wiki.

bmcdonnell-fb, 2021-10-04:
@haneefdm, I added a [blurb](https://github.com/Marus/cortex-debug/wiki/J-Link-Specific-Configuration/_compare/2816a30687083ba767c5426de216f90ef589ac7b) just now. Up to you if it makes sense to describe it more fully.

PhilippHaefele, 2022-01-04:
> You can disable the progress bar and probably prevent this issue if you don't run a GUI by passing `serverArgs": ["-nogui"] in your launch.json. That prevents the JLink DLL to show a GUI progress window.

Did someone test if the error `cannot connect to X server` still happen with the current version and the J-Link commandline version?

haneefdm, 2022-01-04:
I think we should close this. We specifically pass the -nogui option and that is all we know to do.

If they are still seeing the X server problem then it is best they contact SEGGER. Not much we can do as it is like a WSL issue that has its own problems/solutions.

We should let them reopen or open a new issue if they can demonstrate the problem

closed 2022-01-04 by haneefdm