Marus/cortex-debug #362: Multiple targets using OpenOCD

margce, 2020-11-30:
Thanks for the plugin, everything seems to be working fine except that when I want to run more than one instance of the cortex-debugger it doesn't work automatically. 

My environment as follows:
- cortex-debug v0.3.7
- openocd v0.10.0
- MacOS 10.14.6

I noticed that the pluggin will automatically increase the gdb_port starting from 50000 but it doesn't change the tlc_port (4444) and telnet_port (6666) from the default values, so when trying to open another instance of openocd it fails because the ports are already in use.

In my case I have several CAN devices so I generally run more than one debug session at a time. The way I have fixed it is by passing the following launch commands to openocd

```
            "openOCDLaunchCommands": [
                "tcl_port 6667",
                "telnet_port 4445"
            ]
```

Am I missing something on the configuration or is this expected? This is how my launch config looks like

```
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/port/STM32F3XX/Output/Target.elf",
            "name": "Debug IQ-RMONITOR",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "openOCDLaunchCommands": [
                "tcl_port 6667",
                "telnet_port 4445"
            ],
            "osx": {
                "configFiles": ["port/STM32F3XX/debugOpenOcd.cfg"],
                "armToolchainPath": "/usr/local/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/",
            },
            "windows": {
                "configFiles": ["port/STM32F3XX/debugOpenOcd_windows.cfg"],
                "armToolchainPath": "c:/Tools/gcc-arm-none-eabi/bin",
            },
            "runToMain": false,
            "serverArgs": [
            ],
            "preLaunchCommands": [
                "symbol-file port/STM32F3XX/Output/Target.elf",
                "monitor reset halt",
                "break mainApplication",
            ], 
            "postLaunchCommands": [
                "continue"
            ]
            ,"preRestartCommands": [
            ],
            "svdFile": "${workspaceFolder}/port/STM32F3XX/STM32F303.svd"
        },

```

haneefdm, 2020-11-30:
For now, that is the expected behavior. You can also start openocd from the command line and use 
```JSON
"servertype": "external",
"gdbTarget": "hostname:port"
```
If you are using multiple instances of OpenOCD, then you have to do what you're doing except specify different port numbers for tcl_port and telnet_port

haneefdm, 2021-08-01:
All three ports are assigned and incremented according to how OpenOCD does it. In addition, you can specify `numberOfProcessors` and `targetProcessor` so that gdb can connect the appropriate processor and Cortex-Debug reserves the right number of tcp ports.

closed 2021-08-01 by haneefdm