Marus/cortex-debug #364: Feature request: Improved variables view

KrestenHelstrup, 2020-12-01:
Hi
I have two feature requests for the variables view.
1) Allow us to format numbers between hex/dec
2) If you select copy value on an array it should copy all members of the array for easy extraction of data.

Thanks in advance

791314247, 2020-12-01:
顶，强烈要能选hex

haneefdm, 2020-12-01:
You can do this in the Watch window. You do `expr,x` for hex, `expr,b` for binary etc. Pretty much what gdb supports.

Variables window UI is controlled by VSCode but they recently added a feature to control the format but it is unclear how it works across data types and stack frames.

Sorting Globals by name is possible

KrestenHelstrup, 2020-12-02:
Hi haneefdm
I saw the answers from weinand in this thread : https://github.com/Microsoft/vscode/issues/28025
It let me to believe that cortex-debug must implement the formatting somehow. Unfortunately, I'm an embedded developer so I probably can't contribute much to the development.

Do you mean that sorting globals are already a possibility or it is possible to update cortex-debug to do it ?

Thanks in advance

bigbrett, 2020-12-12:
HUGE +1 for this issue. Seems like VSCode supports debuggers implementing this formatting now. This is one of the last features required to make VSCode the best embedded IDE out there!

KrestenHelstrup, 2021-01-18:
@Marus - Is this something that you would consider looking into ?
Alternatively, do you think we could find something in another debug extension we can port ?
Thanks in advance

haneefdm, 2021-01-19:
@KrestenHelstrup I have worked with weinand before. While he added that new feature, we have an issue as well.

We do not interpret values. We just report them as GDB does. Things can get messy and prone to error. Esp. floating point. It should be possible.

One thing that is not clear to me is how to remember which variable has which format between frames and every time you get a breakpoint

KrestenHelstrup, 2021-01-19:
@haneefdm Would it make sense to Weinand for some input ? Maybe he can point in the right direction. 
I'm up for contributing with testing/debugging, but my coding contributions would probably be limited as I'm an embedded C developer.

sidking, 2021-02-05:
Please, please add hexadecimal display for variables in VS Code.  Love the Marus extension and VS Code; adding this would make every embedded developer so much more productive (and happy).

KrestenHelstrup, 2021-04-27:
@Marus & @haneefdm - I have figured out that you can change all variables from hex to decimal and back by writing "set output-radix 16/10" in the debug console. Is it possible to implement an easy way to switch between these ?
That could be a quick way to implement this feature without too much work ?

haneefdm, 2021-04-27:
@sidking You can try creating gdb aliases (or even macros) in your ~/.gdbinit file

https://www.zeuthen.desy.de/dv/documentation/unixguide/infohtml/gdb/Aliases.html

See the end of that page, which may be more appropriate. I have not tried this out but you could do

```
alias rd = set output-radix 10
alias rx = set output-radix 16
```

You maybe thinking buttons or menu items but those take many more mouse/keyboard actions

lagerholm, 2021-06-04:
@haneefdm This didn't work for me.

However this did:
```
alias sor = set output-radix
```
And then used as:
```
sor 10

sor 16
```

Thanks for the tip!

I would also like to add that this is a really sought after feature for many. Myself included. With this `cortex-debug` would be perfect! 😄

hwmaier, 2021-08-13:
> Sorting Globals by name is possible

How can this be achieved?

haneefdm, 2021-08-13:
Sorting (just for Globals) not that hard. This got lost among other things

hwmaier, 2021-08-13:
Ah. I thought this would be a user setting or gdb command. So it sounds like it has to be implemented in the extension.

haneefdm, 2021-08-13:
It is done. It will be in the next release. We don't rely on gdb for globals/statics functions and variables. We get that directly from the elf file. This will add a tiny bit to startup time -- if you have thousands of globals. gdb is needed for locals/scoped variables though.

haneefdm, 2021-08-13:
A side effect or sorting globals is that all the variables starting with `_` float to the top. Esecially ones that start with two or more underscores, which may not be that interesting. Variables with single underscores are generally interesting but not when you have 2 or 3 or 4 underscores -- meaning the author of the C module is trying to hide them or is an intrinsic that you don't mess with.

Oh well, lesser of the two evils

hwmaier, 2021-08-13:
One option would be (but I am asking here too much I suppose) after the list is sorted to find the the first non-underscore entry and then splice this part of the array and insert it at the beginning, so all the underscores move to the end.

haneefdm, 2021-08-13:
@hwmaier  Yes, good idea and 99% sure that people will complain that I don't know how to sort. I thought of putting them all into a (fake) scope under Globals but that is way too much work and likely to break something -- ripple effect.

haneefdm, 2021-08-13:
Okay, this is the best I could about displaying values. It works in Variables and Watch windows for simple values that look like hex or decimal values. There have been times when I could not get the tooltip to popup. Again, this UI element doesn't belong to us and VSCode does not make any promises. I am using the `type` information to display this. Should not break anything ... still looking.

[image]

That whole thing about adding stuff to the context menus did not go anywhere. Sure I can launch something from there but I can't actually change what you see. The thread microsoft/vscode#28025 ended up being not very good and I have to use undocumented APIs and still can't do what we need. The gdbinit is still your best bet and maybe the tooltip can helps as well. And no, can't copy from tooltip

hwmaier, 2021-08-14:
> and 99% sure that people will complain that I don't know how to sort. 

You are right about that. cppvsdbg sorts locals in simple alphabetical order. Underscored vars appear first. Maybe don't trouble yourself about the underscores and just do a simple sort as the cppvsdbg debugger does. That would be a huge improvement making finding variables so much easier. 

> Okay, this is the best I could about displaying values.

If you offer a hover with hex/bin/oct view, you are miles ahead of Microsoft's C++ debugger cppvsdbg which offers only a basic hover. This will be very useful.

haneefdm, 2021-08-16:
Changes in 0.4.2 release. See [Changelog](https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md)

From the original
> Allow us to format numbers between hex/dec.

Yes, but it is all or nothing, ie not on a per variable basis. But registers and variables can be controlled independently.

> If you select copy value on an array it should copy all members of the array for easy extraction of data.

Nope. Tons of work is required because these Windows do not belong to us. They are VSCode's. Perhaps you can file an issue with the core VSCode/Microsoft

> Sort globals by name.

Done. However, variables with double underscores are at the bottom

Closing this issue as most of it is done. Feedback is welcome.

closed 2021-08-16 by haneefdm