Marus/cortex-debug #366: how to runtomain when pressed reset command

wyl2008056, 2020-12-02:
how to runtomain when pressed reset command

KrestenHelstrup, 2020-12-02:
Hi @wyl2008056 
I also had issues with runToMain, so instead I use postLaunchCommands and PostRestartCommands. See attached part of my launch file.

            "postLaunchCommands": [
                "set mem inaccessible-by-default off",
                "monitor clrbp",
                "monitor reset",
                "break main",
                "continue"
            ],
            "postRestartCommands": [
                "break main",
                "continue"
            ]

Br

wyl2008056, 2020-12-07:
Hi，@KrestenHelstrup
thank you !!! 
it working for me.

KrestenHelstrup, 2021-02-02:
@Marus - Is this expected behavior or is it a bug in runToMain ?

TorBlackbeard, 2021-02-27:
"postRestartCommands": [
            "break main",
            "continue"
        ]
Did the trick for me!

Run-to-main works fine for me, so I didn't take the other bits from @KrestenHelstrup's solution.

Just to add my comment on the "Restart" button":

When breaked somewhere in the code, and I press "Restart", the GUI just blinks one time, and the same line is highlighted, suggesting that it did not restart. 
But the it did! 
Proof is, that after pressing "Step over" once, the gui now updates, and highlights the second line i main().

But the above fix does the trick. 

Only annoyance left is: if launch has a "preLaunchTask" (to build the code), that will be executed as well. This can be annoying, if my build is slow. I just want to restart debugging the exact same elf file again.

KrestenHelstrup, 2021-02-27:
Sounds like a mistake if restart is running preLaunchTask again.

haneefdm, 2021-02-27:
This behavior was intentional according to @Marus because processors can very limited breakpoints (as low as 2). Instead of using `runToMain` you can always set a breakpoint in main yourself.

tpilt, 2021-02-27:
Thanks @KrestenHelstrup , and sorry : moved that part of the question to a new ticket:
https://github.com/Marus/cortex-debug/issues/403

closed 2021-07-10 by haneefdm