Marus/cortex-debug #369: vscode November update seems to break breakpoints in assembler files

xnorbt, 2020-12-15:
Since updating Visual Studio Code to current version 1.52.0 (November 20) it seems that setting breakpoints in ARM assembler files is no longer working in Cortex-Debug. It still works in C source files, though.
More precisely, it is no longer possible to set or remove the red breakpoint dots next to line numbers. Manually setting function breakpoints in the 'Breakpoints' tab in debug menu still works.
Rolling back to VSCode 1.51.1 and disabling auto-updates solved the issue, so it seems to have something to do with the november release.

VSCode 1.52.0, Windows 10 x64, Cortex-Debug 0.3.7 with OpenOCD as debug server.

mattico, 2020-12-16:
I'm also having this issue with Rust files (.rs) but not with C files.

mattico, 2020-12-17:
This was fixed in 1.52.1: https://github.com/microsoft/vscode/issues/112564

Marus, 2021-02-14:
This should be properly fixed in v0.3.12. Going to close the ticket.

closed 2021-02-14 by Marus