Marus/cortex-debug #370: Remember scroll position in Cortex Peripheral panels

timblakely, 2020-12-17:
Hopefully this is an easy one for ya :) Love that the Peripherals panel remembers which peripherals/registers are expanded, but it seems whenever I restart debugging the panel's scroll is reset back to the top (possibly after re-parsing the SVD?). This makes debugging peripherals towards the end of the alphabet like `TIMx`, `WWDG`, etc a bit of a chore since it requires re-scrolling down the peripherals panel to find it again. It would be ideal if the panel could remember the last scrolled position and reposition itself there upon next debugging session.

Marus, 2021-01-18:
Unfortunately, while this would be very nice, I don't think this will with the current extension APIs that I have for visual studio code - it is something I will look into though.

manuargue, 2021-01-19:
@timblakely As a workaround for avoiding scrolling down to the bottom, you can focus on the peripheral view and start typing the peripheral name for searching it.

Probably it would be nice to have some sort of "pinning" button on each peripheral so that it can be brought to the top until unpinned.

manuargue, 2021-01-20:
@Marus I could add a "pin" button on each peripheral node so that it stick to the top of the peripheral list. This state can be persisted. It's handy when debugging the same peripheral frequently. What do you think? I can work on this.

Marus, 2021-01-21:
@manuargue - Pin/Unpin at the peripheral layer, which would just sort those ones to the top, I think is a great idea, and I think would help a lot with what @timblakely actually wants to accomplish. Shouldn't be too hard to actually achieve either I don't believe.

If you want to take a stab at this I'd be happy to look at it when ready and include it in the releases.

timblakely, 2021-01-21:
Absolutely, the ability to pin a device to the top of the list would solve it completely, and likely be even more useful when working in situations involving multiple subsystems (DMA, interrupts, etc)!

manuargue, 2021-01-21:
Great, I'll work on this and try to have the pr opened by this weekend.

timblakely, 2021-01-26:
Works like a charm, thank ya much! Unless you've got additional work to do, feel free to close this bug.

Marus, 2021-01-26:
Closing this since the alternative of pinning was released. Thanks again for the PR @manuargue

closed 2021-01-26 by Marus