Marus/cortex-debug #372: AArch64 host support

tbx1024, 2021-01-02:
With the release of M1-based Macs, would it be possible to add aarch64 to supported platforms for this extension? This should also enable use on modern Raspberry Pi boards.
Thank you in advance.

Marus, 2021-01-18:
I'm currently looking into adding support on aarch64 (and 32-bit arm) on Linux to support Raspberry Pi 4 boards - hoping to get this released relatively soon (at least the aarch64 version).

M1-based Macs might be a little bit more difficult as I don't currently have access to any M1 based Mac to build the binary extensions on or test against; although that is something I'm hoping for the future.

Marus, 2021-01-31:
I've added support for linux arm and arm64 to the binary modules. Unfortunately I haven't been successful in getting the binary modules to cross-compile for macOS arm64 - so until I get access to an M1 based Mac I'm not sure I can make real progress on support there.

Most of the functionality should actually still work (assuming you have the toolchain and GDB server working properly on an M1 - the only parts that are currently dependent on the binary modules is SWO output with a serial port input (I do have other planned features that will be dependent, but none of those are in the wild yet).

BradAndreBurns, 2021-02-23:
This is awesome. With a different entension, I'm currently trying to get remote (ssh) debugging in vs code and I have a rather crude implementation (sort of) working between my linux host and the rpi4 (raspberry pi os). My implementation sometimes works, sometimes not.

As it stands right now, if I used vs code with this extension **directly** on a pi4 on raspberry pi os (not remote ssh or anything), will it work for debugging with assembly (*.s) source files?

PhilippHaefele, 2023-02-02:
@haneefdm Correct me, but starting with 1.6.0 (because of https://github.com/Marus/cortex-debug/commit/202c0dd1f0bef42b743f5574c72bd9f51e127039), we should support that.
At least also serialport package has support for ARM64 https://serialport.io/docs/guide-platform-support#supported-platforms-and-architectures . Not sure if there's a working point open on our side?

haneefdm, 2023-02-02:
Yes, it should work with ARM64. But, I have no way of testing it. We don't have any binary dependencies we package anymore.

closed 2023-02-02 by haneefdm